bb-qq/r8152 #345: RTL8156BG support

d-zalewski, 2023-10-12:
Hi,

I found this usb network adapter that is using RTL8156BG chipset instead of RTL8156 or RTL8156B versions.

"This controller improves upon the performance issues commonly associated with the RTL8156 and RTL8156B versions."

https://www.amazon.co.uk/gp/product/B0CGNY7GDY/ref=ox_sc_act_title_1?smid=ATXC3VJYLQDP5&psc=1

Any ideas if its supported by this driver? If not one has tried it yet I'm happy to buy a pair and test it with my DS716+ NAS a Macbook Pro M1

bb-qq, 2023-11-03:
I purchased a product sold in Japan that uses the RTL8156BG and confirmed that **it works**.

As the vendor claims that it consumes 64% less power than the RTL8156, it generates almost no heat during use. I like it very much and it is worth replacing if they are using the RTL8156 product. 😄 

to all:

I have added as much information about RTL8156BG as I can find on the top page. If you find other products using RTL8156BG, please let me know.

y-pp, 2023-11-08:
我也是RTL8156BG芯片的2.5G网卡。 装了驱动后造成nas异常关机。 产品链接 https://item.m.jd.com/product/100069462834.html?utm_user=plusmember&gx=RnAomTM2bjaLwp4Wp4VxD-OmXi19Lvs&gxd=RnAokjJfaDTQmM0Q-9YkC0yIWhW2vMc&ad_od=share&utm_source=androidapp&utm_medium=appshare&utm_campaign=t_335139774&utm_term=CopyURL

TheS1R, 2023-11-08:
UGREEN CM624 works as well!

UGREEN USB C to Ethernet Adapter 2.5G, Ethernet Adapter for Laptop, Aluminum, Plug & Play, Type C(Thunderbolt 4/3) to Rj45 Network Adapter, Compatible for MacBook Air/Pro, XPS, iPhone 15/Plus/Pro/Max https://a.co/d/gPk3cV9

ydzhus, 2024-02-28:
How to understand which chipset without disassembling the adapter?

bb-qq, 2024-06-15:
One user disassembled the adapter and measured the temperature, and I will link to that report. #383

AresROC, 2024-10-06:
Can confirm RTL8156BG UGREEN does work on DS416play Braswell (Front USB), DS418play Apollo Lake (Back USB), DS1821+ (V1000, DSM 7.2)

The same 2.5gbps adaper does NOT work on DS1618+ Denverton (Front & Back USB)
Version 2.18.1-1
DSM 7.1.1

Fon-SY, 2024-12-22:
https://amzn.eu/d/8aZw5Bh

Works now fine one ds720+ back.

At first LAN3 didn’t show up but surprisingly today it did appear.

kunix1979, 2025-08-28:
Hello @bb-qq ,

First of all, thank you for maintaining the drivers for Synology DSM.

I am currently using a DS920+ (Gemini Lake, DSM 7.2.2 build 72806) with a USB 2.5GbE adapter based on the Realtek RTL8156 chipset.  
The driver package installs successfully, but the adapter (revision 14) repeatedly fails with errors such as “read type fail -71 / -110” in dmesg, and the interface (eth2) cannot be brought up.

From what I understand, this issue is related to the newer **RTL8156 Rev.14** hardware revision.  
Could you please let me know if there is any plan or ongoing development to support Rev.14 in the DSM driver package?  
If there are test builds, I would be happy to try them and provide feedback.

Thank you very much for your work and support.
Best regards,