bb-qq/r8152 #346: Could you please release a new version to support DS716+ (Braswell) with DSM 6.2?

soleluna, 2023-10-13:
I have 
DS716+ (BRASWELL) / DSM 6.2 / USB3 RTL8156B 

found only drivers for DSM 7.1 and 7.2:
r8152-braswell-2.17.1-1_7.1.spk 
r8152-braswell-2.17.1-1_7.2.spk 

-> Could you please release a new driver for DSM 6.2?
Thank you for any help, best regards, 
Klaus

Sjekke, 2023-10-13:
Why are you still on 6.2?

soleluna, 2023-10-13:
good question - I am still using OPENERP. This 3rd party app is not available with DSM7. 
Anyone experience with DSM7 running OPENERP inside DOCKER/DSM6 ?

soleluna, 2023-10-14:
I got 2G5 running with an older driver:  
r8152-braswell 2.14.0-2.spk
2500 MBit/s, Vollduplex, MTU 1500

DS716+  (BRASWELL)
Version: 6.2.3-25426 Update 3 
ULANSEN USB 3.0 to 2.5G Adapter (USBA8156B), rear side

bb-qq, 2023-10-15:
I added [r8152-braswell-2.17.1-1_6.2.spk](https://github.com/bb-qq/r8152/releases/download/2.17.1/r8152-braswell-2.17.1-1_6.2.spk), but if you are not experiencing any problems, it is probably a good idea to continue using the current version.

soleluna, 2023-10-15:
In fact, the DS716+ lost the LAN connection with the 8152-braswell 2.14.0-2 overnight.
No ping reaction - maybe the rearside USB3 went to sleep mode? Any issues known concernig SYNO USB3 sleep mode?
Can I read a log file on the SYNO?

Thank you very much for the newest version 2.17.1-1 with BRASWELL DSM6.
I will try with this version and report.
Should I uninstall the old driver before? And if so: How to?

I run before BOND with LAN1 (1Gb) / LAN2 (1Gb)  -  balanced LAN power- connected.
I got  a 2x1Gb connection With my prepared old 1Gb NETGEAR switch - but not to ** one ** single PC.
Is it possible to BOND the USB LAN3 (2.5Gb) / LAN1 (1Gb) - unbalanced - with one IP address for failsafe purpose?
In case of 2.5Gb FAIL this BOND would hopefully fall back to 1Gb.

soleluna, 2023-10-17:
> Should I uninstall the old driver before? And if so: How to?

For uninstall the old driver  r8152-braswell 2.14.0-2.spk, I used issue https://github.com/bb-qq/r8152/issues/299 
- remove via DSM
- deleted the directory (only one found)   usr/syno/etc/packages/r8152
Installed the new driver r8152-braswell-2.17.1-1_6.2.spk & rebooted 

I created a BOND between LAN 1 and LAN 3 
[image]

[image]
Both network states  seemed to be alright:
SYNO 2500 Mb/s     PC     2500 Mb/s
But:   Copy a large FILE to PC was only running with 100MB/sec .... :-(

If I disconnect LAN 1 Ethernet cable (BOND running only with USB3 2.5Gb) copy works as assumed with 210MB/sec
[image]

Confusing....
I will run the next days  DS716+ with only USB3-2.5Gb attached and report hopefully stability.
Again a big thank you very much for adding the newest driver for BRASWELL with DSM 6.2

soleluna, 2023-10-21:
LAN3  [r8152-braswell-2.17.1-1_6.2.spk](https://github.com/bb-qq/r8152/releases/download/2.17.1/r8152-braswell-2.17.1-1_6.2.spk) is running fine for four days - no issues!
USB HOT PLUG activated and succesfully tested with 
bash /var/packages/r8152/scripts/install-udev-rules
REBOOT SYNO tested - success

Everything is fine, except using second LAN connector in parallel for backdoor (DSM-Login,ssh).
LAN2    192.168.1.2    1000 MBit/s, Vollduplex, MTU 1500
LAN3    192.168.1.3    2500 MBit/s, Vollduplex, MTU 1500

On first run, copying SYNO to PC runs with ca. 200MB/sec.
Obviously using SYNO LAN3 USB 2.5Gb with R8152
After SYNO REBOOT copying from SYNO to PC runs only with 100MB/sec...
Seems that SYNO is using after REBOOT LAN2 with slow 1000Mb, 
not LAN3 with 2500Mb.
Both LAN connection are CONNECTED and shown with correct speed.

Has anyone an idea, why SYNO is not using the more speedy port LAN3?

bb-qq, 2023-10-29:
I haven't actually tried it, but I don't think Synology is designed to combine interfaces of different speeds; Linux itself has the ability to specify interface priority, but it doesn't seem to be possible to do so in the GUI.

https://serverfault.com/questions/164442/nic-active-passive-bonding-with-priority

TheS1R, 2023-10-29:
The GUI actually advises against it, but it does not prohibit it.

soleluna, 2023-10-29:
So the SYNO 2.5Gb/1Gb issue is obviously not a problem of bb-qq's marvelous 8152 driver.

My combination  is running fine for more than a week:
DS716+ (BRASWELL)
Version: 6.2.3-25426 Update 3
ULANSEN USB 3.0 to 2.5G Adapter (USBA8156B), front side

Thank you - bb-qq - again supplying the newest driver  2.17.1-1 with BRASWELL DSM6.
From my point issue can be closed.

closed 2023-10-29 by soleluna