bb-qq/r8152 #347: Just updated to DSM 7.2-64570 Update 3, lost 2500 link

aeon3, 2023-10-18:
Hello everyone, I think I messed up.

I updated my 920+ that was working fine with the 2500 driver from bb-qq and now it says:

"failed to start 'rtl8152/rt8153 driver'. please try again later or contact the package developer"

Am I out of luck? What are my options? I know i can't downgrade from this version anymore.

bb-qq, 2023-10-29:
Which version of the driver are you using?
Have you installed the driver for DSM 7.2?

hansgeorgfranke, 2023-11-26:
I was running in the same problem after updaten my DS920+ to DSM 7.2.1-69057 Update 3.
A fresh reinstallation did not help. Starting the package seems to last a lot longer than in the past.
Any hints for troubleshooting?

bb-qq, 2024-04-17:
[Full reinstallation](https://github.com/bb-qq/r8152/wiki/Troubleshooting#uninstall-completely--full-reinstallation) and upgrading to the latest driver may fix the problem.

If this does not work, please create a new issue using the template and attach the log.

closed 2024-04-17 by bb-qq