bb-qq/r8152 #349: Potential issues running TWO of the same network adapters?

Poopypants666, 2023-10-28:
Hi. I have had a very easy and problem free install for a few months on a 920+. Today I added another of the same network adapter (asus usb c-2500) and gave it a shot. It does not seem that the USB adapter is recognized at all by my 920. Does not appear in any networking settings / ifconfig etc. 

Curious if running two of the same exact adapters could/would be problematic. Would it be better to have two different adapters (potentially running of two different drivers)? 

The aforementioned adapter which is not recognized by my 920+, is working / recognized on a Windows 11 machine.

Poopypants666, 2023-10-29:
Hello! I had previous success setting up the built in 2x1GBit LAN ports with Multichannel 3 on my 920+

Then moved to a single 2.5GBit usb adapter via a powered USB hub (which also connects to my UPS and another Synology). No issues there as well. 

In hopes of improving transfer speed and to try Multichannel 3 with 2.5GBit adapters, I purchased the same 2.5GBit adapter today Plugged the adapter into that powered usb data hub and the 920+ does not recognize it. The "old" one still appears, still is working. No issues there. My onboard LAN adapters are also there, working fine but no recognition of the new 2.5 adapter.

Curious if there could/would be issues of having two of the same exact adapters, plugged into the same (powered) usb hub?

The adapter in question is the Asus usb-c2500.

I attempted to clean install the same drivers again via the r1852 github page, but not sure if that would work given it's already a sort of work around (and they were already installed on the machine). That didn't really seem to do anything. I uninstalled the driver, and tried a fresh install, of hopes the Synology would then see BOTH adapters. Sadly not, still just one. 

Fwiw, the newly purchased dongle works / is recognized in a windows 11 machine. So it's not the adapter. Could be an issue with the hub? I plugged in a bunch of other devices and they all seem to be working fine. I do see green connection lights on the hidden USB adapter in question, so it is working(ish). 

Is it better or potentially necessary to have use a different 2.5GBit adapter than the one previously installed? 

Appreciate any help or insight you can lend!

bb-qq, 2023-10-29:
This topic will help you.
https://github.com/bb-qq/r8152/wiki/Troubleshooting#multiple-identical-devices-do-not-work

Poopypants666, 2023-10-29:
OMG ! Thank you so much !!  Super appreciated.

Glad I was on the right trail with the identical network devices m!

bb-qq, 2023-11-03:
This Issue is closed. If you have any questions, feel free to re-open it.

closed 2023-11-03 by bb-qq