bb-qq/r8152 #351: Others RTL8156BG adapters

Sami32, 2023-11-08:
### Description of the problem

> In case it could be helpful to someone i post these two 2.5 Gbps Ethernet adapters using the RTL8156BG chipset:

- https://www.juplink.com/products/2-5g-usb-ethernet-dragon-network-adapter

- https://www.startech.com/en-us/networking-io/us2gc30

### Description of your products

> - the product name of your NAS model
>     - and output of ``uname -a`` command
>     - DSM version x.x
> - the product name of the ethernet dongle

### Description of your environment

> - connect directly to a PC or using a Ethernet switch
>     - PC: NIC model name or chipset name
>     - Ethernet switch: model name
> - cable category (Cat.5e or Cat.6 or Cat.7)

### Output of ``dmesg`` command

````
(paste output here, or attach as a file)
````

### Output of ``lsusb`` command

````
(paste output here)
````

### Output of ``ifconfig -a`` command

````
(paste output here)
````

Sami32, 2023-11-08:
@bb-qq Oups...i probably should have added it to the #345 instead :(
Feel free to close my issue then.

skipdashu, 2023-12-29:
https://www.amazon.com/gp/product/B0BNHK5RGL/ref=ox_sc_saved_image_6?smid=AU7F5I67DFQ5W&th=1

Wavlink USB C to Ethernet Adapter 2.5 Gbps, 2-in-1 USB to 2.5G Ethernet Adapter, USB-C to RJ45 Network Converter

This sub $24 one worked with your driver on my DSM 7.2.1-69057 Update 3, DS224+ following the directions in the readme.  Creates a LAN3 network interface.  Allows Jumbo frames.

I've ordered another to try it on my DS416j.

graphixillusion, 2024-07-07:
How to verify which chip is inside the usb adapter?

WouterJN, 2024-07-26:
have the same question: I bought the cheapest adapter on AliExpress, and it has a Rev. 3100. Does this mean it has an RTL8156B chipset?

I'm also unclear about the exact differences between the RTL8156, RTL8156B, and RTL8156BG chipsets. Only the RTL8156BG has lower power consumption?

dzuelke, 2024-09-17:
Version 31.04 (Revision 3104) means RTL8156BG!

[uni 2.5 Gbps Adapter with USB-A Plug](https://www.amazon.com/dp/B0CZDDTDDT) lists RTL8156B in the description, but the unit I received (and others in European Amazon reviews) reports as version 31.04, meaning RTL8156BG:
[image]

Littlebo81, 2024-12-02:
Looks like many if not most of the adapters use BG chip these days.  i got 3 of em for testing and all 3 r based on BG based on 

their HW id  and they all staying cool after 20 min transfer test. 

https://www.amazon.ca/dp/B0CD1FDKT1   

https://www.amazon.ca/dp/B084L4JL9K

https://www.amazon.ca/dp/B0CWV2Q6HJ

Littlebo81, 2024-12-06:
Pal  got this one 

https://www.amazon.ca/dp/B0CXBRNSC4

reported to be rev 3104   so latest BG chip

and i also got this one 

https://www.amazon.ca/dp/B0856V9FDH

and its also 3104 \ BG  

Ugreen  one [https://www.amazon.ca/dp/B0CWV2Q6HJ](url)  remains unmatched in build quality and materials used,  classy device

bb-qq, 2024-12-31:
> How to verify which chip is inside the usb adapter?

There is no easy way to determine the revision of a chip from its VID/PID. You can use the chip revision shown in the log instead.
When you use the RTL8156BG, the following log appears in dmesg:
```
r8152 2-2:1.0 eth2: chip rev 14
```
As already commented, you may refer to the Version of the UI as well.