bb-qq/r8152 #352: TUC-ET2G recognized as 1000 Mbps on DS1520+

ahensley, 2023-11-12:
### Description of the problem

Hi, it seems that the TRENDnet TUC-ET2G might not be fully compatible with the DS1520+. I installed the driver and was able to see it in the list of network interfaces, but it was listed as "1000 Mbps, Full duplex, MTU 1500". Thinking it was a negotiation issue I tried different ethernet cables, USB ports on the DiskStation, ports on the switch, etc. Only ever saw it at 1 GbE in DSM. I also confirmed that the same dongle works at 2.5 GbE on Windows, so I don't think it's a defective unit.

Eventually I just ordered another model with confirmed compatibility with the DS1520+ and have not had any issues with it, so thank you for all the work put into making that possible! 😊 Mostly I am just submitting this to go though the process to get the possible compatibility issue added to the wiki, however I am happy to help troubleshoot it if there's interest.

### Description of your products

> - DS1520+
>     - Linux Polaris 4.4.302+ #64570 SMP Thu Jul 20 00:06:38 CST 2023 x86_64 GNU/Linux synology_geminilake_1520+
>     - DSM 7.2-64570 Update 3
> - TRENDnet TUC-ET2G (v2.0R)

### Description of your environment

> - connected to switch
>     - Ethernet switch: TRENDnet TEG-3102WS
> - cable category: Cat.6

### Output of ``dmesg`` command
Will provide if there is interest in troubleshooting, as I'd need to swap back to the TUC-ET2G first.

### Output of ``lsusb`` command
Will provide if there is interest in troubleshooting.

### Output of ``ifconfig -a`` command
Will provide if there is interest in troubleshooting.

bb-qq, 2023-11-15:
Could you paste the result of `lsusb`? I would like to confirm the device is recognized as a USB3 device.

cpm1, 2023-11-21:
I'm experiencing something similar with DS918+ and an RTL8152 adapter. When first installed, it worked with full 2.5G speed, then I had to reboot the DS918+ and the adapter was gone. Hence I reinstalled the driver (made sure nothing old was left over) and while the device basically works again, it only works in 1Gbit mode.

`dmesg` reveals USB warnings like `usb 1-1: not running at top speed` when device is attached. 

`cat /sys/bus/usb/devices/1-1/speed` reveals it is running only in High Speed mode (480 Mbit) instead of Super Speed like in the beginning. I'm not sure what exactly happened as I didn't update DSM (7.2-64570 Update 1) after the initial setup of the adapter.

cpm1, 2023-11-21:
Alright .. you can somewhat disregard the above ... just after I wrote it, I made some more tests with an additional hub inbetween to no avail. Then I had the why-not-thought of turning the USB-C connector 180 degree around (my devices came with a small USB-C to USB-A adapter) and retried ... and guess what, the error message disappeared, the device connected as full Super Speed device and I can operate at 2.5G again.

So, for whatever reason, this little adapter became wonky and caused all the trouble for me. Probably no help to the thread creator, but if he also has such an adapter, it surely is worth trying :)

closed 2023-12-02 by ahensley