bb-qq/r8152 #358: DS216+II not stable with the latest driver

aqsandy, 2023-11-29:
### Description of the problem

> Replace this line with your answer.
The ethernet drive still shows in the ifconfig but it dropped from the list in `lsusb`. The device is not reachable after couple of hours even without large data transfer. No other usb devices attached to the Nas.

### Description of your products

> - the product name of your NAS model
Linux  3.10.108 #69057 SMP Sat Sep 23 21:44:56 CST 2023 x86_64 GNU/Linux synology_braswell_216+II
>     - and output of ``uname -a`` command
>     - DSM version 7.2
> - the product name of the ethernet dongle
R8156B series Dongle

### Description of your environment

> - connect directly to a PC or using a Ethernet switch
>     - PC: NIC model name or chipset name
Realtek USB 2.5GbE Family Controller  / USB\VID_0BDA&PID_8156&REV_3100

>     - Ethernet switch: model name
2.5GE switch from amazon (https://www.amazon.com/dp/B0C5D2Y4FH?ref=ppx_yo2ov_dt_b_product_details&th=1)
> - cable category (Cat.5e or Cat.6 or Cat.7)

 Cat 6 cable
### Output of ``dmesg`` command
 Attached.
[dmseg.txt](https://github.com/bb-qq/r8152/files/13506613/dmseg.txt)

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (xhci_hcd 0000:00:14.0) hub
  |__1-5         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 65006B70CCCA6000)
|__usb2          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (xhci_hcd 0000:00:14.0) hub
````

### Output of ``ifconfig -a`` command

````
docker0   Link encap:Ethernet  HWaddr 02:42:FA:5C:E9:DE
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          inet6 addr: fe80::42:faff:fe5c:e9de/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:258 (258.0 B)

docker22d Link encap:Ethernet  HWaddr C6:9D:4D:D8:B1:88
          inet6 addr: fe80::c49d:4dff:fed8:b188/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:516 (516.0 B)

eth0      Link encap:Ethernet  HWaddr 00:11:32:70:56:CD
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:22425832 errors:0 dropped:0 overruns:0 frame:0
          TX packets:57060406 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1924711388 (1.7 GiB)  TX bytes:58099931307 (54.1 GiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:129940 errors:0 dropped:0 overruns:0 frame:0
          TX packets:129940 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:16456306 (15.6 MiB)  TX bytes:16456306 (15.6 MiB)

ovs-syste Link encap:Ethernet  HWaddr 62:67:73:AE:B4:BF
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

ovs_eth0  Link encap:Ethernet  HWaddr 00:11:32:70:56:CD
          inet addr:x.x.x.x  Bcast:x.x.x.x  Mask:255.255.255.0
          inet6 addr: x.x.x.x Scope:Global
          inet6 addr: x.x.x.x/64 Scope:Global
          inet6 addr: fe80::211:32ff:fe70:56cd/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:22383761 errors:0 dropped:0 overruns:0 frame:0
          TX packets:46001386 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:1922153320 (1.7 GiB)  TX bytes:57445230931 (53.5 GiB)

ovs_eth1  Link encap:Ethernet  HWaddr C8:4D:44:25:2D:3A
          inet addr:x.x.x.x Bcast:x.x.x.x  Mask:255.255.255.0
          inet6 addr: fe80::ca4d:44ff:fe25:2d3a/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:61009 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6791 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:8416111 (8.0 MiB)  TX bytes:15397326 (14.6 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

bb-qq, 2024-01-06:
```
[ 1093.768942] usb 2-2: reset SuperSpeed USB device number 2 using xhci_hcd
[ 1098.786910] usb 2-2: device descriptor read/8, error -110
[ 1099.068081] usb 2-2: reset SuperSpeed USB device number 2 using xhci_hcd
[ 1099.086394] usb 2-2: Parent hub missing LPM exit latency info.  Power management will be impacted.
[ 1099.097424] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff880273ecdf00
[ 1099.107574] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff880273ecdf40
[ 1099.117773] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff880273ecdf80
```

These logs indicate that the issue was caused by the device itself or the host controller level. I don't know anything from a driver standpoint, try replacing the device or USB cable.

meisiek, 2024-01-06:
The adapter works on my DS216+ only if plugged in the front usb port. It disconnects or throttles down to 1 Gb if plugged in a back usb port.

bb-qq, 2024-02-25:
Also, [Troubleshooting page](https://github.com/bb-qq/r8152/wiki/Troubleshooting) may help.

bb-qq, 2024-04-18:
I close this issue as there have been no replies for some time.
Please feel free to re-open the issue.