bb-qq/r8152 #359: Please provide support for the 1512+ (cedarview) units stuck on DSM 6.2

ThomasMorley-T3D, 2023-12-01:
I'd like to try your updated code on my old 1512+ (cedarview). You older 6.x packages kind of worked but I'm hoping things will be better with your new code.

bb-qq, 2024-01-07:
I added the cedarview driver for DSM 6.2.
https://github.com/bb-qq/r8152/releases/download/2.17.1/r8152-cedarview-2.17.1-1_6.2.spk

ThomasMorley-T3D, 2024-01-07:
Thank you for the quick addition.  The driver appears to be more stable than the old versions. Connection states 2.5G connection. I was hoping for a miracle on this old hardware but sadly, I'm only achieving around 1G transfers through it (I think it's CPU bound as CPU use is pegged at 99% at times during transfers). On a positive note it does give me one more stable network connection to the server. With multi-channel SMB (experimental on this unit and must be turned on by cmd changes) this unit can reach over 2.5G throughput with the 3 connections. Thank you again for your time on this project!

closed 2024-01-07 by ThomasMorley-T3D