bb-qq/r8152 #360: cannot install r8152-apollolake-2.17.1-1_7.1.spk on my  DS918+

github-0815, 2023-12-13:
### Description of the problem

cannot install r8152-apollolake-2.17.1-1_7.1.spk on my  DS918+

As recommended, I tried installation via the Packet Manager (manual installation), chose the packet and uploaded it to the NAS, and started installation via GUI. 

Failure was reported, and, without closing the GUI installation window, I run the command
	sudo install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su /opt/sbin/spk_su
but installation failed and the following message was thrown:
	install: cannot stat '/var/packages/r8152/target/r8152/spk_su': No such file or directory

I did this multiple times, without or with NAS reboot, and manually deleting the r8152 directory:
	$ rmdir /volume1/@appconf/r8152
But installation did not work.

Is there anything wrong with the package, or maybe in the configuration of my user account / settings of the NAS?

### Description of your products

DS918+
Linux synology_nas 4.4.180+ #42962 SMP Mon May 29 14:38:23 CST 2023 x86_64 GNU/Linux synology_apollolake_918+  
DSM 7.1.1-42962 Update 6
DIGITUS DN-3028 USB Type-C™ Gigabit Ethernet Adapter 2.5G, USB-C™ + USB A (USB3.1/3.0)

### Description of your environment

Ethernet dongle connected to front USB port, 
CAT 5E cable connected to the 2G5 switch
PC connected via the same type of USB to Ethernet dongle (Digitus) CAT 5E cable to the switch
PC mainboard: MSI MPG Z390M GAMING EDGE AC (MS-7B50)

8 Port 2.5G Ethernet Switch with 10G SFP Uplink, NICGIGA Unmanaged 2.5Gb Network Switch
Cable category: Cat.5e

### Output of ``dmesg`` command
(see attached file: github_0815_output_of_dmesg_command.txt)
[github_0815_output_of_dmesg_command.txt](https://github.com/bb-qq/r8152/files/13666576/github_0815_output_of_dmesg_command.txt)

github-0815, 2023-12-15:
Anybody out there?

The problem (r8152 driver cannot be installed on DS918+) still persists.

Here some dditional information to my previous post:

Test installation of the aqc111 package did not work either. The problem therefore IMHO is not caused by the package itself, but by the actual NAS configuration / access / account settings and permissions. As a user, I have general r/w permissions.

During installation (following the sequence depicted in my first contribution, using the sudo install command line as requested), the necessary package directories are not created in the /var/packages path, neither for the r8152 package nor for the aqc111 package. 

Any idea? 
Any help would be very much appreciated!

Namssorg, 2023-12-16:
Coincidentally, I am trying to do the same install on a 918+ right now, and am running into this same issue. Install failed (as expected) and after logging in through SSH to run the sudo command, and rebooting, the package is still showing up with an install error.

github-0815, 2023-12-16:
The last comment by Namssorg points to the question if there is a general installation problem with the r8152 builds for the 918+ (I tried installation of the actual build (r8152-apollolake-2.17.1-1_7.1.spk) as well as of the previous build (r8152-apollolake-2.16.3-4_7.1.spk), but both failed.
Or is there something wrong with the actual  918+ DSM 7.1.1-42962 Update 6 regarding manual package installation?

github-0815, 2023-12-16:
Hi Namssorg,
Which is your current DSM version?
My DS918+ runs DSM 7.1.1-42962 Update 6
There is a newer version (DSM 7.2) available, but users of the 7.1.1 version get no notification:
https://www.synology.com/de-de/releaseNote/DSM?model=DS918%2B
If your DSM version is DSM 7.2, then I would try to update first, and then retry installation of the appropriate r8152 package.

Namssorg, 2023-12-16:
I am using DSM 7.1.1-42962 Update 6.

teeesss, 2024-02-14:
I am not sure if this is the same issue, but I tried using the newest package [1.3.3.0-12](https://github.com/bb-qq/aqc111/releases/tag/1.3.3.0-12) for 7.2 and I have a 1019+ and it failed and and running: `sudo install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su /opt/sbin/spk_su` then I tried older versions: [1.3.3.0-5 DSM-7.x](https://github.com/bb-qq/aqc111/releases/tag/1.3.3.0-5) and kept going up and the driver would start and then stop. 
I eventually worked my way back up to [1.3.3.0-12](https://github.com/bb-qq/aqc111/releases/tag/1.3.3.0-12) and the driver finally started. I tried with and without the UGreen 2.5G adapter plugged in, but it worked when it was plugged in.

For sure not easy and I assumed it would never work. I did not reboot.

bb-qq, 2024-06-23:
This is an old issue so I would like to close it.
If using the latest updated package does not solve the problem, feel free to reopen it.

closed 2024-06-23 by bb-qq