bb-qq/r8152 #362: Does not work for DS920+ DSM 7.2.1

SYStem-Max, 2023-12-25:
### Description of the problem

When I enter "sudo install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su /opt/sbin/spk_su" into SSH, I get error: "cannot stat '/var/packages/r8152/target/r8152/spk_su': No such file or directory". Any suggestions?

### Description of your products

>  DS920+
>     - and output of ``uname -a`` command
>     - DSM version 7.2.1
> Ethernet 2.5G KS-IS KS-714

### Description of your environment

> - connect directly to a PC or using a Ethernet switch
>     - PC: NIC model name or chipset name
>     - Ethernet switch: model name
> - cable category (Cat.5e or Cat.6 or Cat.7)

### Output of ``dmesg`` command

````
(paste output here, or attach as a file)
````

### Output of ``lsusb`` command

````
(paste output here)
````

### Output of ``ifconfig -a`` command

````
(paste output here)
````

michaelsargent, 2023-12-29:
Hi SYStem-Max,
Did you download the correct package from https://github.com/bb-qq/r8152/releases
Did you try to install the package that was downloaded using manual install?  You will get a "Failed to install" message.  Close that and then run the command "sudo install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su /opt/sbin/spk_su"

If you haven't installed it with the package manager (and failed), you will get file not found.

bb-qq, 2024-06-15:
@michaelsargent , thanks for the sound advice.
Since there has been no response for a long time, this issue is closed. Please feel free to re-open it if you have any questions.

closed 2024-06-15 by bb-qq