bb-qq/r8152 #366: USB ethernet adapter drops off during the night

coinbirb, 2024-01-07:
### Description of the problem

> Hiya! First of all, thanks for doing this, very cool to have the ability to add faster connectivity to older devices. :)

I have a DS1019+ that I wanted to give some more bandwidth, so I attached a Realtek 8156B based USB adapter. The instructions worked pretty much, and the link comes up and delivers reasonable performance. The problem is that it drops unexpectedly - either after a certain time, or during the night, maybe when some maintenance routine runs or something? 

When that happens the entire network stack seems to be very unhappy, and I can't SSH into the NAS anymore, even if I have one of the Gbit ports connected. The machine still responds to ICMP requests on the other IP, but no SSH, no web UI, which means I have to reboot.

Happy to provide any additional information that may come in handy.

### Description of your products

> Synology DS1019+
> Linux DS1019+ 4.4.302+ #69057 SMP Mon Nov 13 14:21:00 CST 2023 x86_64 GNU/Linux synology_apollolake_1019+
> DSM 7.2.1-69057 Update 3
> Digitus DN-3028 (RTL8156B chipset)

### Description of your environment

> connected to TP-Link BE85
> Cat. 6

### Output of ``dmesg`` command
dmesg doesn't have anything interesting because I have to reboot every time. I've attached kernel logs that might show something interesting.
[jan6.txt](https://github.com/bb-qq/r8152/files/13853683/jan6.txt)

bb-qq, 2024-02-04:
It seems that there is something wrong with the USB host stack or at the signal level.

And since geminilake USB is mature and stable on this platform, it may be a device side issue. Can you try another product? Preferably one with RTL8156BG chip.

Also you might find some insights from the [Troubleshooting page](https://github.com/bb-qq/r8152/wiki/Troubleshooting).

coinbirb, 2024-02-05:
Yeah, I've replaced it with the Club3d one and used this adapter on a different machine, where it's stable. All good, thanks for taking a look!

closed 2024-02-05 by coinbirb