bb-qq/r8152 #368: DS1815+ with DSM7.1.1 and avoton erry by installing the driver

nkmtsm, 2024-01-13:
### Description of the problem
The driver will be installed, tested, all available versions from r8152-avoton-2.15.0-9.spk - r8152-avoton-2.17.1-1_7.1.spk same result installation failure.
From DSm side, pls reinstall the driver again..
Intersting: on my old DS412+ (dsm6.2.4 the driver runns very well with good 2.5Bit performance.

> Replace this line with your answer.
So what to do?
regards Norbert

### Description of your products

DS1815+ with DS7.1.1 and avoton 
Linux nas3 3.10.108 #42962 SMP Mon May 29 14:36:28 CST 2023 x86_64 GNU/Linux synology_avoton_1815+
>     - DSM 7.1.1-42962 Upd 2

### Description of your environment

> - connect directly to a PC or using a Ethernet switch
Tested direct attached on all 4 USB 3.0 port + adding 2 seperated USB3.0 hubs
>     - PC: NIC model name or chipset name
All tested adapters
ULT-WIIQ USB-A 2,5 GB Ethernet-Adapter
plugable 2,5-GB-USB-C Ethernet-Adapter
CLUB 3D CAC-1420
ASUSTOR AS-U2.5G
Tested Cat5e, Cat6, and Cat7 cables

### Output of ``dmesg`` command
[83634.506650] Loading kernel module for a network device with CAP_SYS_MODULE (deprecated).  Use CAP_NET_ADMIN and alias netdev- instead.
....

[102777.649266] Loading kernel module for a network device with CAP_SYS_MODULE (deprecated).  Use CAP_NET_ADMIN and alias netdev- instead.

````
(paste output here, or attach as a file)
````

### Output of ``lsusb`` command
ash-4.4# lsusb
|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:16.0) hub
  |__1-1         8087:07db:0002 09  2.00  480MBit/s 0mA 1IF  ( ffffffd1ffffffb2ffffffdbffffffad) hub
    |__1-1.1     f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 65004CD0F0A28F78)
|__usb2          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__2-4         2109:2822:9014 09  2.10  480MBit/s 0mA 1IF  (VIA Labs, Inc.          USB2.0 Hub              000000001) hub
|__usb3          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__3-4         2109:0822:9014 09  3.20 5000MBit/s 0mA 1IF  (VIA Labs, Inc.          USB3.1 Hub              000000000) hub
ash-4.4# lsusb
|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:16.0) hub
  |__1-1         8087:07db:0002 09  2.00  480MBit/s 0mA 1IF  ( ffffffd1ffffffb2ffffffdbffffffad) hub
    |__1-1.1     f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 65004CD0F0A28F78)
|__usb2          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__2-4         2109:2822:9014 09  2.10  480MBit/s 0mA 1IF  (VIA Labs, Inc.          USB2.0 Hub              000000001) hub
|__usb3          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__3-4         2109:0822:9014 09  3.20 5000MBit/s 0mA 1IF  (VIA Labs, Inc.          USB3.1 Hub              000000000) hub

````
(paste output here)
````

### Output of ``ifconfig -a`` command bas system has 4 x 1Gbit connections
bond0     Link encap:Ethernet  HWaddr 00:11:32:40:01:09
          inet addr:192.168.178.43  Bcast:192.168.178.255  Mask:255.255.255.0
          inet6 addr: fe80::211:32ff:fe40:109/64 Scope:Link
          UP BROADCAST RUNNING MASTER MULTICAST  MTU:1500  Metric:1
          RX packets:39919790 errors:0 dropped:0 overruns:0 frame:0
          TX packets:17193425 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:55348762005 (51.5 GiB)  TX bytes:2657595255 (2.4 GiB)

docker0   Link encap:Ethernet  HWaddr 02:42:3C:4E:1E:5C
          inet addr:172.18.0.1  Bcast:172.18.255.255  Mask:255.255.0.0
          inet6 addr: fe80::42:3cff:fe4e:1e5c/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:9985 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:619142 (604.6 KiB)

docker60d Link encap:Ethernet  HWaddr 86:44:92:1E:5D:4A
          inet6 addr: fe80::8444:92ff:fe1e:5d4a/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:9988 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:619400 (604.8 KiB)

eth0      Link encap:Ethernet  HWaddr 00:11:32:40:01:09
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:38475224 errors:0 dropped:0 overruns:0 frame:0
          TX packets:18729 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:55102416506 (51.3 GiB)  TX bytes:1674106 (1.5 MiB)

eth1      Link encap:Ethernet  HWaddr 00:11:32:40:01:09
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:1444566 errors:0 dropped:0 overruns:0 frame:0
          TX packets:17174696 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:246345499 (234.9 MiB)  TX bytes:2655921149 (2.4 GiB)

eth2      Link encap:Ethernet  HWaddr 00:11:32:40:01:0B
          inet addr:172.17.43.43  Bcast:172.17.43.255  Mask:255.255.255.0
          inet6 addr: fe80::211:32ff:fe40:10b/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2000923 errors:0 dropped:0 overruns:0 frame:0
          TX packets:19576 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:185159660 (176.5 MiB)  TX bytes:1629242 (1.5 MiB)

eth3      Link encap:Ethernet  HWaddr 00:11:32:40:01:0C
          inet addr:10.0.0.43  Bcast:10.0.0.255  Mask:255.255.255.0
          inet6 addr: fe80::211:32ff:fe40:10c/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2010216 errors:0 dropped:0 overruns:0 frame:0
          TX packets:18919 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:188919333 (180.1 MiB)  TX bytes:1535835 (1.4 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:1111419 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1111419 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:390138192 (372.0 MiB)  TX bytes:390138192 (372.0 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

ash-4.4#

````
(paste output here)
````

nkmtsm, 2024-01-17:
Hello together no info no help no thing?

jpman67, 2024-01-22:
I have a DS1815+ with DSM7.1.1 as well. Exact same issue.

bb-qq, 2024-02-25:
Could you give me the entire `dmesg` log?

And did you run `lsusb` when connecting the device? 
The output of `lsusb` does not include any Realtek devices which should be shown whether my driver works or not.

nkmtsm, 2024-03-18:
ash-4.4# lsusb
|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:16.0) hub
  |__1-1         8087:07db:0002 09  2.00  480MBit/s 0mA 1IF  ( ffffffd1ffffffb2ffffffdbffffffad) hub
    |__1-1.1     f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 65004CD0F0A28F78)
|__usb2          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__2-4         2109:2822:9014 09  2.10  480MBit/s 0mA 1IF  (VIA Labs, Inc.          USB2.0 Hub              000000001) hub
    |__2-4.4     0bda:8156:3100 00  2.10  480MBit/s 200mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 001000001)
|__usb3          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__3-4         2109:0822:9014 09  3.20 5000MBit/s 0mA 1IF  (VIA Labs, Inc.          USB3.1 Hub              000000000) hub
    |__3-4.1     0bda:8156:3100 00  3.20 5000MBit/s  0IFs (Realtek USB 10/100/1G/2.5G LAN 001000000)
    |__3-4.3     0bda:8156:3104 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 4013000001)
ash-4.4# ifconfig eth5
eth5      Link encap:Ethernet  HWaddr 00:E0:4C:F1:77:61
          inet addr:192.168.179.40  Bcast:192.168.179.255  Mask:255.255.255.0
          inet6 addr: fe80::2e0:4cff:fef1:7761/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1983 errors:0 dropped:0 overruns:0 frame:0
          TX packets:61 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:181397 (177.1 KiB)  TX bytes:23549 (22.9 KiB)

ash-4.4# /var/packages/r8152/scripts/start-stop-status stop
++ dirname /var/packages/r8152/scripts/start-stop-status
+ script_root=/var/packages/r8152/scripts
++ realpath -m /var/packages/r8152/scripts/..
+ package_root=/var/packages/r8152
++ basename /var/packages/r8152
+ driver_name=r8152
+ driver_root=/var/packages/r8152/target/r8152
+ log_root=/var/packages/r8152/var/log
+ mkdir -p /var/packages/r8152/var/log
+ exec
++ tee -a /var/packages/r8152/var/log/start-stop-status.log
+ echo ----------
----------
+ date
Mon Mar 18 18:30:13 CET 2024
+ grep -E '^(version|create_time)' /var/packages/r8152/INFO
version="2.16.3-4"
create_time="20230319-14:37:13"
+ echo stop
stop
+ case $1 in
+ set_interfaces down
+ action=down
+ target_interface_name=
++ ls /sys/class/net
+ all_interface_names='bond0
bonding_masters
docker0
dockerb4521c9
eth0
eth1
eth2
eth3
eth4
eth5
lo
sit0'
+ '[' -n ']'
+ '[' '!' -e /sys/class/net/ ']'
+ for interface_name in ${all_interface_names}
+ [[ -n '' ]]
+ '[' '!' -e /sys/class/net/bond0/device/driver ']'
+ continue
+ for interface_name in ${all_interface_names}
+ [[ -n '' ]]
+ '[' '!' -e /sys/class/net/bonding_masters/device/driver ']'
+ continue
+ for interface_name in ${all_interface_names}
[... 110 lines cut ...]
+ mkdir -p /var/packages/r8152/var/log
+ exec
++ tee -a /var/packages/r8152/var/log/start-stop-status.log
+ echo ----------
----------
+ date
Mon Mar 18 18:32:11 CET 2024
+ grep -E '^(version|create_time)' /var/packages/r8152/INFO
version="2.16.3-4"
create_time="20230319-14:37:13"
+ echo start
start
+ case $1 in
+ '[' -e /var/packages/r8152/scripts/apply-memory-setting ']'
+ '[' -w /sys/module/usbcore/parameters/autosuspend ']'
++ count_usb_interfaces
++ ls -l /sys/class/net/
++ grep /usb
++ wc -l
+ initial_count=0
+ '[' -e /var/packages/r8152/target/r8152/mii.ko ']'
+ '[' -e /var/packages/r8152/target/r8152/usbnet.ko ']'
+ /sbin/insmod /var/packages/r8152/target/r8152/r8152.ko
+ '[' -r /usr/lib/udev/rules.d/51-usb-r8152-net.rules ']'
+ exit 0

nkmtsm, 2024-03-18:
[dmesg.zip](https://github.com/bb-qq/r8152/files/14640343/dmesg.zip)

nkmtsm, 2024-03-18:
If you want to see it direcly, you can here login over teamviewer session too.

closed 2024-03-18 by nkmtsm

nkmtsm, 2024-04-01:
> @nkmtsm <https://github.com/nkmtsm> , did you connect multiple Realtek
> NICs at the same time?
> If so, could you try to connect one device?
> And, could you also try to enable the hot-plug feature described on this
> page
> ?

Hi, a interesting Thing Happens. I Update MS ds1815+ with 7.1 ti ds1817+ to
7
2.1 and your diver runns very Well. Just now i try Out the 20 diffent
Models of the USB 2.5 nics. Results will sending. Regards norbert