bb-qq/r8152 #369: Problem with 2,5 Gb speed on DS214Play

devosdan, 2024-01-15:
### Description of the problem

>Problem with 2,5 Gb speed on DS214Play

I have my NAS connected with a Ugreen adapter (New with realtek RTL8156BG) through a 2,5 Gb 8 port switch to my PC with 2,5 Gb network connector.
But checking the speed with Iperf3 I get 80 MB as a maximum.
- Checked the Ugreen adapter on another PC and there it has a speed of 275 MB (Amazing), so the adapter is ok.
- Changing to the other USB 3.0 connections (front and rear), still have 80 MB.

- Have another NAS (Asustor AS3202T) with a Asustor 2,5 Gb adapter on my PC and the speed is up to 280 MB.MTU settings on it = 9000 as on my PC.

- Manualy I set the MTU of my DS214Play to MTU 9000,but checking with the command ip a, I see that it did not change to 9000, still is on MTU 1500.
- Setting the MTU on the PC to 1500, the speed drops to 40 MB . 

My question: Is there a possibility to force the DS214Play (with a script) to accept the MTU 9000, because I am very sure that this is the problem. 

By the way : I followed all the instructions as mentioned on github. Driver r8152-2.17.1 for the 7.1 DSM and Evansport cpu properly installed and confirmed by my NAS.

````

````

devosdan, 2024-01-21:
After a few days of testing my son finaly found the solution. He checked all the existing r8152 drivers on GitHub and told me to install the following version 2.15.0-4 DSM7.0. And tada I got my speed with Iperf3 tested to 140 MB/s instead of 80 MB/s Also I checked the MTU setting in my nas and was confirmed MTU 9000. So I am a happy guy.
So the Ugreen adapter (New with realtek RTL8156BG) is a perfect match when using this older driver that is aproved for DSM 7.0. while I have DSM 7.1 installed.

Sjekke, 2024-01-22:
@bb-qq how is this possible?

bb-qq, 2024-02-26:
I released [the version 2.17.1-2](https://github.com/bb-qq/r8152/releases/tag/2.17.1-2) which is almost same configuration as  2.15.0-4. Could you try it?

devosdan, 2024-02-26:
Well, I just installed the version 2.17.1-2 and can confirm that the speed is up and around 140MB/s and the settings of MTU 9000 in my nas (networkinterface) is ok. So many thanks for adapting the old driver version.

closed 2024-06-15 by bb-qq