bb-qq/r8152 #370: DS920+ and UGREEN adapter works but capped at ~40mb/s

333bdm, 2024-01-21:
### Description of the problem

> Replace this line with your answer.

### Description of your products

> - the product name of your NAS model
>     - and output of ``uname -a`` command
>     - DSM version x.x
> - the product name of the ethernet dongle

### Description of your environment

> - connect directly to a PC or using a Ethernet switch
>     - PC: NIC model name or chipset name
>     - Ethernet switch: model name
> - cable category (Cat.5e or Cat.6 or Cat.7)

### Output of ``dmesg`` command

````
(paste output here, or attach as a file)
````

### Output of ``lsusb`` command

````
(paste output here)
````

### Output of ``ifconfig -a`` command

````
(paste output here)
````

edw00d4711, 2024-01-26:
same problem

Linux version 4.4.302+ (root@build7) (gcc version 12.2.0 (GCC) )  CST 2024
Brand: Synology
Model: DS-918+

**parts of dmesg**
 _r8152: module verification failed: signature and/or required key missing - tainting kernel
usbcore: registered new interface driver r8152
usb 2-1: reset SuperSpeed USB device number 2 using xhci_hcd
netif_napi_add() called with weight 256 on device eth%d
r8152 2-1:1.0 eth2: v2.17.1 (2023/06/13)
r8152 2-1:1.0 eth2: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
r8152 2-1:1.0 eth2: chip rev 14
r8152 2-1:1.0 eth2: rx_buf_sz will be overridden to 16384
IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready
r8152 2-1:1.0 eth2: carrier on_

frkaar, 2024-02-05:
I have the same issue!
Previously the speed was 2500Mbps but I noticed a slowdown in file transfert => 40mo/s (previously >200mo/s), then I check the network section on my DS918+ and I realized that the speed was listed as 1000Mbps.
I don't why the adapter's characteristics changed :-/

bb-qq, 2024-02-25:
Did you check the Troubleshooting page?
https://github.com/bb-qq/r8152/wiki/Troubleshooting#performance-issues

ossource, 2024-03-04:
I had this issue, too. It is a comparability issue between Host USB and dongle USB chip.
You can add another USB host chip in row (aka USB hub with 3.0 speed ;) )  You can check the USB speed with 'lsusb' over command line.

ydzhus, 2024-03-15:
A similar problem.

Scheme:
NAS DS720+ USB 3.0 (lsusb) --> Ugreen 5 adapter (USB 3.0 to Type C - 5Gbps) --> USB-C 2500Mbps (RTL8156BG) --> Vention CAB CAT 7 (0.5v) --> Keenetic kn-1811 (2.5g port) --> wifi ax 2400 notebook 0.2 m.

50 megabytes per second for downloading from the NAS, 170-180 megabytes per second for writing to the NAS.

Changing cables and adapter to Elough 10А did not help.
I expect a 3.1 - 10Gbps adapter, then we'll see.

I don't know where to look anymore

csihama, 2024-03-24:
Similar issue happened with me when using a Club3d CAC-1420 3.2 Gen1 Type-A 2.5 Gbit Ethernet adapter (RTL-8156) and B75M-D3H motherboard. I was only getting around 40 MB/s ~ 342 Mbps. When plugging the 2.5 Gbit Ethernet adapter into one of the front USB ports, speed is now at 2.37 Gbit/s and issue seems resolved.

The front panel uses another USB xHCI Host controller that is actually capable of 5 Gbps USB... The rear ones are 3.0 and can only have 480 Mbit/s max... hopefully someone can learn from this experience.

Sjekke, 2024-03-24:
Thanks for the feedback, but this was mentioned many times before here. Nevertheless thkx.

lschmies, 2024-03-28:
I had a similar problem. My Ugreen 2.5gbit Lan adapter is USB-C. Therefore you need a USB-C to USB-A converter.
The direction in which I inserted the Lan adapter into the USB-C to USB-A converter decides for me whether there is a USB 2 or USB 3 connection!
There are cheap adapters that are not wired correctly!

ydzhus, 2024-03-29:
What you say is correct.
In my case, writing to the NAS is always stable at 170 megabytes per second. But reading from the NAS is about 150 megabytes per second only on the first connection, then it drops to 70-80 megabytes per second and does not rise again. It feels like the adapter is being protected.

I tried 3 USB 3.0\3.1 to Type C adapters
1. USB 3.0 (lsusb)
2. Ugreen (USB 3.0 to Type C - 5Gbps)
3. Baseus (USB 3.1 to Type C - 10Gbps)

The situation is the same.
But there is a nuance, I connect via Wi-Fi.
In a week the SSD case will arrive and I will check everything using the 7e-8e cable with the above adapters. Then I’ll write back.
There is a possibility that the laptop's Wi-Fi card may not be stable for reception, although the SSD drive in the laptop is fast.

It’s strange that for recording everything is stable at 170 megabytes per second, this is the limit of my HDD drives, but for downloading only 70-80 megabytes per second.

It's all somehow strange.

bb-qq, 2024-04-01:
Did you measure throughput by iperf?
The result of iperf may indicate additional insights.

ydzhus, 2024-05-28:
[image]

????

bb-qq, 2024-06-15:
Hi all, the cause varies regarding performance issues, so could you create a separate issue and fill the template?
This issue continues to handle the issue on the first comment and I am awaiting the result of iperf.

ydzhus, 2024-06-15:
> Did you measure throughput by iperf? The result of iperf may indicate additional insights.

In my case, when I raised the virtual machine and assigned the adapter as USB, it turned off, started making a sound and started showing 100 megabytes of connection. No matter where I put it, it’s always 100 megabits.
So, I sold it. Since it heated up at 2500 megabits until it crashed .

ydzhus, 2024-06-15:
[image]
[image]
[image]
[image]

bb-qq, 2025-01-02:
> This issue continues to handle the issue on the first comment and I am awaiting the result of iperf.

Since there has been no response for a long time, this issue is closed. Please feel free to re-open it if you have any questions.

closed 2025-01-02 by bb-qq