bb-qq/r8152 #374: DS115J 2.5gbit/s possibility

GIAGIA35, 2024-02-18:
### Description of the problem

My Synology NAS is a DS115j with a Marvell Armada 370 88F6707 CPU in it. I tried to patch it with the r8152-armada37xx-2.16.3-4_7.1 spk file but after loading it, I have an error message that says "this package is not supported but the diskstation platform or is incompatible with the DSm version." Am I using the wrong spk file or is it not just possible to upgrade a DS115J to 2,5gbit/s speed LAN network ?

### Description of your products

> DS115J
> DSM version 7.1.1

bb-qq, 2024-02-25:
I created a package for armada370. Could you try it?
https://github.com/bb-qq/r8152/releases/download/2.17.1/r8152-armada370-2.17.1-1_7.1.spk

GIAGIA35, 2024-02-28:
It works fine! thank you. Howerver i've ust noticed the DS115j is only equiped with 2.0 usb port. Is it a dead end ?

ossource, 2024-03-04:
obviously, yes.

closed 2024-04-17 by bb-qq