bb-qq/r8152 #376: DS-420+ & Ugreen Adapter 2.5: No automatic start on boot ?

KorT3XXX, 2024-02-23:
Hello,

My UGREEN adapter is working fine with the front usb and I can get speed transfers up to 240Mo/s which is great.

But I need to start manually the driver after NAS reboot in package center otherwise the interface wont come up.
Btw, if I change the usb port, the interface is gone and will come up after I restart the driver in the package center.

What should I do to automatically start the driver on boot.

DSM 7.2.1-69057 Update 3

Thank you

bb-qq, 2024-02-24:
Maybe there is something different in the boot sequence of your NAS than usual, since once you manually start the package, the driver should be loaded automatically after the next boot.

I don't know if this will work, but could you try enabling hotplugging on this page?
https://github.com/bb-qq/r8152/wiki/Troubleshooting#hot-plugging-does-not-work

bb-qq, 2024-04-17:
I close this issue as there have been no replies for some time.
Please feel free to re-open the issue.

closed 2024-04-17 by bb-qq