bb-qq/r8152 #377: Can’t installe r8152 on DS418 DSM7.2.1

Toniohec, 2024-02-23:
Hi, I can’t install the driver r8152 through SSH once I manually “failed” to install it in DSM.

I have a DS418 (rtd1296), I downloaded the suitable driver for it ( https://github.com/bb-qq/r8152/releases/download/2.17.1/r8152-rtd1296-2.17.1-1_7.2.spk) but when I type the command in SSH “ sudo install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su /opt/sbin/spk_su” it asks for password, I type my password then enter and nothing happens, no line appears. I didn’t buy any adapter yet, wanted to be sure to be able to install driver first.

### Output of ``dmesg``

### Output of ``lsusb`` command

|__usb2          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller xhci-hcd.2.auto) hub
|__usb3          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller xhci-hcd.2.auto) hub
|__usb4          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller xhci-hcd.5.auto) hub
|__usb5          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller xhci-hcd.5.auto) hub

### Output of ``ifconfig -a`` command
bond0     Link encap:Ethernet  HWaddr 00:11:32:81:A5:88  
          inet addr:192.168.1.61  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: 2a01:e0a:8e5:8a70:211:32ff:fe81:a588/64 Scope:Global
          inet6 addr: fe80::211:32ff:fe81:a588/64 Scope:Link
          UP BROADCAST RUNNING MASTER MULTICAST  MTU:1500  Metric:1
          RX packets:10164 errors:0 dropped:0 overruns:0 frame:0
          TX packets:12023 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:2198856 (2.0 MiB)  TX bytes:5982903 (5.7 MiB)

eth0      Link encap:Ethernet  HWaddr 00:11:32:81:A5:88  
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:8462 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5644 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:2059489 (1.9 MiB)  TX bytes:3649112 (3.4 MiB)
          Interrupt:11 

eth1      Link encap:Ethernet  HWaddr 00:11:32:81:A5:89  
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:1702 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6379 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:139367 (136.1 KiB)  TX bytes:2333791 (2.2 MiB)
          Interrupt:28 base 0x4000  

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:8979 errors:0 dropped:0 overruns:0 frame:0
          TX packets:8979 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:2131676 (2.0 MiB)  TX bytes:2131676 (2.0 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

Thank you

Toniohec, 2024-02-23:
Toniohec@DS418_Tonio:~$ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.302+ (root@build8) (gcc version 12.2.0 (GCC) ) #69057 SMP Fri Jan 12 17:01:55 CST 2024
[    0.000000] Boot CPU: AArch64 Processor [410fd034]
[    0.000000] cma: Reserved 32 MiB at 0x000000007e000000
[    0.000000] On node 0 totalpages: 524288
[    0.000000]   DMA zone: 8192 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 524288 pages, LIFO batch:31
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv0.1 detected in firmware.
[    0.000000] psci: Conflicting PSCI version detected.
[    0.000000] PERCPU: Embedded 15 pages/cpu @ffffffc07df78000 s20504 r8192 d32744 u61440
[    0.000000] pcpu-alloc: s20504 r8192 d32744 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: enabling workaround for ARM erratum 845719
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 516096
[    0.000000] Kernel command line: ip=off console=ttyS0,115200 root=/dev/md0 rw syno_hdd_detect=4,5,6,7 syno_hdd_enable=85,86,82,81 HddHotplug=4 syno_usb_vbus_gpio=23@xhci-hcd.2.auto@0,22@xhci-hcd.5.auto@0 syno_hw_version=DS418 hd_power_on_seq=4 ihd_num=4 netif_num=2 audio_version=1012363 syno_fw_version=M.209 U-boot=2012.07-g68fd7ba (Nov 20 2018 - 11:17:34)
[    0.000000] SYNO GPIO hdd detect pin: 4 5 6 7 
[    0.000000] SYNO GPIO hdd enable pin: 85 86 82 81 
[    0.000000] Support HDD Hotplug.
[    0.000000] USB Vbus GPIO Control:
[    0.000000]  Gpp#23 - Host: xhci-hcd.2.auto      - Port:1 - Polarity: ACTIVE_HIGH
[    0.000000]  Gpp#22 - Host: xhci-hcd.5.auto      - Port:1 - Polarity: ACTIVE_HIGH
[    0.000000] Synology Hardware Version: DS418
[    0.000000] Internal HD num: 4
[    0.000000] Internal netif num: 2
[    0.000000] PID hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] software IO TLB: mapped [mem 0x77a00000-0x7ba00000] (64MB)
[    0.000000] Memory: 1871920K/2097152K available (5139K kernel code, 305K rwdata, 1808K rodata, 256K init, 445K bss, 192464K reserved, 32768K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vmalloc : 0xffffff8000000000 - 0xffffffbdbfff0000   (   246 GB)
                   vmemmap : 0xffffffbdc0000000 - 0xffffffbfc0000000   (     8 GB maximum)
                             0xffffffbdc0000000 - 0xffffffbdc2000000   (    32 MB actual)
                   fixed   : 0xffffffbffa7fd000 - 0xffffffbffac00000   (  4108 KB)
                   PCI I/O : 0xffffffbffae00000 - 0xffffffbffbe00000   (    16 MB)
                   modules : 0xffffffbffc000000 - 0xffffffc000000000   (    64 MB)
                   memory  : 0xffffffc000000000 - 0xffffffc080000000   (  2048 MB)
                     .init : 0xffffffc00094a000 - 0xffffffc00098a000   (   256 KB)
                     .text : 0xffffffc000280000 - 0xffffffc0009495f4   (  6950 KB)
                     .data : 0xffffffc000998000 - 0xffffffc0009e46c0   (   306 KB)
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  CONFIG_RCU_FANOUT set to non-default value of 32
[    0.000000] NR_IRQS:64 nr_irqs:64 0
[    0.000000] Architected cp15 timer(s) running at 27.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x63a1e71a3, max_idle_ns: 440795203123 ns
[    0.000005] sched_clock: 56 bits at 27MHz, resolution 37ns, wraps every 4398046511093ns
[    0.000077] [RTK-TIMER0] set mode: CLOCK_EVT_MODE_SHUTDOWN
[    0.000089] [RTK-TIMER0] clocksource 0 register HZ
[    0.000098] clocksource: rtk_timer0_counter: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 70787423951 ns
[    0.000146] [RTK-TIMER0] set mode: CLOCK_EVT_MODE_SHUTDOWN
[    0.000155] [RTK-TIMER1] set mode: CLOCK_EVT_MODE_SHUTDOWN
[    0.000164] [RTK-TIMER1] clocksource 1 register HZ
[    0.000173] clocksource: rtk_timer1_counter: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 579169832313 ns
[    0.000439] [RTK_MEM_REMAP] initialized
[    0.000705] Calibrating delay loop (skipped), value calculated using timer frequency.. 54.00 BogoMIPS (lpj=270000)
[    0.000719] pid_max: default: 32768 minimum: 301
[    0.000867] Security Framework initialized
[    0.000893] AppArmor: AppArmor initialized
[    0.000960] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000971] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes)
[    0.001674] Initializing cgroup subsys io
[    0.001691] Initializing cgroup subsys memory
[    0.001732] Initializing cgroup subsys devices
[... 861 lines cut ...]
[  115.402169] NFSD: starting 90-second grace period (net ffffffc0009d9980)
[  115.532228] ata2.00: disabling queued TRIM support
[  115.533497] ata2.00: disabling queued TRIM support
[  115.533515] ata2.00: configured for UDMA/133
[  115.537903] ata2: EH complete
[  116.207311] ata3.00: configured for UDMA/133
[  116.211730] ata3: EH complete
[  117.029282] ata4.00: configured for UDMA/133
[  117.033714] ata4: EH complete
[... cut ...]
[  121.529053] iSCSI:target_core_ep.c:795:ep_attach_hba RODSP plugin for epio is enabled.
[  121.537208] iSCSI:target_core_ep.c:802:ep_attach_hba ODX Token Manager is enabled.
[  121.900100] workqueue: max_active 1024 requested for vhost_scsi is out of range, clamping between 1 and 512
[  124.058217] findhostd uses obsolete (PF_INET,SOCK_PACKET)
[  139.481149] fuse init (API version 7.23)
[  141.231595] Synotify use 16384 event queue size
[  141.236537] Synotify use 16384 event queue size
[  144.718661] Synotify use 16384 event queue size

bb-qq, 2024-02-24:
> I type my password then enter and nothing happens, no line appears.

This is the expected result.

Toniohec, 2024-02-24:
I thought I read somewhere it would reinstall somehow through SSH.
I still have the error when I look up for the package in DSM.
Still ok ?
Thanks

bb-qq, 2024-02-25:
After installing `spk_su` using SSH, did you manually install the driver package again?

Toniohec, 2024-02-25:
No I didn’t, now I did.. I feel dumb… It seems to be working now I installed it again manually.
But I read some ppl saying they can’t get it work properly with their DS418. The can go PC to NAS but NAS to PC the connection shuts down. Any USB adapter you recommend ?
Thank you for everything

bb-qq, 2024-02-26:
> they can’t get it work properly with their DS418.

Oh, I forgot that. Currently, models using rtd1296 seem to have a critical issue to work. You may encounter problems with DS418...

mvoong, 2024-02-28:
Used StarTech.com US2GA30 adaptor and just got it working on my DS418 with the r8152-rtd1296-2.17.1-1_7.2 release. A previous version only worked at 1 Gbps.

Toniohec, 2024-02-28:
>  just got it working on my DS418
You didn't get any connection shutting down ?

mvoong, 2024-02-28:
Seems fine for me. Using front port

mvoong, 2024-02-29:
2.17.1-1 seems more stable

Toniohec, 2024-02-29:
Thanks for your tests @mvoong, when you say

> 2.17.1-1 seems more stable

are you refering to _7.1 ? more stable is not full stable I presume ?

mvoong, 2024-03-01:
No difference with powered USB hub. iperf3 works fine but as soon as I try to copy a file using SMB, the connection goes down. At this point I may just upgrade to a DS923+

wchow77, 2024-03-25:
> > they can’t get it work properly with their DS418.
> 
> Oh, I forgot that. Currently, models using rtd1296 seem to have a critical issue to work. You may encounter problems with DS418...

Confirmed with 2.17.1-2 on a DS-220j with a rtd1296 and generic RTL8156BG USB-A (Aliexpress). 

It installs and gets an IP address.  I get the login screen from a browser and I can view file shares in Windows but the minute I try to login or transfer files it freezes and stops working.  After that it fails to get an IP address on subsequent reboots.  At first I thought it was because it was connected to a multi-gig port on the ASUS RT-AX89X router but plugged it into a 2.5GbE port on a LIANGUO 8 Port 2.5GbE Ethernet Switch. It got an IP address but once I tried to transfer files it froze up and failed again.

ziv01, 2024-04-05:
> > > they can’t get it work properly with their DS418.
> > 
> > 
> > Oh, I forgot that. Currently, models using rtd1296 seem to have a critical issue to work. You may encounter problems with DS418...
> 
> Confirmed with 2.17.1-2 on a DS-220j with a rtd1296 and generic RTL8156BG USB-A (Aliexpress).
> 
> It installs and gets an IP address. I get the login screen from a browser and I can view file shares in Windows but the minute I try to login or transfer files it freezes and stops working. After that it fails to get an IP address on subsequent reboots. At first I thought it was because it was connected to a multi-gig port on the ASUS RT-AX89X router but plugged it into a 2.5GbE port on a LIANGUO 8 Port 2.5GbE Ethernet Switch. It got an IP address but once I tried to transfer files it froze up and failed again.
> 
> I actually don't use any of Synology's advanced features so I'm replacing it with the new Ugreen NAS with the latest hardware. At least the USB adapter works on Windows.

I am trying to install an TRL8156 USB-a card. Driver installed correctly, when I connect the USB NIC it doesn't show up and if I try to restrt the driver/service it throws a error. It will let me stop it and won't let start it