bb-qq/r8152 #380: DS214play & DS218play not quite working random disconnect & limited speed

lawleo, 2024-02-26:
[image]
2.17.2 is a bit better than 2.17.1. but still have problem on both machine
DS218play is unusable
DS214play iperf3 result is good, but still only 80-90MB/s during file copy on 4TB WD raid 1 configuration

bb-qq, 2024-02-26:
> DS214play iperf3 result is good, but still only 80-90MB/s during file copy on 4TB WD raid 1 configuration

Disk transfer rate or CPU may be the bottleneck. In the latter case, enabling SMB v3 may improve the situation.

> Also, for driver 2.17.1, I can only set the MTU to 8000 instead of maximum 9000 from the DSM GUI, not sure why

This is due to the change in receive buffer size; 2.17.1 has a receive buffer of 8k, and the MTU is also limited to this value.

lawleo, 2024-02-28:
> Disk transfer rate or CPU may be the bottleneck. In the latter case, enabling SMB v3 may improve the situation.

I believe it should be CPU bottleneck because either WD 4TB run on my PC get ~200MB/s.  SMBv3 is enabled by default in Synology NAS, but said to say if I change the MTU other than 1518 on my windows 11 PC with 2.5G network, the mapped network drive will just freeze.  May be the situation will be better if the driver work in DS218play

guerravic, 2024-03-08:
I confirm that the Realtek RTD1296 cpu disconnects the network at the beginning of the file copy.

bb-qq, 2024-06-15:
Regading DS214play, the latest drivers have changes that affect performance, so please try it.
Regading DS218play, there are known stability issues. See #275 for more information.

closed 2024-06-15 by bb-qq