bb-qq/r8152 #381: 802.3ad LACP bond missing after reboot

phiten, 2024-02-27:
### Description of the problem

I have two RTL8156B on my DS918+. Both get detected and are working. I can create a 802.3ad LACP bond which also creates and works fine.
[image]
"Bond 1" consists of the two 1G internal NICs
"Bond 2" consists of the two 2.5G RTL8156B external NICs
Both bonds work and can be accessed. I can confirm that in my switch:
[image]
ch2 is "Bond 1"
ch3 is "Bond 2"

But after a reboot the bond is missing and both RTL8156B show as individual NIC again:
[image]
The bond consisting of the two internal NICs is fine.

I took the requested commands with a working bond:

### Description of your products

> - DS918+
>     - Linux hostname 4.4.302+ #69057 SMP Fri Jan 12 17:02:59 CST 2024 x86_64 GNU/Linux synology_apollolake_918+
>     - DSM version DSM 7.2.1-69057 Update 4
> - Aliexpress RTL8156B USB3.0 Ethernet Adapter

### Description of your environment

> - connected to a switch
>     - PC: irrelevant
>     - Ethernet switch: Lancom GS-4530XUP
> - cable category: Cat.6/a

### Output of ``dmesg`` command

````
[ds918+_dmesg.txt](https://github.com/bb-qq/r8152/files/14425985/ds918%2B_dmesg.txt)
````

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 650078EBA7E41684)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         0bda:8156:3104 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 4013000001)
  |__2-2         0bda:8156:3100 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 0013000001)

````

### Output of ``ifconfig -a`` command

````
bond0     Link encap:Ethernet  HWaddr 00:11:32:7E:EB:D3
          inet addr:10.100.10.140  Bcast:10.100.10.255  Mask:255.255.255.0
          inet6 addr: fe80::211:32ff:fe7e:ebd3/64 Scope:Link
          UP BROADCAST RUNNING MASTER MULTICAST  MTU:1500  Metric:1
          RX packets:4849 errors:0 dropped:0 overruns:0 frame:0
          TX packets:9766 errors:0 dropped:1 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:765808 (747.8 KiB)  TX bytes:12471527 (11.8 MiB)

bond1     Link encap:Ethernet  HWaddr 00:E0:4C:68:02:B3
          inet addr:10.10.10.26  Bcast:10.10.10.255  Mask:255.255.255.0
          inet6 addr: fe80::2e0:4cff:fe68:2b3/64 Scope:Link
          inet6 addr: 2a04::/64 Scope:Global
          UP BROADCAST RUNNING MASTER MULTICAST  MTU:1500  Metric:1
          RX packets:6019 errors:0 dropped:0 overruns:0 frame:0
          TX packets:14990 errors:0 dropped:32 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:877196 (856.6 KiB)  TX bytes:18375158 (17.5 MiB)

eth0      Link encap:Ethernet  HWaddr 00:11:32:7E:EB:D3
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:1368 errors:0 dropped:0 overruns:0 frame:0
          TX packets:9631 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:167372 (163.4 KiB)  TX bytes:12425679 (11.8 MiB)

eth1      Link encap:Ethernet  HWaddr 00:11:32:7E:EB:D3
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:3481 errors:0 dropped:0 overruns:0 frame:0
          TX packets:135 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:598436 (584.4 KiB)  TX bytes:45848 (44.7 KiB)

eth2      Link encap:Ethernet  HWaddr 00:E0:4C:68:02:B3
          inet addr:169.254.178.11  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:1049 errors:0 dropped:0 overruns:0 frame:0
          TX packets:15009 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:110623 (108.0 KiB)  TX bytes:18385274 (17.5 MiB)

eth3      Link encap:Ethernet  HWaddr 00:E0:4C:68:02:B3
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:4970 errors:0 dropped:0 overruns:0 frame:0
          TX packets:115 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:766573 (748.6 KiB)  TX bytes:15133 (14.7 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:9266 errors:0 dropped:0 overruns:0 frame:0
          TX packets:9266 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:1313956 (1.2 MiB)  TX bytes:1313956 (1.2 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

````

FlorianHeigl, 2024-03-09:
I had the same issues with a bond (1x 1g internal + 1x 2.5g external). It'll work, but come apart on reboot.
I suspect they initialize their bonding way before USB devices are fully loaded.

bb-qq, 2024-04-17:
Under the current mechanism, drivers are loaded after the package system is ready, so LACP BOND, which is prepared at an earlier stage, is not restored at startup.

If the drivers are loaded at the same time as other kernel modules are loaded at startup, the LACP configuration can be enabled. But I have not found a good way to do that.

ektorasdj, 2024-11-30:
Hello, thank you bb-qq for giving us the ability to upgrade the network of our NAS. Do you think it's possible to re-create the bonding with the same settings as it was before reboot through a script that will be called at boot and sleep for some seconds until the USB devices are recognized? I tried creating something like this with the help of chatgpt but while the bond is created all the network settings and attached nics are missing from its configuration.

bb-qq, 2024-12-31:
The driver can be loaded early by extracting the spk package and copying `r8152.ko` directly into the system under `/lib/modules/`.
However, I do not recommend this method as it modifies the system.

zhengran14, 2025-01-14:
Hi, do you use the same RTL8156B? I use two 2.5g usb with different bands and when I set LACP, there are no bond1 in web interface, but with ssh I can see bond1, and also bond1 is not useful.

bb-qq, 2025-02-16:
No, a configuration file for auto-loading must be placed under `/lib/modules-load.d/`.

By the way, what is your motivation to configure BOND with two USB adapters?
Of course throughput will not be improved, and if stability problems arise, they will be at the USB level, so two devices will fail at the same time.

zhengran14, 2025-02-17:
> No, a configuration file for auto-loading must be placed under `/lib/modules-load.d/`.
>
> By the way, what is your motivation to configure BOND with two USB adapters? Of course throughput will not be [...]

For me, I have many devices at home. When one device accesses the NAS at full speed, other devices accessing it will become slow because of the full bandwidth usage. Currently, my approach is to have one 2.5G connection for pure data access and another 2.5G for other devices, which does resolve my issue. However, I hope that if I can set up LACP, it can handle the load balancing itself, rather than needing to pre-assign different NAS IP addresses for different devices.

As for the stability of USB devices, at least in the many years I’ve used them, I’ve never encountered any instability. Moreover, my DS918+ device doesn't support additional PCIe devices, so the USB network card is a last resort.

bb-qq, 2025-03-20:
Thanks for the explanation. I understand the situation.

Realtek has recently released the RTL8157 which has 5Gbps speed and is not very expensive, so if the above approach does not work for you, please consider using this product.

ektorasdj, 2025-04-14:
@bb-qq  Hi! Could you help me test loading the module earlier during boot?

I created a boot-up task in Task Scheduler that runs:

insmod /lib/modules/r8152.ko

Unfortunately, the bond still breaks and reverts to two separate ports with DHCP enabled but no IP assigned—until I reboot without the bond setup.

[image]

I also tried creating a .conf file in /lib/modules-load.d/ with just r8152 inside (and proper permissions), but after rebooting, the NAS became completely unreachable—even from the non-USB ports. I ended up needing to reinstall everything from scratch, so I might’ve done something wrong.

Thank you in advance

ektorasdj, 2025-04-15:
@bb-qq I was finally able to achieve what I wanted using two steps:

    I created a script that rebuilds the bond interface in the same way Synology's GUI (Control Panel) does. After running it, the bond interface shows up and works correctly.

    I added a rule under /usr/lib/udev/rules.d/ to ensure that the correct USB network adapter consistently gets the same interface name (e.g., eth2). This is important because Synology assigns the bond’s MAC address based on the first eth device.

I'm not 100% certain the script handles everything perfectly, but to summarize:

    It loads the bonding module

    Recreates the bond with the same parameters used by the GUI

    Flushes eth2 and eth3

    Brings all interfaces down and up again

    Then finally requests a DHCP lease on bond0

Thanks, and let me know if anyone wants to try the script on their setup—I'd be happy to share it.

audibleroot, 2025-05-28:
has anyone else gotten this to work?