bb-qq/r8152 #382: SA6400 new chip Epyc7002 should choose which driver? DSM is 7.2.

jesselashen, 2024-02-28:
The new SA6400 | AMD EPYC 7272 model, which driver should we choose to install?

flowsasa, 2024-03-12:
我也遇到了同样的问题

bb-qq, 2024-04-18:
I added the driver for epyc7002.
https://github.com/bb-qq/r8152/releases/download/2.17.1-2/r8152-epyc7002-2.17.1-2_7.2.spk

I would appreciate it if let me know whether it works well or not.

flowsasa, 2024-04-19:
> I added the driver for epyc7002. https://github.com/bb-qq/r8152/releases/download/2.17.1-2/r8152-epyc7002-2.17.1-2_7.2.spk
> 
> I would appreciate it if let me know whether it works well or not.

Perfect,This driver is working on my SA6400.
[image]
[image]

closed 2024-06-13 by bb-qq