bb-qq/r8152 #384: Geminilake Support for DSM 6.2

bankfly, 2024-03-15:
I am currently using DS920+ with CPU model: Geminilake. Due to the need to continue using DS PHOTOS, the system version remains at DSM6.2.

Upon reviewing the latest release 2.17.1-1, I noticed that for Geminilake, there is support for _7.2 and _7.1, but no support for Geminilake_6.2. Will there be support for Geminilake_6.2 in future versions?

Furthermore, I have also checked the Issues and previous older versions. Can I use the driver version r8152-geminilake-2.14.0-3.spk? The network card model I am using is: UGREEN USB C to Ethernet Adapter 2.5G (RTL8156BG/Type-A).

Thank you very much for your work and assistance.

TheS1R, 2024-03-15:
DSM 6.2, which is only minimally supported now, goes fully EOL on 1 October 2024. Have you considered using Synology Photos, which supersedes DS Photos? For security alone, you should migrate to the newest available DSM and Synology utilities and applications.

FYI, I currently run DSM 7.2.1-69057 Update 4 on a DS918+ and multiple DS718+ NASs.

bb-qq, 2024-04-18:
Geminilake is new enough and has good performance that I also strongly recommend you move to DSM7.2. However, I know you have your own circumstances, so I have added the driver for DSM6.2.
https://github.com/bb-qq/r8152/releases/download/2.17.1-2/r8152-geminilake-2.17.1-2_6.2.spk

closed 2024-06-13 by bb-qq