bb-qq/r8152 #387: Network port was visible in DSM after install, now no longer visible, DS218+ and ULT-WIIQ Adapter

gituser823, 2024-03-23:
### Description of the problem

I installed the 2.17.1-1 Release according to Description.
The adapter was then visible in DSM with 2,5 Gbit.

I dont fully remember what I did, but I think I rebooted and removed the 1000Gbit network cable, that was attached to the internal network port.

After the reboot the 2,5Gbit was no longer visible in DSM and I couldnt connect via it.

I reinstalled the driver and when that didnt work, installed 2.17.1-2 Release.

But I am no longer able to see the 2,5 Port show up in DSM.

### Description of your products

DS218+

uname -a
Linux 218plus 4.4.302+ #64570 SMP Thu Jul 20 00:07:29 CST 2023 x86_64 GNU/Linux synology_apollolake_218+

DSM 7.2-64570 Update 3

dongle:
 Hersteller ‏ : ‎ ULT-WIIQ
ASIN ‏ : ‎ B09GBFZ9DZ 

https://www.amazon.de/gp/product/B09GBFZ9DZ/

### Description of your environment

> - connect directly to a PC or using a Ethernet switch

I did not do this yet, because it showed up in DSM originally, so the  Adapter was detected and should work.

Ethernet Switch:

Zyxel MG-108-ZZ0101F Multi-Gigabit Unmanaged Switch

Cable: Cat.8

### Output of ``dmesg`` command

````
[dmesg.txt](https://github.com/bb-qq/r8152/files/14733321/dmesg.txt)

````

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-2         0bda:8156:3104 00  2.10  480MBit/s 200mA 2IFs (Realtek USB 10/100/1G/2.5G LAN 4013000001)
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 65007B9D07CA6064)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         0480:a200:0000 00  3.00 5000MBit/s 896mA 1IF  (TOSHIBA External USB 3.0 20150206014965F)
````

### Output of ``ifconfig -a`` command

````
docker0   Link encap:Ethernet  HWaddr 02:42:A6:C1:79:52  
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          inet6 addr: fe80::42:a6ff:fec1:7952/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:15799 errors:0 dropped:0 overruns:0 frame:0
          TX packets:25451 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:893506 (872.5 KiB)  TX bytes:120939680 (115.3 MiB)

docker-48 Link encap:Ethernet  HWaddr 02:42:B9:33:2C:4A  
          inet addr:172.22.0.1  Bcast:172.22.255.255  Mask:255.255.0.0
          inet6 addr: fe80::42:b9ff:fe33:2c4a/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:438 (438.0 B)

docker-64 Link encap:Ethernet  HWaddr 02:42:17:0A:5F:E7  
          inet addr:172.19.0.1  Bcast:172.19.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

docker-ab Link encap:Ethernet  HWaddr 02:42:C4:39:40:C7  
          inet addr:172.20.0.1  Bcast:172.20.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

docker-c7 Link encap:Ethernet  HWaddr 02:42:38:1C:CD:C3  
          inet addr:172.18.0.1  Bcast:172.18.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

docker-cd Link encap:Ethernet  HWaddr 02:42:32:A5:77:46  
          inet addr:172.21.0.1  Bcast:172.21.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

docker160 Link encap:Ethernet  HWaddr 96:63:25:EC:73:A0  
          inet6 addr: fe80::9463:25ff:feec:73a0/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:10 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:876 (876.0 B)

docker659 Link encap:Ethernet  HWaddr DA:6D:00:F6:46:76  
          inet6 addr: fe80::d86d:ff:fef6:4676/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:15799 errors:0 dropped:0 overruns:0 frame:0
          TX packets:25456 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:1114692 (1.0 MiB)  TX bytes:120940118 (115.3 MiB)

eth0      Link encap:Ethernet  HWaddr 00:11:32:86:0E:F4  
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:110133 errors:0 dropped:0 overruns:0 frame:0
          TX packets:27614 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:129949790 (123.9 MiB)  TX bytes:8514382 (8.1 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:872 errors:0 dropped:0 overruns:0 frame:0
          TX packets:872 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:124284 (121.3 KiB)  TX bytes:124284 (121.3 KiB)

ovs-syste Link encap:Ethernet  HWaddr DE:DD:8E:BD:61:69  
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

ovs_eth0  Link encap:Ethernet  HWaddr 00:11:32:86:0E:F4  
          inet addr:192.168.1.26  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:50113 errors:0 dropped:0 overruns:0 frame:0
          TX packets:24817 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:125283372 (119.4 MiB)  TX bytes:8329780 (7.9 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          inet addr:10.8.0.1  P-t-P:10.8.0.2  Mask:255.255.255.255
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

````

bb-qq, 2024-06-15:
The driver does not seem to be loaded at all.
Could you try starting the driver manually from the Package Center? If the error persists, could you try full re-installation?
https://github.com/bb-qq/r8152/wiki/Troubleshooting#uninstall-completely--full-reinstallation

gituser823, 2024-06-15:
I already got it working in the meantime.

closed 2024-06-15 by gituser823