bb-qq/r8152 #390: Package suddenly not worked

dateno1, 2024-04-10:
### Description of the problem

> Replace this line with your answer.

### Description of your products

> - the product name of your NAS model
>     - and output of ``uname -a`` command
>     - DSM version x.x
> - the product name of the ethernet dongle
DS918+ (xpenology)
Linux NAS 4.4.302+ #69057 SMP Fri Jan 12 17:02:59 CST 2024 x86_64 GNU/Linux synology_apollolake_918+
Cablemate 2.5G USB Ethernet (0bda:8156:3000)
### Description of your environment

> - connect directly to a PC or using a Ethernet switch
>     - PC: NIC model name or chipset name
>     - Ethernet switch: model name
> - cable category (Cat.5e or Cat.6 or Cat.7)
PC used same RTL8156
Connected by switch with multiple 2.5G devices (Other work well)

Cable is Cat.6

Switch is QNAP QSW Series (5P)

### Output of ``dmesg`` command

````
[dmesg.txt](https://github.com/bb-qq/r8152/files/14926857/dmesg.txt)

````

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:14.0) hub
  |__1-7         0bda:0129:3960 ff  2.00  480MBit/s 500mA 1IF  (Generic USB2.0-CRW 20100201396000000)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:14.0) hub
  |__2-1         f400:f400:0100 00  3.00 5000MBit/s 896mA 1IF  (SanDisk Ultra Fit 4C530001071029109505)
  |__2-3         0bda:8156:3000 00  3.20 5000MBit/s 512mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 000000001)
````

USB Keyboard is removed before type it

### Output of ``ifconfig -a`` command

````
docker0   Link encap:Ethernet  HWaddr 02:42:4D:64:10:99
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth0      Link encap:Ethernet  HWaddr 9C:5C:8E:20:F5:25
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:85030 errors:0 dropped:0 overruns:0 frame:0
          TX packets:155707 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:8964180 (8.5 MiB)  TX bytes:213353842 (203.4 MiB)

eth1      Link encap:Ethernet  HWaddr A0:CE:C8:C8:14:5F
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:75 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:8721 (8.5 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:19774 errors:0 dropped:0 overruns:0 frame:0
          TX packets:19774 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:2764065 (2.6 MiB)  TX bytes:2764065 (2.6 MiB)

ovs-syste Link encap:Ethernet  HWaddr 7A:89:7E:36:BE:27
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

ovs_eth0  Link encap:Ethernet  HWaddr 9C:5C:8E:20:F5:25
          inet addr:192.168.1.129  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: 2001:470:24:2ff:9e5c:8eff:fe20:f525/64 Scope:Global
          inet6 addr: fe80::9e5c:8eff:fe20:f525/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:81367 errors:0 dropped:0 overruns:0 frame:0
          TX packets:19687 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:7569999 (7.2 MiB)  TX bytes:205920319 (196.3 MiB)

ovs_eth1  Link encap:Ethernet  HWaddr A0:CE:C8:C8:14:5F
          inet addr:192.168.11.1  Bcast:192.168.11.255  Mask:255.255.255.0
          inet6 addr: fe80::a2ce:c8ff:fec8:145f/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:75 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:8721 (8.5 KiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tap021132 Link encap:Ethernet  HWaddr 46:25:DD:1F:46:80
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:675 errors:0 dropped:0 overruns:0 frame:0
          TX packets:44100 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:51045 (49.8 KiB)  TX bytes:3833479 (3.6 MiB)
````
---------------------------------------
It work well at today morning but not work after reboot NAS

I fount no LAN 2 on Control panel and checked package manager

It need to fix and click fix i can found LAN 2 on DSM but can't communication with other device

Reboot it again package broken again

I tried
remove -> reboot -> reinstall -> reboot
same result

$route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         192.168.1.1     0.0.0.0         UG    0      0        0 ovs_eth0
172.17.0.0      0.0.0.0         255.255.0.0     U     0      0        0 docker0
192.168.1.0     0.0.0.0         255.255.255.0   U     0      0        0 ovs_eth0
192.168.11.0    0.0.0.0         255.255.255.0   U     0      0        0 ovs_eth1

$ping 192.168.11.1
PING 192.168.11.1 (192.168.11.1) 56(84) bytes of data.
64 bytes from 192.168.11.1: icmp_seq=1 ttl=64 time=0.034 ms

$ping 192.168.11.3
PING 192.168.11.3 (192.168.11.3) 56(84) bytes of data.
From 192.168.11.1 icmp_seq=1 Destination Host Unreachable

$lsmod|grep r8
r8152                 224445  0
r8168                 544464  0
r8169                  71712  0
mii                     4099  2 r8169,usbnet
usbcore               210297  12 etxhci_hcd,r8152,usblp,uhci_hcd,usb_storage,ehci_hcd,ehci_pci,usbhid,usbnet,cdc_ncm,xhci_hcd,xhci_pci

Tried Force ReInstall DSM but nothing changed

bb-qq, 2024-04-17:
Did you try full re-installation?
https://github.com/bb-qq/r8152/wiki/Troubleshooting#uninstall-completely--full-reinstallation

dateno1, 2024-04-17:
> Did you try full re-installation? https://github.com/bb-qq/r8152/wiki/Troubleshooting#uninstall-completely--full-reinstallation

Yes I tried that (remove is uninstall&delete appconf)

but nothing changed

Jcambass, 2024-04-23:
👋 I'm experiencing the same problem. After installation the NIC is recognised. After a restart of DSM the NIC will not be recognised anymore.

DSM Version: DSM 7.2.1-69057 Update 5
Adapter: [CLUB 3D CAC-1420](https://amzn.to/2ZPmzKD) (RTL8156/Type-A)
Device: DS1522+

taxi911, 2024-05-15:
For Xpnology user + ARCLoader 24.5.13
tested on noname AliExpress USB3.0 to LAN dongle on rtl8152 chip
sorry but i dont have a 2.5G dongle
TRY:
1.Install as usual
[https://github.com/bb-qq/r8152]
2. After reboot you have broken pakage
3.Reboot go to ARCLoader config mode
4. Go to "Show ARC Loader"
5. Go to DSM Modules
6. Go to "Choose Modules"
7. Find r8152 and uncheck it
8. Find realteck (Realteck PHY driver) tick this
9. Rebuild Loader 
Everything should work and after reboot too.

dateno1, 2024-05-15:
> For Xpnology user + ARCLoader 24.5.13 tested on noname AliExpress USB3.0 to LAN dongle on rtl8152 chip sorry but i dont have a 2.5G dongle TRY: 1.Install as usual [https://github.com/bb-qq/r8152] 2. [...]

I was used ARPL

At first time all OK. but some day i was reboot NAS and package is broken

No DSM or bootloiader chagend

taxi911, 2024-05-16:
> > For Xpnology user + ARCLoader 24.5.13 tested on noname AliExpress USB3.0 to LAN dongle on rtl8152 chip sorry but i dont have a 2.5G dongle TRY: 1.Install as usual [https://github.com/bb-qq/r8152] 2. After reboot you have broken pakage 3.Reboot go to ARCLoader config mode 4. Go to "Show ARC Loader" 5. Go to DSM Modules 6. Go to "Choose Modules" 7. Find r8152 and uncheck it 8. Find realteck (Realteck PHY driver) tick this 9. Rebuild Loader Everything should work and after reboot too.
> 
> I was used ARPL
> 
> At first time all OK. but some day i was reboot NAS and package is broken
> 
> No DSM or bootloiader changed
> 
> OK I will reCheck bootloader setting

Try to deselect r8152 driver in Loader Modules

- ARPL v1.1 Beta  have a r8152 driver in the:
Advanced menu - Modules - Coose modules to include - r8152

- ARPL i18
Modules menu - Show/Select modules - r8152

Another way (not automatically)
- manualy Restore package in the package center after next reboot
- via ssh somthing like this (do not remeber right command)  synopkg restart <package name>

I dont find any solusion that restart broken package after boot automatically not via Scheduler and not via script.
Sorry for my English.

dateno1, 2024-05-16:
> > > For Xpnology user + ARCLoader 24.5.13 tested on noname AliExpress USB3.0 to LAN dongle on rtl8152 chip sorry but i dont have a 2.5G dongle TRY: 1.Install as usual [https://github.com/bb-qq/r8152] 2. After reboot you have broken pakage 3.Reboot go to ARCLoader config mode 4. Go to "Show ARC Loader" 5. Go to DSM Modules 6. Go to "Choose Modules" 7. Find r8152 and uncheck it 8. Find realteck (Realteck PHY driver) tick this 9. Rebuild Loader Everything should work and after reboot too.
> > 
> > 
> > I was used ARPL
> > At first time all OK. but some day i was reboot NAS and package is broken
> > No DSM or bootloiader changed
> > OK I will reCheck bootloader setting
> 
> Try to deselect r8152 driver in Loader Modules
> 
>     * ARPL v1.1 Beta  have a r8152 driver in the:
>       Advanced menu - Modules - Coose modules to include - r8152
> 
>     * ARPL i18
>       Modules menu - Show/Select modules - r8152
> 
> 
> Another way (not automatically)
> 
>     * manualy Restore package in the package center after next reboot
> 
>     * via ssh somthing like this (do not remeber right command)  synopkg restart
> 
> 
> I dont find any solusion that restart broken package after boot automatically not via Scheduler and not via script. Sorry for my English.

Thanks for help

Problem is fixed

I don't know why it broken

closed 2024-05-16 by dateno1