bb-qq/r8152 #391: 2.5G Ethernet very slow on DS214play (evansport)

N0b0dy0101, 2024-04-16:
### Description of the problem
Hi, would really appreciate if you can help solve this problem with my DS214Play
My 2.5GbE adapters are detected with correctly from my PC and NAS, but iperf test are very disappointing.
I tried my 2 adapters with PC-PC, and iperf3 results are good and expected, but with PC-NAS it shows below values only.

iperf3 -s
```
-----------------------------------------------------------
Server listening on 5201 (test #1)
-----------------------------------------------------------
Accepted connection from 10.0.4.3, port 49726
[  5] local 10.0.4.105 port 5201 connected to 10.0.4.3 port 49727
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  84.4 MBytes   708 Mbits/sec                  
[  5]   1.00-2.00   sec  84.6 MBytes   710 Mbits/sec                  
[  5]   2.00-3.00   sec  84.8 MBytes   712 Mbits/sec                  
[  5]   3.00-4.00   sec  89.3 MBytes   749 Mbits/sec                  
[  5]   4.00-5.00   sec  89.0 MBytes   747 Mbits/sec                  
[  5]   5.00-6.00   sec  77.6 MBytes   651 Mbits/sec                  
[  5]   6.00-7.00   sec  86.8 MBytes   728 Mbits/sec                  
[  5]   7.00-8.00   sec  87.4 MBytes   733 Mbits/sec                  
[  5]   8.00-9.00   sec  90.5 MBytes   759 Mbits/sec                  
[  5]   9.00-10.00  sec  77.6 MBytes   650 Mbits/sec                  
[  5]  10.00-10.00  sec  55.1 KBytes   689 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.00  sec   852 MBytes   715 Mbits/sec                  receiver
```

### Description of your products
Model: DS214play
CPU: INTEL Atom CE5335
USB Device: USB 10/100/1G/2.5G LAN - Realtek Semiconductor Corp.
DSM Version: DSM 7.1.1-42962 Update 5
Ethernet Dongle: Unitek 2.5G Ethernet Adapter
Linux DiskStation 3.2.101 #42962 SMP PREEMPT Fri Mar 24 00:27:46 CST 2023 i686 GNU/Linux synology_evansport_214play

### Description of your environment
Ethernet switch:  TP-Link TL-SH1008 8-Port 2.5GbE Switch
Both my NAS and my PC are connected to my 2.5GbE switch using Cat6 cables and indicators are showing both are connected via 2.5GbE

### Output of ``dmesg`` command
(https://github.com/bb-qq/r8152/files/14986597/dmesg.txt)

### Output of ``lsusb`` command
|__usb1          1d6b:0002:0302 09  2.00  480MBit/s   0mA 1IF  (ehci_hcd 0000:01:0d.0) hub
|__usb2          1d6b:0002:0302 09  2.00  480MBit/s   0mA 1IF  (ehci_hcd 0000:01:0d.1) hub
  |__2-1         05e3:0726:9910 00  2.00  480MBit/s 500mA 1IF  ( 000000009910)
|__usb3          1d6b:0002:0302 09  2.00  480MBit/s   0mA 1IF  (ehci_hcd 0000:01:0d.2) hub
|__usb4          1d6b:0002:0302 09  2.00  480MBit/s   0mA 1IF  (Linux 3.2.101 etxhci_hcd-161118 Etron xHCI Host Controller 0000:02:00.0) hub
|__usb5          1d6b:0003:0302 09  3.00 5000MBit/s   0mA 1IF  (Linux 3.2.101 etxhci_hcd-161118 Etron xHCI Host Controller 0000:02:00.0) hub
  |__5-1         0bda:8156:3104 00  3.20 5000MBit/s  64mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 4013000001)

### Output of ``ifconfig -a`` command
eth1      Link encap:Ethernet  HWaddr 00:E0:4C:40:BD:D6  
          inet addr:10.0.4.105  Bcast:10.0.4.255  Mask:255.255.255.0
          inet6 addr: fe80::2e0:4cff:fe40:bdd6/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1292373 errors:0 dropped:0 overruns:0 frame:0
          TX packets:479374 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1891606341 (1.7 GiB)  TX bytes:32765263 (31.2 MiB)

VenimK, 2024-04-17:
whats does 'ethtool eth1 | grep Speed' say on your system

ON my NAS
	Speed: 2500Mb/s

N0b0dy0101, 2024-04-17:
> ethtool eth1 | grep Speed

Hi VenimK, yes it's the same, it says
Speed: 2500Mb/s

transformcube, 2024-04-25:
The processing power of INTEL Atom CE5335 is not powerful enough for USB 2.5Gbps adapter, like current Intel N5105 just enough for single 10Gbps adapter marginally.

N0b0dy0101, 2024-04-25:
> The processing power of INTEL Atom CE5335 is not powerful enough for USB 2.5Gbps adapter, like current Intel N5105 just enough for single 10Gbps adapter marginally.

That explains why I tried different brands, still is slower than the onboard 1gbps nic.  Too bad, would not be able to utilize my 2.5gbps switch.  Thanks for the info ****.

VenimK, 2024-04-25:
Are you running xpenology
If so.
My give it a try-out with the ARC loader

bb-qq, 2024-06-15:
Did you try to increase MTU value?
This page may help.
https://github.com/bb-qq/r8152/wiki/Troubleshooting#performance-issues