bb-qq/r8152 #397: DS923+ Problem after update (DSM 7.2.1)

PbkTomas, 2024-04-28:
### Description of the problem

> Replace this line with your answer.

I was using the r8152-r1000-2.16.3-4_7.2 version well,
After updating DS923+ to DSM 7.2.1-69057 Update 5 a few days ago
LAN card connection is not working due to a problem.

r8152-r1000-2.17.1-1_7.2 Division
Both r8152-r1000-2.17.12.1-2_7.2 have driver packages in use,
The network interface comes out in the disconnected state.

### Description of your products

> - DS923+
> - DSM 7.2.1-69057 Update 5
> - Next-2501GU3 (RTL8152)

### Description of your environment

> - connect 2.5 Gib Ethernet switch
> - Ethernet switch : TP-LINK TL-SG105S-M2
> - cable category (Cat.7)

### Output of ``dmesg`` command

````
[    2.533006] AMD-Vi: Unable to write to IOMMU perf counter.

[    4.296524] Cannot find slot node of this ata_port.
[    4.301436] Cannot find slot node of this ata_port.

[    4.674156] ata4 : SATA link down (SStatus 0 SControl 300)
[    4.679656] ata4 : No present pin info for SATA link down event
[    4.685610] ata5 : SATA link down (SStatus 0 SControl 300)
[    4.691102] ata5 : No present pin info for SATA link down event
[    4.702158] ata3 : SATA link down (SStatus 0 SControl 300)
[    4.707655] ata3 : No present pin info for SATA link down event

[   50.715233] md0: detected capacity change from 0 to 2549940224
[   50.896460] usb 2-3: new SuperSpeed USB device number 2 using xhci_hcd
[   51.985936] EXT4-fs (md0): couldn't mount as ext3 due to feature incompatibilities
[   52.010884] EXT4-fs (md0): mounted filesystem with ordered data mode. Opts: (null)
[   52.055433] EXT4-fs (md0): couldn't mount as ext3 due to feature incompatibilities
[   52.111479] EXT4-fs (md0): mounted filesystem with ordered data mode. Opts: barrier=1
[   52.197803] loop: module loaded
[   56.413258] EXT4-fs (md0): couldn't mount as ext3 due to feature incompatibilities
[   56.435603] EXT4-fs (md0): mounted filesystem with ordered data mode. Opts: (null)
[   56.645847] EXT4-fs (md0): couldn't mount as ext3 due to feature incompatibilities
[   56.668903] EXT4-fs (md0): mounted filesystem with ordered data mode. Opts: barrier=1
[   56.769132] systemd[1]: systemd 219 running in system mode. (+PAM -AUDIT -SELINUX -IMA -APPARMOR -SMACK +SYSVINIT -UTMP -LIBCRYPTSETUP -GCRYPT -GNUTLS -ACL +XZ -LZ4 -SECCOMP +BLKID -ELFUTILS +KMOD -IDN)

[   58.757877] systemd[1]: Cannot add dependency job for unit synorelayd.service, ignoring: Unit synorelayd.service failed to load: No such file or directory.
[   58.771880] systemd[1]: Cannot add dependency job for unit pkg-install-high-priority-builtin.service, ignoring: Unit pkg-install-high-priority-builtin.service failed to load: No such file or directory.
[   58.789772] systemd[1]: Cannot add dependency job for unit pkg-synorelayd-reload-package-config.service, ignoring: Unit synorelayd.service failed to load: No such file or directory.
[   58.805919] systemd[1]: Cannot add dependency job for unit pkg-synorelayd-reload-fixed-package-config.service, ignoring: Unit synorelayd.service failed to load: No such file or directory.
[   58.822601] systemd[1]: Cannot add dependency job for unit pkgctl-Node.js_v12.service, ignoring: Unit pkgctl-Node.js_v12.service failed to load: No such file or directory.
[   58.837894] systemd[1]: Cannot add dependency job for unit pkg-install-low-priority-builtin.service, ignoring: Unit pkg-install-low-priority-builtin.service failed to load: No such file or directory.
[   58.855682] systemd[1]: Cannot add dependency job for unit pkg-synobrm-recovery-restart.service, ignoring: Unit pkg-synobrm-recovery-restart.service failed to load: No such file or directory.

[   59.638934] systemd[1]: Starting QuickConnect's slice.
[   59.643876] flashcache: flashcache-1.0-2023-09-23-22:09 initialized
[   59.647708] flashcache_syno: flashcache-1.0-syno-v25-pin-file-2023-09-23-22:09-debug initialized

[   59.755448]  Reinit 1 8001
[   59.755462]  Reinit 2 1
[   59.765497]  Reinit 1 8001
[   59.765510]  Reinit 2 1
[   59.766157] r8168: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
[   59.766159] r8168  Copyright (C) 2017  Realtek NIC software team <nicfae@realtek.com>
                This program comes with ABSOLUTELY NO WARRANTY; for details, please see <http://www.gnu.org/licenses/>.
                This is free software, and you are welcome to redistribute it under certain conditions; see <http://www.gnu.org/licenses/>.

[   62.737846]  Reinit 1 8001
[   62.740659]  Reinit 2 1
[   62.753182]  Reinit 1 8001
[   62.755994]  Reinit 2 1
[   62.768849]  Reinit 1 8001
[   62.771661]  Reinit 2 1
[   62.805162] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   63.197032] eth1: 0xffffc9000009c000, 90:09:d0:32:a2:1f, IRQ 310
[   63.208822]  Reinit 1 8001
[   63.211630]  Reinit 2 1
[   63.224164]  Reinit 1 8001
[   63.226972]  Reinit 2 1
[   63.239831]  Reinit 1 8001
[   63.242638]  Reinit 2 1
[   63.276162] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
[   63.475308] 8021q: 802.1Q VLAN Support v1.8

[   65.792005] md2: detected capacity change from 0 to 23971736322048
[   66.789789] flashcache_syno: /dev/vg1/volume_1 exclude check success

[   67.073686] Btrfs loaded, crc32c=crc32c-intel
[   67.279515] BTRFS: device label 2023.02.15-05:27:55 v42962
[   67.290589]  Reinit 1 8001
[   67.290602]  Reinit 2 1

[   67.820269]  Reinit 1 8001
[   67.823079]  Reinit 2 1

````

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:06:00.3) hub
  |__1-1         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 65002BAD07238D95)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:06:00.3) hub
  |__2-3         0bda:8156:3100 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 001000001)
````

### Output of ``ifconfig -a`` command

````
eth0      Link encap:Ethernet  HWaddr 90:09:D0:32:A2:1E
          inet addr:192.168.1.223  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::9209:d0ff:fe32:a21e/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:39414 errors:0 dropped:0 overruns:0 frame:0
          TX packets:32997 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:9774115 (9.3 MiB)  TX bytes:29671454 (28.2 MiB)
          Interrupt:53 base 0x8000

eth1      Link encap:Ethernet  HWaddr 90:09:D0:32:A2:1F
          inet addr:192.168.1.224  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::9209:d0ff:fe32:a21f/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:7696 errors:0 dropped:0 overruns:0 frame:0
          TX packets:313 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1457556 (1.3 MiB)  TX bytes:44428 (43.3 KiB)
          Interrupt:54 base 0xc000

eth2      Link encap:Ethernet  HWaddr 00:E0:4C:68:12:B9
          inet addr:192.168.1.221  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:56594 errors:0 dropped:0 overruns:0 frame:0
          TX packets:56594 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:11042562 (10.5 MiB)  TX bytes:11042562 (10.5 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

JeffreyRon, 2024-04-29:
I have a similar problem after updating. I'm on the same 7.2.1-69057 Update 5, using the PLUGABLE 2.5G USB-C AND USB TO ETHERNET ADAPTER. It was working great before, but now my speeds are in the 90Mb/s. I was getting 2.35Gb/sec before. I've had to revert back to the regular built in 1Gb/sec connector and now I'm getting a normal, just under 1Gb/sec.

PbkTomas, 2024-05-12:
I suddenly got a driver. I couldn't try many times on the day the error occurred after the update, but it must have been a temporary problem.

closed 2024-05-12 by PbkTomas