bb-qq/r8152 #400: DS420+ does not recognize 2.5GB USB Ethernet Adapter

YonnJungpapa, 2024-05-13:
### Description of the problem

Hello, I'm attempting to use two 2.5G adapters on my DS420+ but only one interface is recognized, leaving me in need of assistance.

I've inserted USB LAN adapters into the USB 3.0 ports on both the front and back of the NAS, but only one side is recognized while the other isn't.

These USB LAN adapters are recognized well on my PC. Even when I insert one adapter through a USB hub, it still doesn't get recognized.

I'd appreciate any help.

### Description of your products

Model: DS420+
CPU: Intel Celeron J4025
DSM Version: DSM 7.2.1-69057 Update 5
Ethernet dongle product name: NEXT-2501GU3

### Description of your environment

Ethernet switch: TRENDnet 8-Port Unmanaged 2.5G Switch
I access the NAS through an Ethernet switch. Both the NAS and laptop are using Cat8 cables and 2.5G USB Ethernet Adapters.
All the 2.5G USB Ethernet Adapters I have are functioning properly when connected to a PC.

### Output of ``dmesg`` command

[dmesg.txt](https://github.com/bb-qq/r8152/files/15298089/dmesg.txt)

### Output of ``lsusb`` command
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F00124B05CC4565)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         0bda:8156:3100 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 001000001)
  |__2-2         0bda:8156:3100 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 001000000)

### Output of ``ifconfig -a`` command

eth0      Link encap:Ethernet  HWaddr 00:11:32:EB:E2:17
          inet addr:172.30.1.30  Bcast:172.30.1.255  Mask:255.255.255.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:88 base 0xc000

eth1      Link encap:Ethernet  HWaddr 00:11:32:EB:E2:18
          inet addr:172.30.1.40  Bcast:172.30.1.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:368396 errors:0 dropped:0 overruns:0 frame:0
          TX packets:907339 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:27684184 (26.4 MiB)  TX bytes:2984363908 (2.7 GiB)
          Interrupt:89 base 0xe000

eth2      Link encap:Ethernet  HWaddr 00:E0:4C:68:13:2D
          inet addr:172.30.1.42  Bcast:172.30.1.255  Mask:255.255.255.0
          inet6 addr: fe80::2e0:4cff:fe68:132d/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:14850 errors:0 dropped:0 overruns:0 frame:0
          TX packets:7038 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:4749960 (4.5 MiB)  TX bytes:1017843 (993.9 KiB)

eth3      Link encap:Ethernet  HWaddr 00:E0:4C:68:11:E4
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:144103 errors:0 dropped:0 overruns:0 frame:0
          TX packets:144103 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:24187383 (23.0 MiB)  TX bytes:24187383 (23.0 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

sammam2, 2024-05-16:
Did you check the Troubleshooting page?
https://github.com/bb-qq/r8152/wiki/Troubleshooting#performance-issues

Other Issues / Tips
Hot plugging does not work
The hot plug feature is disabled by default. To enable it, execute the following command:

sudo bash /var/packages/r8152/scripts/install-udev-rules

Note: This command modifies the system files of the DSM. Uninstalling the driver package will discard that change.

Multiple identical devices do not work
Please enable the hot plug feature described above.

Since the same products have the same serial number, connecting them to one NAS at the same time will trigger the workaround process of the Linux kernel. This will prevent recognition. Using the hot plug feature avoids this problem.

This is not a problem when combining different products.

bb-qq, 2024-06-13:
I am closing this issue as the method suggested in one previous comment should resolve this issue.
If not, feel free to re-open it.

closed 2024-06-13 by bb-qq