bb-qq/r8152 #403: Can IPv6 be disable?

SiLenCe-WB, 2024-06-11:
### Description of the problem

Even if I disable IPv6 from my network interface, when I reboot the system it's back on. How can I disable it permanently from boot?

### Description of your products

> DS418play
> DSM 7.2.1-69057 Update 5
> SABRENT ‎NT-S25G (2.5 gigabit ethernet)

bb-qq, 2024-06-15:
There is no way to disable IPv6 at this time, as the driver does not respect the settings on the DSM side.
Could you please use the task scheduler to work around the problem?

SiLenCe-WB, 2024-06-15:
I have created a task schedule script that runs this command as root and it works 👍

sysctl -w net.ipv6.conf.all.disable_ipv6=1

bb-qq, 2024-06-15:
I would like to close the issue as I do not plan to change the driver at this time.

closed 2024-06-15 by bb-qq