bb-qq/r8152 #405: ds920+ download only 1Gbps when I use this plugin

vencenty, 2024-06-13:
### Description of the problem
[image]

### Description of your products

DS920+
DSM 7.2.1-69057 Update 5

### Description of your environment

I use ds920+ (using 2.5G network card) to connect to 2.5G switch, and 2.5G switch to Mac (using 2.5G network card).
At this time, the speed of uploading from Mac to Synology is 2.5G, and the speed of downloading Synology files from Mac to local is only 1G.
I also have another ugreenNas, and I can run at full 2.5G speed by connecting Mac and uGreenNas through a switch.

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-1         0463:ffff:0100 00  1.10   12MBit/s 20mA 1IF  (EATON SANTAK TG-BOX Blank)
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 6500276A082DF931)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-2         0bda:8156:3104 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 4013000001)
````

### Output of ``ifconfig -a`` command

````
root@SynologyNas:~# ifconfig -a
bond0     Link encap:Ethernet  HWaddr 90:09:D0:28:2B:F5
          inet addr:192.168.50.10  Bcast:192.168.50.255  Mask:255.255.255.0
          inet6 addr: 2408:8207:2591:4ee0:9209:d0ff:fe28:2bf5/64 Scope:Global
          inet6 addr: fe80::9209:d0ff:fe28:2bf5/64 Scope:Link
          UP BROADCAST RUNNING MASTER MULTICAST  MTU:1500  Metric:1
          RX packets:19767 errors:0 dropped:0 overruns:0 frame:0
          TX packets:148090 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:5414997 (5.1 MiB)  TX bytes:843164006 (804.1 MiB)

docker0   Link encap:Ethernet  HWaddr 02:42:FE:1C:43:EA
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          inet6 addr: fe80::42:feff:fe1c:43ea/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:90764 errors:0 dropped:0 overruns:0 frame:0
          TX packets:113095 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:831843306 (793.3 MiB)  TX bytes:2055224465 (1.9 GiB)

docker945 Link encap:Ethernet  HWaddr 76:EC:A2:28:0E:1A
          inet6 addr: fe80::74ec:a2ff:fe28:e1a/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:480 (480.0 B)

docker-76 Link encap:Ethernet  HWaddr 02:42:1A:C1:C5:4E
          inet addr:172.18.0.1  Bcast:172.18.255.255  Mask:255.255.0.0
          inet6 addr: fe80::42:1aff:fec1:c54e/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:183 errors:0 dropped:0 overruns:0 frame:0
          TX packets:141 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:29612 (28.9 KiB)  TX bytes:26769 (26.1 KiB)

docker5e7 Link encap:Ethernet  HWaddr BA:05:4E:DD:5C:ED
          inet6 addr: fe80::b805:4eff:fedd:5ced/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:400 errors:0 dropped:0 overruns:0 frame:0
          TX packets:337 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:59796 (58.3 KiB)  TX bytes:503045 (491.2 KiB)

docker6f0 Link encap:Ethernet  HWaddr DE:46:96:22:47:72
          inet6 addr: fe80::dc46:96ff:fe22:4772/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:168 (168.0 B)

docker793 Link encap:Ethernet  HWaddr 7A:60:6E:69:47:1A
          inet6 addr: fe80::7860:6eff:fe69:471a/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:183 errors:0 dropped:0 overruns:0 frame:0
          TX packets:146 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:32174 (31.4 KiB)  TX bytes:27207 (26.5 KiB)

docker88b Link encap:Ethernet  HWaddr CA:A2:FB:3E:F3:7E
          inet6 addr: fe80::c8a2:fbff:fe3e:f37e/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:89542 errors:0 dropped:0 overruns:0 frame:0
          TX packets:112116 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:832967673 (794.3 MiB)  TX bytes:2054410213 (1.9 GiB)

dockercc1 Link encap:Ethernet  HWaddr 36:A5:A6:16:A9:8F
          inet6 addr: fe80::34a5:a6ff:fe16:a98f/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:822 errors:0 dropped:0 overruns:0 frame:0
          TX packets:657 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:86533 (84.5 KiB)  TX bytes:312389 (305.0 KiB)

dockere43 Link encap:Ethernet  HWaddr A6:5C:17:1A:FF:F6
          inet6 addr: fe80::a45c:17ff:fe1a:fff6/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:7 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:522 (522.0 B)

eth0      Link encap:Ethernet  HWaddr 90:09:D0:28:2B:F5
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:12392 errors:0 dropped:0 overruns:0 frame:0
          TX packets:57968 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:4369381 (4.1 MiB)  TX bytes:43765544 (41.7 MiB)
          Interrupt:99 base 0x4000

eth1      Link encap:Ethernet  HWaddr 90:09:D0:28:2B:F6
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:7375 errors:0 dropped:0 overruns:0 frame:0
          TX packets:90122 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1045616 (1021.1 KiB)  TX bytes:799398462 (762.3 MiB)
          Interrupt:98

eth2      Link encap:Ethernet  HWaddr 00:E0:4C:68:10:AE
          inet addr:192.168.50.100  Bcast:192.168.50.255  Mask:255.255.255.0
          inet6 addr: fe80::2e0:4cff:fe68:10ae/64 Scope:Link
          inet6 addr: 2408:8207:2591:4ee0:2e0:4cff:fe68:10ae/64 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:1463157 errors:0 dropped:0 overruns:0 frame:0
          TX packets:281 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:2142460036 (1.9 GiB)  TX bytes:53826 (52.5 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:11201 errors:0 dropped:0 overruns:0 frame:0
          TX packets:11201 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:3133166 (2.9 MiB)  TX bytes:3133166 (2.9 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

vencenty, 2024-06-13:
please help me

bb-qq, 2024-06-15:
Did you also increase the MTU value on the Ugreen NAS side?
And measuring throughputs using iperf may indicate some additional insights.

bb-qq, 2025-01-03:
Since there has been no response for a long time, this issue is closed. Please feel free to re-open.

closed 2025-01-03 by bb-qq