bb-qq/r8152 #406: r1852 adaper regconnized but it does not appear in network interface

willng-max, 2024-06-16:
### Description of the problem

I have a LLANO 2.5gbs network adapter (type c, with adapter to type A)
Device is recognize in info center after installing driver & restart NAS. However, it does not appear in network interface

### Description of your products
DS923+
DSM 7.2.1-69057 build 5
Linux will_nas 4.4.302+ #69057 SMP Fri Jan 12 17:01:14 CST 2024 x86_64 GNU/Linux synology_r1000_923+
LLANO 2-in-1 Type-C USB RJ45 Ethernet adapter

### Output of ``lsusb`` command
willy@will_nas:~$ lsusb
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:12.2) hub
  |__1-3         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Generic Mass Storage 663210CF)
|__usb2          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:13.2) hub
|__usb3          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:10.0) hub
|__usb4          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:10.0) hub
  |__4-1         0bda:8156:3104 00  3.20 5000MBit/s 256mA 2IFs (Realtek USB 10/100/1G/2.5G LAN 4013000001)

### Output of ``ifconfig -a`` command

willy@will_nas:~$ ifconfig -a
docker0   Link encap:Ethernet  HWaddr 02:42:05:08:53:D4
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

docker-11 Link encap:Ethernet  HWaddr 02:42:57:0C:98:99
          inet addr:172.19.0.1  Bcast:172.19.255.255  Mask:255.255.0.0
          inet6 addr: fe80::42:57ff:fe0c:9899/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:406 errors:0 dropped:0 overruns:0 frame:0
          TX packets:375 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:431008 (420.9 KiB)  TX bytes:37057 (36.1 KiB)

docker1be Link encap:Ethernet  HWaddr 56:17:E1:0C:F6:DA
          inet6 addr: fe80::5417:e1ff:fe0c:f6da/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:27496 errors:0 dropped:0 overruns:0 frame:0
          TX packets:21853 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:11333149 (10.8 MiB)  TX bytes:7196035 (6.8 MiB)

docker819 Link encap:Ethernet  HWaddr 82:9E:AE:DF:4C:93
          inet6 addr: fe80::809e:aeff:fedf:4c93/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:9 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:606 (606.0 B)

dockerb9a Link encap:Ethernet  HWaddr 4E:85:03:B9:D8:15
          inet6 addr: fe80::4c85:3ff:feb9:d815/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:12122 errors:0 dropped:0 overruns:0 frame:0
          TX packets:13637 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:1579622 (1.5 MiB)  TX bytes:4601332 (4.3 MiB)

dockerf6e Link encap:Ethernet  HWaddr 72:F8:0A:A6:C9:AE
          inet6 addr: fe80::70f8:aff:fea6:c9ae/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:9362 errors:0 dropped:0 overruns:0 frame:0
          TX packets:13485 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:5579608 (5.3 MiB)  TX bytes:6299383 (6.0 MiB)

eth0      Link encap:Ethernet  HWaddr 00:E0:4C:68:E3:F7
          inet addr:192.168.88.8  Bcast:192.168.88.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:101382 errors:0 dropped:0 overruns:0 frame:0
          TX packets:269801 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:101054544 (96.3 MiB)  TX bytes:367431223 (350.4 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:19514 errors:0 dropped:0 overruns:0 frame:0
          TX packets:19514 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:3541719 (3.3 MiB)  TX bytes:3541719 (3.3 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

usb0      Link encap:Ethernet  HWaddr 00:E0:4C:68:00:0C
          inet addr:169.254.170.254  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

bb-qq, 2024-06-23:
Did you manually start the driver package from the Package Center?

willng-max, 2024-06-24:
[image]
[image]
Yes, I also tried stopping & starting the driver package again

bb-qq, 2024-09-22:
Could you attach the output of `dmesg`?

closed 2025-01-02 by bb-qq