bb-qq/r8152 #408: r8152-2.18.1-1 fails to build on kernel 6.9.x

torehl, 2024-06-28:
Hi,

    tried building the latest driver on 6.9.  Patch available?

root@fedora39vm:/usr/src/r8152-2.18.1-1# dkms install -m r8152 -v 2.18.1-1 -k `uname -r` 
Sign command: /lib/modules/6.9.7-100.fc39.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Creating symlink /var/lib/dkms/r8152/2.18.1-1/source -> /usr/src/r8152-2.18.1-1

Building module:
Cleaning build area...
Building module(s).....(bad exit status: 2)
Failed command:
make -j12 KERNELRELEASE=6.9.7-100.fc39.x86_64 -C /lib/modules/6.9.7-100.fc39.x86_64/build M=/var/lib/dkms/r8152/2.18.1-1/build
Error! Bad return status for module build on kernel: 6.9.7-100.fc39.x86_64 (x86_64)
Consult /var/lib/dkms/r8152/2.18.1-1/build/make.log for more information.

and from make.log

root@srl-torel01:/usr/src/r8152-2.18.1-1# cat /var/lib/dkms/r8152/2.18.1-1/build/make.log
DKMS make.log for r8152-2.18.1-1 for kernel 6.9.7-100.fc39.x86_64 (x86_64)
Fri Jun 28 10:04:46 AM CEST 2024
make: Entering directory '/usr/src/kernels/6.9.7-100.fc39.x86_64'
  CC [M]  /var/lib/dkms/r8152/2.18.1-1/build/r8152.o
/var/lib/dkms/r8152/2.18.1-1/build/r8152.c: In function ‘rtl8152_get_drvinfo’:
/var/lib/dkms/r8152/2.18.1-1/build/r8152.c:22803:9: error: implicit declaration of function ‘strlcpy’; did you mean ‘strncpy’? [-Werror=implicit-function-declaration]
22803 |         strlcpy(info->driver, MODULENAME, sizeof(info->driver));
      |         ^~~~~~~
      |         strncpy
/var/lib/dkms/r8152/2.18.1-1/build/r8152.c: At top level:
/var/lib/dkms/r8152/2.18.1-1/build/r8152.c:23822:20: error: initialization of ‘int (*)(struct net_device *, struct ethtool_keee *)’ from incompatible pointer type ‘int (*)(struct net_device *, struct ethtool_eee *)’ [-Werror=incompatible-pointer-types]
23822 |         .get_eee = rtl_ethtool_get_eee,
      |                    ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/r8152/2.18.1-1/build/r8152.c:23822:20: note: (near initialization for ‘ops.get_eee’)
/var/lib/dkms/r8152/2.18.1-1/build/r8152.c:23823:20: error: initialization of ‘int (*)(struct net_device *, struct ethtool_keee *)’ from incompatible pointer type ‘int (*)(struct net_device *, struct ethtool_eee *)’ [-Werror=incompatible-pointer-types]
23823 |         .set_eee = rtl_ethtool_set_eee,
      |                    ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/r8152/2.18.1-1/build/r8152.c:23823:20: note: (near initialization for ‘ops.set_eee’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:244: /var/lib/dkms/r8152/2.18.1-1/build/r8152.o] Error 1
make[1]: *** [/usr/src/kernels/6.9.7-100.fc39.x86_64/Makefile:1930: /var/lib/dkms/r8152/2.18.1-1/build] Error 2
make: *** [Makefile:252: __sub-make] Error 2
make: Leaving directory '/usr/src/kernels/6.9.7-100.fc39.x86_64'

torehl, 2024-06-28:
https://bbs.archlinux.org/viewtopic.php?id=296912

bb-qq, 2024-09-22:
The source code for this driver is based on that released by Realtek. At the moment I have no plans to make compatible with the new kernel by myself.

bb-qq, 2025-01-02:
Since Synology is still using the version4 series kernel, I have no plans to do anything to support it. The driver source code itself has been updated recently, so the problem might be resolved.

closed 2025-01-02 by bb-qq