bb-qq/r8152 #411: DS1520+ with TRENDnet TUC-ET2G

mikeswanson, 2024-07-15:
Confirming that the TRENDnet TUC-ET2G USB-C to RJ-45 Ethernet Adapter is working while plugged-in to the back of my Synology DS1520+. Repeated 2GB file copies result in ~270 MB/sec transfer speeds.

The TUC-ET2G adapter comes with a USB-C to USB-A adapter that must be connected in the correct orientation. Even though there's a very small notch on the connector, it's easy to miss, and if you connect it backwards, you'll only be able to achieve a 1G connection.

### Description of your products

Synology DS1520+
DSM 7.2.1-69057
3 Seagate IronWolf Pro 16TB drives
SSD read cache only (no write cache)

### Description of your environment

Connected from PC to TRENDnet TEG-S380 8-Port Unmanaged 2.5G Switch
Cat 5e cables
Enable Jumbo Frame, the MTU value is 9000
Reported 2500 Mbps, Full duplex connection

mikeswanson, 2024-07-20:
Just a quick update that it's been 5 days, and the connection has remained stable at 2500 Mbps through multiple cloud backups and many multi-GB file transfers.

mikeswanson, 2024-07-27:
One final note that after two weeks, the connection has remained stable and without issue.

gearboxsg, 2024-10-18:
I had to thank you LOL this was the issue with the ugreen ethernet adapter too.. I had to flip it around because I was using a usb C adapter too.. thank you so much.. it was frustrating me for the better part of an hour

mikeswanson, 2024-10-18:
Glad you were able to figure it out, @gearboxsg!

While I'm here, I'll report that this connection has remained rock solid ever since I added the adapter. I've had no issues at all.