bb-qq/r8152 #414: Please help me resolve the package service execution failure on Synology DS923+.

rlawnsdud117, 2024-07-22:
### Description of the problem
[image]

> I accidentally deleted the /var/packages/r8152 folder and /opt/sbin/spk_su during an overloaded driver reinstall process 1 day ago, and now I'm getting this message.
> I accidentally deleted the /var/packages/r8152 folder and /opt/sbin/spk_su while reinstalling drivers a day ago.
> When I tried to restart the package, I received a 'package service execution failure' message.

### Description of your products

> - the product name of your NAS model
>     - and output of ``uname -a`` command
>      Linux SpringTreeNAS 4.4.302+ #69057 SMP Fri Jan 12 17:01:14 CST 2024 x86_64 GNU/Linux synology_r1000_923+
>     - DSM version 7.2
> - the product name of the ethernet dongle :  EFM ipTIME U25G 
> [image]
### Description of your environment

> - connect directly to a PC or using a Ethernet switch
>     - PC: NIC model name or chipset name : Realtek Gaming 2.5GbE Family Controller 
>     - Ethernet switch: model name : NEXT-7255GH
> - cable category (Cat.5e or Cat.6 or Cat.7) : Cat. 6 , cat 7

### Output of ``dmesg`` command

````
(paste output here, or attach as a file)
[dmesg.txt](https://github.com/user-attachments/files/16330283/dmesg.txt)
````

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:06:00.3) hub
  |__1-1         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 65002B971F238D11)
  |__1-3         05e3:0610:0656 09  2.10  480MBit/s 100mA 1IF  (GenesysLogic USB2.1 Hub ffffffd6ffffffa3ffffffebffffffcb) hub
    |__1-3.3     0bda:8156:3104 00  2.10  480MBit/s 200mA 2IFs (Realtek USB 10/100/1G/2.5G LAN 401300E04C68008D)
    |__1-3.4     051d:0002:0106 00  2.00   12MBit/s 100mA 1IF  (American Power Conversion Back-UPS BX950MI  FW:295201G -302201G  9B2113A30468)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:06:00.3) hub
````

### Output of ``ifconfig -a`` command

````
docker0   Link encap:Ethernet  HWaddr 02:42:77:7F:CB:00
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          inet6 addr: fe80::42:77ff:fe7f:cb00/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:210487 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1533667 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:3071245692 (2.8 GiB)  TX bytes:3190255767 (2.9 GiB)

docker-2b Link encap:Ethernet  HWaddr 02:42:DF:FF:2E:BD
          inet addr:172.18.0.1  Bcast:172.18.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

docker-2f Link encap:Ethernet  HWaddr 02:42:72:CA:E8:10
          inet addr:172.21.0.1  Bcast:172.21.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

docker236 Link encap:Ethernet  HWaddr 92:D1:34:B3:80:13
          inet6 addr: fe80::90d1:34ff:feb3:8013/64 Scope:Link
[... 116 lines cut ...]
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

ovs_eth1  Link encap:Ethernet  HWaddr 90:09:D0:32:6E:CB
          inet addr:192.168.50.161  Bcast:192.168.50.255  Mask:255.255.255.0
          inet6 addr: fe80::9209:d0ff:fe32:6ecb/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1596639 errors:0 dropped:0 overruns:0 frame:0
          TX packets:305413 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:320541904 (305.6 MiB)  TX bytes:3106097706 (2.8 GiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          inet addr:10.8.0.8  P-t-P:10.8.0.8  Mask:255.255.255.0
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:18 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500
          RX bytes:864 (864.0 B)  TX bytes:0 (0.0 B)````

transformcube, 2024-09-11:
The CPU of DS923+ is AMD Ryzen R1600, the driver must be compiled for this CPU, otherwise it will not work.

bb-qq, 2024-09-18:
Did you try full re-installation?
https://github.com/bb-qq/r8152/wiki/Troubleshooting#uninstall-completely--full-reinstallation

If your issue is not resolved, please submit logs described below:
https://github.com/bb-qq/r8152/wiki/Troubleshooting#how-to-investigate-if-the-driver-does-not-work-properly

bb-qq, 2025-01-02:
Since there has been no response for a long time, this issue is closed. Please feel free to re-open it.

closed 2025-01-02 by bb-qq