bb-qq/r8152 #415: The hub crashes during a transfer of more than 797Mo

Matthieu7777, 2024-07-24:
### Description of the problem

> I have a NAS DS218play on DSM 7.2.1-69057 Update 5 and I installed the driver r8152-rtd1296-2.18.1-1_7.2.spk. The new LAN shows up as 2500 Mbps, Full duplex, MTU 1500.
LAN 1 (originally at 100 Mb/s) and LAN 2 (the new one) do not appear as LAN 1 and 2 but as LAN and LAN. However, it works, and access from a device works well at a higher speed (I haven't done any specific measurements yet, but a small file copies about 5 times faster than before. That's pretty good).

> My problem is that as soon as I try to transfer a somewhat large file, the LAN 2 network interface crashes, and I have to fall back on LAN 1.
I was able to transfer small files, but as soon as they are larger, it doesn't work and causes the connection to crash.
700Mo OK
913Mo not OK
1.3Go not OK
Basically, it seems to crash starting at 797Mo. At this point, the transfer stalls, and the network interface crashes.

> The driver is still running, and LAN 2 as well, but the NAS is no longer visible on the local network, and I have to unplug and replug the cable to see it again.

### Description of your products

> NAS Synology DS218play
> Linux Nas 4.4.302+ #69057 SMP Fri Jan 12 17:01:55 CST 2024 aarch64 GNU/Linux synology_rtd1296_ds218play
> DSM 7.2.1-69057 Update 5
> UGREEN 2.5G USB 3.0 to RJ45 Ethernet Network Adapter 2.5 Gigabit USB LAN Plug Play [(https://www.amazon.fr/dp/B0CWV2Q6HJ?psc=1&ref=ppx_yo2ov_dt_b_product_details)](https://amzn.eu/d/04TRxhOA)

### Description of your environment

> Connected via CAT8 Ethernet UGreen to an EERO Pro E6 Wifi router at 2.5Gb/s
> MacBook Air M2
> Eero Pro E6
> Cat8

### Output of ``dmesg`` command

````
[### Output of ``dmesg`` command.txt](https://github.com/user-attachments/files/16360263/Output.of.dmesg.command.txt)
````

### Output of ``lsusb`` command

````
[### Output of ``lsusb`` command.txt](https://github.com/user-attachments/files/16360280/Output.of.lsusb.command.txt)
````

### Output of ``ifconfig -a`` command

````
[### Output of ``ifconfig -a`` command.txt](https://github.com/user-attachments/files/16360272/Output.of.ifconfig.-a.command.txt)
````

rlawnsdud117, 2024-07-27:
Similarly, I can't send more than 700 pages.

Matthieu7777, 2024-07-27:
> Similarly, I can't send more than 700 mb.

If you find a solution please come back here to explain it

rlawnsdud117, 2024-07-28:
> > Similarly, I can't send more than 700 mb.
> 
> If you find a solution please come back here to explain it. What Nas model have you got ?

I am using a Synology DS923+. I don’t have a solution either.

arigon, 2024-07-28:
I've experienced the same problem with the same UGREEN adapter with the DS216+II. It will crash the whole USB stack until restart of the NAS. I'm replacing it with the DS224+ anyway, so I will give it another try - maybe it's the UGREEN.
I tested these versions: 2.18.1-1, 2.17.1-2, 2.17.1-1

Here some DMESG logs when trying to replugin the UPS USB cable (USB ports on the back):
[image]

rlawnsdud117, 2024-07-29:
On the DS923+, I am using a BasicGear 4-port USB hub and connecting an EFM ipTIME U25G with a Realtek RTL8156B chipset. 
I attempted to use version 2.18.1-1, but encountered installation issues. I tested the following versions: 2.18.1-1, 2.17.1-2, 2.17.1-1. 
The rear ports are connected through the BasicGear 4-port USB hub, with both the UPS cable and the EFM ipTIME U25G connected.

[image]

When uploading files larger than 700 MB, the LAN connection gets disconnected.

Matthieu7777, 2024-07-29:
Hello, and thank you for your feedback. In a way it reassures me because I thought it came from my Nas which has only one gigabyte of RAM. I thought it could be because the flash memory was filled and the transfer stopped from the moment the memory was filled. A priori, if you use Nas of different models, the problem should not come from there. This potentially means that an update could overcome this concern.

Matthieu7777, 2024-07-30:
Curiously yesterday, I was able to transfer from my MacBook a Files of one, three gigabytes without worry, even if the transfer froze for a moment around 700, 800 mega to then continue at a relatively fast speed

rlawnsdud117, 2024-07-31:
> Curiously yesterday, I was able to transfer from my MacBook a Files of one, three gigabytes without worry, even if the transfer froze for a moment around 700, 800 mega to then continue at a relatively fast speed

Are you using a powered USB hub by any chance?

Matthieu7777, 2024-07-31:
No it’s a UGREEN 2.5G USB 3.0 to RJ45 Ethernet Network Adapter 2.5 Gigabit USB LAN Plug Play [(https://www.amazon.fr/dp/B0CWV2Q6HJ?psc=1&ref=ppx_yo2ov_dt_b_product_details)](https://amzn.eu/d/04TRxhOA)
But I deleted all the software that was not essential and that I could delete. Maybe it freed up a little memory. It seems to me that in the activity monitor, this is not the case.
Maybe that's why I was able to transfer larger files than before. I don't know.

rlawnsdud117, 2024-08-01:
> No it’s a UGREEN 2.5G USB 3.0 to RJ45 Ethernet Network Adapter 2.5 Gigabit USB LAN Plug Play [...]

Thank you!

Matthieu7777, 2024-08-05:
Just to complete the description of my problem, for a few days, I have had Plex crashes on my Apple TV repeatedly. Up to a crash every five minutes for example. Each time I had to unplug and reconnect the port. 2.5Gb Ethernet. I don't know if it helps to understand the problem, but at least it may give some information.

Matthieu7777, 2024-08-16:
Update.

My network interface keeps crashing so I had to keep my interface at 100k/s to keep a stable connection. It crashes when I make a transfer, when I read a video on Plex or even for no particular reason. It can crash several times when playing an episode of a TV series or a movie. If you have any ideas it would help me.

transformcube, 2024-09-11:
Try enable jumbo frame (MTU 9000), see it work or not.

Matthieu7777, 2024-09-11:
I tried but it seemed to be relatively unstable. But I may not have insisted long enough.

Why do you suggest that solution?

Matthieu7777, 2024-09-11:
I just activated this option. I'll try to wait a little longer to see if it stabilizes. And if it solves the problem. In any case, thank you for your comment.

Matthieu7777, 2024-09-11:
I tried and it crashes even faster I have the impression. I will leave this setting in place for the moment and see if it changes anything.

Sjekke, 2024-09-11:
Every device must support MTU9000 or it has not sense doing it.

Matthieu7777, 2024-09-12:
I had tried this version before installing the more recent version in beta for the same results unfortunately. I will try again to see if it stabilizes the situation. But a priori this was not the case.

Matthieu7777, 2024-09-12:
I had version 2.18.1-1 and I couldn't install version 2.17.1-2 so I decided to uninstall the driver to be able to install 2.17.1-2 again.

Unfortunately, I had forgotten that I have the updated version of DSM 7.2.2-72803. Following this latter, update, my driver still worked, but now that I have uninstalled, it is impossible to reinstall it, regardless of the version of the driver I choose.

I should never have tried this method. Do you know if a version of the driver compatible with this version of the DSM will be put online? And if so, on what date do you think?

transformcube, 2024-09-13:
[https://emby.media/community/index.php?/topic/26325-best-way-to-cleanly-and-completely-uninstall/#comment-259010](url)
using ssh to manual uninstall package:
1. Navigate to /volume1/@appstore then delete the application you want to remove.
2. Navigate to /var/packages see if there is a folder named the same as your app and remove it
3. Navigate to /usr/local see if there is a folder named the same as your app and remove it

Matthieu7777, 2024-09-18:
I don't know how to access to this folders and delete it (if here).
I tried to restart the command line installation but it doesn't work anymore.

bb-qq, 2024-09-18:
Unfortunately, some users of rtd1296 based products reported stability issues. For details, please refer this issue:
https://github.com/bb-qq/r8152/issues/275

If you are using a recent product that does not use rtd1296, the cause may be different. If you can create a separate Issue and attach the logs to it, I would like to investigate.

Matthieu7777, 2024-09-18:
I found the problem for the reinstallation, I used rtd1619b but it didn't work. I used rtd1619 again (without the B) and I was able to reinstall the driver.

So I reinstalled the r8152-rtd1619-2.17.1-2_7.2.spk version of the driver avoiding the next beta version and I find my additional network interface but the stability problems are still present unfortunately.

I still haven't solved the problem at the origin of this post so if you have other ideas, I'm interested.

Matthieu7777, 2024-09-19:
I understood that there was a script to relaunch the driver in case of a bug. I can't find it or know how to install it. Can anyone answer me and explain the procedure?

bb-qq, 2024-12-31:
It should work stably on the rtd1619 platform. Have you read [the troubleshooting page](https://github.com/bb-qq/r8152/wiki/Troubleshooting)?

Gordemm77, 2025-02-24:
Are there any news concerning this issue? I do have the same problems with my DS218play. I'm able to send data to the NAS, but as soon as I try to receive data from my NAS the network connection crashes

olib937, 2025-03-27:
Hi @bb-qq got the same exact same issue on a DS218play with an ASUS WL-USB USB-C2500 adapter.
I've read the troubleshooting page but did not find the solution... is there any fix for this ? Thx !

bb-qq, 2025-06-07:
Since I haven't heard back from the owner of this issue for a while, I am closing it for now. Feel free to reopen it if you need to.

For stability issues with DS218play, please see #275.

closed 2025-06-07 by bb-qq