bb-qq/r8152 #419: DS1513+ and Asus 2500 is slow

cupanther, 2024-08-08:
I have a DS1513+ and can only get 1.2G one way and 1G the other way. I am using the ASUS USB-C 2500 adapter. I have verified the USB port accepts 5G, set the MTU to the recommended setting, verified SMB 3 is enabled and checked that the ethernet adapter reports it can do 2.5G. The adapter is plugged directly into the left USB3 port, but have also tried the other one and no difference. Is there anything else I need to check or change?

transformcube, 2024-09-09:
USB Lan Dongle performence mainly affected by CPU, the low network transfer rate due to Intel Atom D2700 not powerful enough to push USB-C 2500 chips.  Just like pairing an RTX 4090 with a $47 CPU
https://www.youtube.com/watch?v=pEQ8bD8l3b0

bb-qq, 2024-09-22:
USB performance does not seem to be sufficient for older platforms. Perhaps USB-connected HDDs are about the same speed.