bb-qq/r8152 #422: Unable to use multiple driver versions with DS920+ and US2GA30

bikerjake, 2024-08-25:
### Description of the problem

I am having issues getting my DS920+ running with multiple drivers after following the installation steps (including running the command) and rebooted after.

After installation / reboot, when I try to run: r8152-geminilake-2.17.1-1_7.1.spk it fails with a message 'Failed to run the package service'. 

I have also tried older versions such as
* r8152-geminilake-2.15.0-4.spk
* r8152-geminilake-2.15.0-5.spk
* r8152-geminilake-2.15.0-6.spk
Which succeed to install, reboot NAS, manually start (then shows running, but about 10/15 seconds later stops).

### Description of your products

> - the product name of your NAS model: DS920+
>     - and output of ``uname -a`` command:
`Linux SynologyHome 4.4.302+ #69057 SMP Fri Jan 12 17:02:28 CST 2024 x86_64 GNU/Linux synology_geminilake_920+`
>     - DSM version 7.2.1-69057
> - the product name of the ethernet dongle: StarTech US2GA30

### Description of your environment

### Output of ``dmesg`` command

````
[dmesgOutput.txt](https://github.com/user-attachments/files/16740646/dmesgOutput.txt)

````

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F000CDE8DEFD613)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         0bda:8156:3104 00  3.20 5000MBit/s 256mA 2IFs (Realtek USB 10/100/1G/2.5G LAN 4013000001)

````

### Output of ``ifconfig -a`` command

````
(paste output here)
````

bikerjake, 2024-08-25:
Update on this: I managed to get this working with the following driver ( r8152-geminilake-2.16.3-4_7.2.spk ) after seeing a Reddit post ( https://www.reddit.com/r/synology/comments/14ynhht/need_help_plugable_25g_usb_c_and_usb_to_ethernet/ ) with someone having the same issue with the driver running then stopping.
This is now working with my StarTech US2GA30 (Note this is the USB A version NOT C).

bb-qq, 2024-09-19:
Could you attach logs described here?
https://github.com/bb-qq/r8152/wiki/Troubleshooting#how-to-investigate-if-the-driver-does-not-work-properly

bb-qq, 2025-01-02:
I am closing this issue as the problem seems to be resolved.

closed 2025-01-02 by bb-qq