bb-qq/r8152 #424: Not work with 7.2.2

dateno1, 2024-08-31:
### Description of the problem

> Replace this line with your answer.

### Description of your products

> - the product name of your NAS model
>     - and output of ``uname -a`` command
>     - DSM version x.x
> - the product name of the ethernet dongle
Model : DS918+ (Using ARC bootloader 24.08.27)
Hardware : VC65R-G039M (i5-6500T)
Version : 7.2.2
Kernel : Linux Test 4.4.302+ #72803 SMP Mon Aug 19 19:59:09 CST 2024 x86_64 GNU/Linux synology_apollolake_918+

### Description of your environment

> - connect directly to a PC or using a Ethernet switch
>     - PC: NIC model name or chipset name
>     - Ethernet switch: model name
> - cable category (Cat.5e or Cat.6 or Cat.7)

NIC : RTL8156
No switch (1:1 Direct Connected)
Cable : CAT.6

### Output of ``dmesg`` command

````
[dmesg.txt](https://github.com/user-attachments/files/16823841/dmesg.txt)

````

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:14.0) hub
  |__1-2         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (JetFlash Mass Storage Device B8X4DF3X)
  |__1-7         0bda:0129:3960 ff  2.00  480MBit/s 500mA 1IF  (Generic USB2.0-CRW 20100201396000000)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:14.0) hub
  |__2-1         0bda:8156:3000 00  3.20 5000MBit/s 512mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 000000001)
  |__2-5         05e3:0626:0656 09  3.20 5000MBit/s 0mA 1IF  (GenesysLogic USB3.1 Hub ffffffd6ffffffa3ffffffebffffffcb) hub

````

### Output of ``ifconfig -a`` command

````
eth0      Link encap:Ethernet  HWaddr 78:24:AF:04:42:B1
          inet addr:192.168.12.166  Bcast:192.168.12.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:16201 errors:0 dropped:0 overruns:0 frame:0
          TX packets:15237 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:16870583 (16.0 MiB)  TX bytes:8323641 (7.9 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:6348 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6348 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:852040 (832.0 KiB)  TX bytes:852040 (832.0 KiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

All work well with 7.2.1 but not work with 7.2.2

[image]

guandms, 2024-09-04:
I also have the same issue; my new ASUSTOR AS-U2.5G was installed correctly, followed every step, and the speed is limited to 1000MB.
I'm also on 7.2.2 :(

transformcube, 2024-09-11:
Have you tried the r8152-apollolake-2.17.1-2_7.2.spk specific DSM7.2 driver?

dateno1, 2024-09-11:
> Have you tried the r8152-apollolake-2.17.1-2_7.2.spk specific DSM7.2 driver?

Tested with same version (lastest)

bb-qq, 2024-09-18:
Did you try full re-installation?
https://github.com/bb-qq/r8152/wiki/Troubleshooting#uninstall-completely--full-reinstallation

If your issue is not resolved, please submit logs described below:
https://github.com/bb-qq/r8152/wiki/Troubleshooting#how-to-investigate-if-the-driver-does-not-work-properly

dateno1, 2024-09-18:
ReInstallation not fix problem

It fixed but your driver removed&Bootloader's Driver enabled

bb-qq, 2024-09-18:
Does this mean that the r8152 driver is also present on the bootloader side? If so, there may be a conflict problem.

dateno1, 2024-09-19:
> Does this mean that the r8152 driver is also present on the bootloader side? If so, there may be a conflict problem.

1. At first time that not loaded (but Package not work)

2. Enable driver in bootloader error removed on Package Center

3. Remove Package and test RTL8156 (work well)

yvanliang, 2024-09-19:
The same issue on DS920. First I upgraded DSM from 7.2.1 to 7.2.2 (with 2.16 driver), the driver works well. Then I tried to upgrade the driver to 2.18, it failed to start.

bb-qq, 2024-09-22:
It appears that the `cdc_ncm` driver must be disabled in order to use the r8152 driver with ARPL. See here for details.
https://github.com/bb-qq/r8152/wiki/Troubleshooting#hot-plugging-does-not-work

bb-qq, 2024-11-23:
I can't think of anything more I can do, so I'm closing this issue. Feel free to re-open it if you have any questions.

closed 2024-11-23 by bb-qq