bb-qq/r8152 #426: ds918+ DSM 7.2.1 U5 - driver installation failed

fra-wwilhelm, 2024-09-06:
### Description of the problem

DSM gives an error mesage that driver installation / repair failed
[r8152-apollolake-2.17.1-2_7.2.spk](https://github.com/bb-qq/r8152/releases/download/2.17.1-2/r8152-apollolake-2.17.1-2_7.2.spk)

### Description of your products
´´´´
ds918+ / apollolake
DSM 7.2.1 U5
Linux homeland 4.4.302+ #69057 SMP Fri Jan 12 17:02:59 CST 2024 x86_64 GNU/Linux synology_apollolake_918+
uni USB Ethernet Adapter 2,5G, USB 3.0
´´´´
https://www.amazon.de/uni-Ethernet-treiberfreier-Netzwerkadapter-Chromebook-Grau/dp/B0CZDDTDDT/ref=sr_1_3?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=AITBQEE08LMX&dib=eyJ2IjoiMSJ9.LwwIUVBlHK3yVAHCW8tcvJhAPrlZN4ieeIw7BkFj8PT-7i0uaj79_xab_g_ge82mM32R2HposXOkprQWqbYbdHjtQtThzSMvWaKZ4b79Q49I5UWynaqlFxW_FyKwhaWkIZXn1DfxtN8ULLPcpNo9PvxY7OOa7hKxnMcY_aBHxB6H7dAMBILu2eK6e_040nE0ezifltnQVY1ZJkReDIsYUpcWJW2-x4A9Mgd4PMKzbbc.Pc4ElTKNq6FGOrKo6MMD5Ki_7CFOh040uurdBWXm6UU&dib_tag=se&keywords=uni+ethernet+2.5&qid=1725604167&sprefix=uni+ethernet+2.5%2Caps%2C99&sr=8-3

### Description of your environment

connected via switch
Cat5e

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F0097EB92E5F113)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         0bda:8156:3104 00  3.20 5000MBit/s 256mA 2IFs (Realtek USB 10/100/1G/2.5G LAN 401000001)

````
````
ll /opt/sbin/spk_su
-rwsr-xr-x 1 root root 20952 Sep  5 17:15 /opt/sbin/spk_su
````

````
Sep 05 17:54:26 homeland synoscgi_SYNO.Core.Package.Installation_1_upgrade[19885]: SYNOPKGMGR: Failed to install package, spk=[/volume1/@tmp/upload_tmp.194240]
 result=[{"action":"repair","beta":false,"betaIncoming":false,"broken_by":"install_corruption","error":{"code":278,"description":"\"failed to run postinst script\""},
"finished":true,"installReboot":false,"installing":true,"language":"enu","last_stage":"start","package":"r8152",
"packageName":"RTL8152/RTL8153 driver","pid":19885,
"scripts":[{"code":0,"message":"","type":"preupgrade"},{"code":0,"message":"","type":"preuninst"},
{"code":0,"message":"","type":"postuninst"},{"code":0,"message":"","type":"preinst"},
{"code":5,"message":"","type":"postinst"}],"spk":"/volume1/@tmp/upload_tmp.194240","stage":"install_failed_and_start_failed","status":"broken","status_code":150,"status_description":"failed to complete installation"
,"success":false,"username":"XXXXX","version":"2.17.1-2"}]

````

bb-qq, 2024-09-19:
```
{"code":5,"message":"","type":"postinst"}],"spk":"/volume1/@tmp/upload_tmp.194240","stage":"install_failed_and_start_failed","status":"broken","status_code":150,"status_description":"failed to complete installation"
,"success":false,"username":"XXXXX","version":"2.17.1-2"}]
```
The above line indicates below condition is a failure since the error status is `5`,
```
    if [ -x "${spk_su}" ]; then
```

but its permission seems fine as you pointed out.  It is wired...
```
ll /opt/sbin/spk_su
-rwsr-xr-x 1 root root 20952 Sep  5 17:15 /opt/sbin/spk_su
```

Could you check the permission of parent directories?

fra-wwilhelm, 2024-09-19:
Thank you for the feedback,

indeed, the permissions of the parent directory were wrong.
After fixing this install went as expected, adapter is recognized and driver starts in DSM.

Thanks, toopic can be closed,

closed 2024-09-19 by fra-wwilhelm

bruduju, 2025-05-23:
thanks, 
I'm having the same issue. Which should be the permissions of the parent directory and how to fix them please ?

streithausen, 2025-05-23:
````
#ls -ld /opt/sbin/
drwxr-xr-x 2 root root 4096 Sep  5  2024 /opt/sbin/
#ls -ld /opt/
drwx--x--x 4 root root 4096 Sep  5  2024 /opt/
````
chmod is your friend: 
````
chmod 711 /opt
chmod 755 /opt/sbin
````