bb-qq/r8152 #431: DS218+ DSM 7.2.2 | MTU 9000 | CLUB 3D CAC-1420 (RTL8156/Type-A) - Wake on LAN

mcury1985, 2024-09-20:
Hi, everything is working perfectly and with MTU 9000 and SMB3, I can reach up to 180MB/s and the speed doesn't drop anymore.

I would just like to ask about the WOL feature, any one of you guys know how to activate it for the USB NIC ?

Thanks very much for this driver, it saved me and now I can keep my NAS for much longer !! :)

bb-qq, 2024-10-05:
Unfortunately, WoL does not appear to be supported.

closed 2025-01-13 by bb-qq

henkbart, 2025-08-30:
No, it is a proven seolution. And it still hybernates!!!

Sjekke, 2025-08-31:
Thx @henkbart it's seems to be working, nevertheless the hibernate doesn't start after 10min like programmed but later ...

henkbart, 2025-08-31:
Hello @Sjekke .
To hybernate the Synology, you have to check your complete network traffic.
If there is any communication to the Nas , it don't Hybernate.
I have 9 Synology NAS machines and they all go into hybernation (Hybernations is set to 10 minutes).
But i had to check my network before i got this.
Any app of programm checking if the NAS is present, breaks the hybernation timer.
Henk