bb-qq/r8152 #432: Problem installing R1000 driver on my new DS923+ running DSM 7.2.2-72806

bill01568, 2024-09-21:
Hi,

I had good luck installing one of the drivers on my older Synology NAS and successfully getting 2.5gbe throughput, so I was looking forward to an easy install/setup of the appropriate driver on my newly purchased DS923+. Alas, I encountered an issue at the first step.

Checking the Synology page you linked (thank you!) it shows that the DS923+ uses AMD Ryzen R1600, so I should retrieve the driver for architecture R1000.

Since my DS923+ is running DSM 7.2.2-72806, I downloaded r8152-v1000-2.18.1-1_7.2.spk.

When I used package manager to try to install that driver manually, it immediately said "This package is not supported on the platform of Synology NAS or is incompatible with the current DSM version."

What am I doing wrong?

Thanks,
Bill

bb-qq, 2024-09-22:
Please try the driver for `r1000`.
https://github.com/bb-qq/r8152/releases/download/2.18.1-1/r8152-r1000-2.18.1-1_7.2.spk

bill01568, 2024-09-22:
Much better! Thank you!

quantvol, 2026-03-01:
Any luck on the DS723+? I'm in the same boat, 1Gbps only

TheS1R, 2026-03-01:
> Thanks for your reply. Are you using the UGREEN or another adapter?

wisdPi 5 Gbps USB NIC and Synology 10 Gbps module (_E10G22-T1-Mini Rj45_).