bb-qq/r8152 #443: Unable to configure MTU above 1500 using 2.18.1-2 pre-release in the GUI

Wysie, 2024-10-11:
### Description of the problem

Unable to configure MTU above 1500 using 2.18.1-2 pre-release in the GUI

`/bin/get_key_value /etc.defaults/synoinfo.conf supportMTU
`
returns a yes value.

Running this in SSH seems to fix it though (if I check via ifconfig -a):

`sudo ifconfig eth2 mtu 9000
`

### Description of your products
Model(platform): DS420+ (geminilake)
DSM version: DSM 7.2.1-69057 Update 5
Driver version: 2.18.1-2 pre-release
Ethernet device: WisdPi WP-UT5 5Gbps (RTL8157/Type-A)

### Description of your environment

> - connect directly to a PC or using a Ethernet switch
>     - PC: NIC model name or chipset name
>     - Ethernet switch: model name
> - cable category (Cat.5e or Cat.6 or Cat.7)

bb-qq, 2024-12-31:
Is the situation the same with the latest version?
https://github.com/bb-qq/r8152/releases/tag/2.19.2-1

DKWien, 2025-01-12:
I am using Synology 220+ with 
- USB 5GbE adapter WP-UT5 RTL8157
- driver 2.19.2-1
- DSM 7.2.2-72806 Update 2

before changing to MTU 9000: 

[image]

then:

`sudo ifconfig eth2 mtu 9000`

after:

[image]

some info with screenshots:

[image]

still I get :  **Speed: Unknown!** :

[image]

bb-qq, 2025-02-16:
At this time, the link speed is not displayed correctly due to a limitation of the Realtek driver. However, if the link speed is Unknown, it means that the link is running at 5Gbps.

I will try to fix this when 5Gbps adapters become available in my country.

fabio1605, 2025-03-24:
to add same issue with me - DS916 - Unable to go above 1500 MTU

However the sudo ifconfig eth2 mtu 9000  does also work for me

My information is available on #503  if you want more specs out of my data

bb-qq, 2025-06-08:
I can set the MTU to 9000 from the GUI on my DS918+ without any problems. I think the contents of `synoinfo.conf` may be related to this issue, but I'm not sure. Could someone please edit the contents and investigate?