bb-qq/r8152 #447: Failed to run postinst script for r8152, 5 with r8152-r1000-2.18.1-1_7.2

Poohbear2024, 2024-10-18:
### Description of the problem
r8152-r1000-2.18.1-1_7.2 installation failed with error 5

synopkg.log:2024/10/18 17:33:00 install r8152 2.18.1-2 Begin postinst
synopkg.log:2024/10/18 17:33:00 install r8152 2.18.1-2 End postinst ret=[5]
synopkg.log:2024/10/18 17:33:00 install r8152 2.18.1-2 failed

### Description of your products

DS923+
Linux Poohbear101 4.4.302+ #72806 SMP Thu Sep 5 13:41:01 CST 2024 x86_64 GNU/Linux synology_r1000_923+
DSM 7.2.2-72806

Unbranded Realtek USB 10/100/1G/2.5G LAN - Realtek Semiconductor Corp.

### issue Description

Unable to install r8152-r1000-2.18.1-1_7.2 drive and Synology driver showed error.

synopkgmgr.log:2024/10/18 11:06:19 Failed to run postinst script for r8152, 5
synopkgmgr.log:2024/10/18 11:06:19 Failed to install package, spk=[/volume1/@tmp/upload_tmp.119400] result=[{"action":"install","beta":false,"betaIncoming":false,
"broken_by":"install_corruption","error":{"code":276,"description":"failed to run postinst script"},"finished":true,"installReboot":false,"installing":true,"language":"enu","last_stage":"postinstall","package":"r8152","packageName":"RTL8152/RTL8153 driver","pid":13440,"scripts":[{"code":0,"message":"","type":"preinst"},{"code":5,"message":"","type":"postinst"}],"spk":"/volume1/@tmp/upload_tmp.119400","stage":"install_failed","status":"broken","status_code":150,"status_description":"failed to complete installation","success":false,"username":"xxxx","version":"2.18.1-1"}]

### workaround

Install previous release r8152-r1000-2.16.3-4_7.2, is able to install and running successfully

[putty_github.log](https://github.com/user-attachments/files/17434741/putty_github.log)

bb-qq, 2024-12-31:
Did you trye [Full reinstallation](https://github.com/bb-qq/r8152/wiki/Troubleshooting#uninstall-completely--full-reinstallation)?

Poohbear2024, 2025-01-14:
After following full reinstallation procedure to remove installed package, can successfully installed r8152-r1000-2.19.2-1_7.2 release.

closed 2025-01-14 by Poohbear2024