bb-qq/r8152 #451: Unable to use two (identical) adapters on Synology 1815+ - Only one works

thelowkeyguy, 2024-10-22:
### Description of the problem

I'm getting this error when I tried to insert a second (identical) adapter (I'm using the latest version: r8152-avoton-2.18.1-2_7.1.spk. Before I was using version r8152-avoton-2.17.1-1_7.1.spk with same problem: unable to use both adapters at same time, but with old version I don´t get the kernel dump)

[8974079.889774] usb 3-3: reset SuperSpeed USB device number 2 using etxhci_hcd-170202
[8974080.085655] r8152 3-3:1.0 eth4: v2.18.1 (2024/05/20)
[8974080.091450] r8152 3-3:1.0 eth4: This product is covered by one or more of the following patents:
                 		US6,570,884, US6,115,776, and US6,327,625.

[8974080.109471] r8152 3-3:1.0 eth4: chip rev 14
[8974080.114423] usbcore: registered new interface driver r8152
[8974080.225317] IPv6: ADDRCONF(NETDEV_UP): eth4: link is not ready
[8974081.209562] iSCSI:iscsi_target.c:545:iscsit_add_np CORE[0] - Added Network Portal: 10.0.0.1:3260 on iSCSI/TCP
[8974083.426354] IPv6: ADDRCONF(NETDEV_CHANGE): eth4: link becomes ready
[8974083.434693] r8152 3-3:1.0 eth4: carrier on
[8974083.933678] iSCSI:iscsi_target.c:545:iscsit_add_np CORE[0] - Added Network Portal: [fe80::823f:5dff:fef5:657e]:3260 on iSCSI/TCP
[8974450.354610] usb 2-4: new high-speed USB device number 2 using etxhci_hcd-170202
[8974450.374555] usb 2-4: not running at top speed
[8974453.844576] usb 2-4: new high-speed USB device number 2 using etxhci_hcd-170202
[8974453.864323] usb 2-4: not running at top speed
[8974457.334341] usb 2-4: new high-speed USB device number 2 using etxhci_hcd-170202
[8974457.354096] usb 2-4: not running at top speed
[8974460.824120] usb 2-4: new high-speed USB device number 2 using etxhci_hcd-170202
[8974460.843864] usb 2-4: not running at top speed
[8974463.161042] usb 3-4: new SuperSpeed USB device number 3 using etxhci_hcd-170202
[8974463.187309] drivers/usb/core/ethub.c (2055) Same device found. Change serial to 401000000 
[8974463.200582] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 2 ep 2 with no TDs queued?
[8974463.210446] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 2 ep 3 with no TDs queued?
[8974463.220305] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 2 ep 6 with no TDs queued?
[8974464.334033] usb 3-4: reset SuperSpeed USB device number 3 using etxhci_hcd-170202
[8974464.356649] usb 3-4: device firmware changed
[8974464.361680] usb 3-4: USB disconnect, device number 3
[8974464.386743] r8152 3-4:1.0: read type=0x00cc, index=0xd818 fail -19
[8974464.393853] ------------[ cut here ]------------
[8974464.399220] WARNING: at //source/r8152/r8152.c:1395 ocp_read_word+0x7e/0x90 [r8152]()
[8974464.408181] Modules linked in: r8152(OF) nfnetlink xfrm_user xfrm_algo bridge stp aufs macvlan veth xt_conntrack xt_addrtype nf_conntrack_ipv6 nf_defrag_ipv6 ip6table_filter ip6_tables ipt_MASQUERADE xt_REDIRECT xt_nat fuse iptable_nat nf_nat_ipv4 nf_nat xt_recent xt_iprange xt_limit xt_state xt_tcpudp xt_multiport xt_LOG nf_conntrack_ipv4 nf_defrag_ipv4 nf_conntrack iptable_filter ip_tables x_tables 8021q vhost_scsi(O) vhost(O) tcm_loop(O) iscsi_target_mod(O) target_core_ep(O) target_core_multi_file(O) target_core_file(O) target_core_iblock(O) target_core_mod(O) syno_extent_pool(PO) rodsp_ep(O) vfat fat udf isofs quota_v2 quota_tree synoacl_vfs(O) raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx nfsd btrfs zstd_decompress ecryptfs zstd_compress xxhash raid6_pq xor lockd rpcsec_gss_krb5
[8974464.488280]  auth_rpcgss sunrpc leds_lp3943 aesni_intel glue_helper lrw gf128mul ablk_helper avoton_synobios(PO) hid_generic usbhid hid usblp usb_storage qede(O) qed(O) mlx5_core(O) mlx4_en(O) mlx4_core(O) mlx_compat(O) atlantic_v2(O) atlantic(O) tn40xx(O) i40e(O) ixgbe(O) be2net(O) i2c_algo_bit igb(O) e1000e(O) sg dm_snapshot crc_itu_t crc_ccitt psnap p8022 llc zlib_deflate libcrc32c hfsplus md4 hmac sit tunnel4 ipv6 flashcache_syno(O) flashcache(O) syno_flashcache_control(O) dm_mod crc32c_intel cryptd arc4 sha256_generic sha1_generic ecb aes_x86_64 authenc des_generic ansi_cprng cts md5 cbc cpufreq_powersave cpufreq_performance acpi_cpufreq mperf processor thermal_sys cpufreq_stats freq_table vxlan ip_tunnel zram(C) loop etxhci_hcd xhci_hcd uhci_hcd ehci_pci ehci_hcd usbcore usb_common [last unloaded: r8152]

[8974464.567563] CPU: 1 PID: 9887 Comm: kworker/1:5 Tainted: PF        C O 3.10.108 #42962
[8974464.576519] Hardware name: Insyde MohonPeak/Type2 - Board Product Name1, BIOS M.111 2017/05/09
[8974464.586383] Workqueue: events driver_set_config_work [usbcore]
[8974464.593105]  ffffffff8147dfd1 ffffffff8146f5e1 0000000000000002 ffff8800191f7bcc
[8974464.601609]  ffff88000a778830 ffff880015717a50 ffff880015717000 ffffffffa040b8fe
[8974464.610110]  ffffffed0a778830 ffffffff0a778800 ffff88000a778800 ffff8800157176c0
[8974464.618599] Call Trace:
[8974464.621534]  [<ffffffff8147dfd1>] ? dump_stack+0xc/0x15
[8974464.627577]  [<ffffffff8146f5e1>] ? warn_slowpath_common+0x56/0x6c
[8974464.634692]  [<ffffffffa040b8fe>] ? ocp_read_word+0x7e/0x90 [r8152]
[8974464.641902]  [<ffffffffa04272d2>] ? rtl8152_probe+0x592/0x1020 [r8152]
[8974464.649433]  [<ffffffffa0012a98>] ? usb_probe_interface+0xa8/0x1f0 [usbcore]
[8974464.657517]  [<ffffffff81317a1f>] ? really_probe+0x5f/0x1f0
[8974464.663947]  [<ffffffff81317bb0>] ? really_probe+0x1f0/0x1f0
[8974464.670465]  [<ffffffff81315e2e>] ? bus_for_each_drv+0x5e/0x90
[8974464.677187]  [<ffffffff81317980>] ? device_attach+0x90/0xd0
[8974464.683622]  [<ffffffff81316e88>] ? bus_probe_device+0x88/0xb0
[8974464.690343]  [<ffffffff8131502d>] ? device_add+0x54d/0x710
[8974464.696685]  [<ffffffff811a15c2>] ? sysfs_merge_group+0x52/0xe0
[8974464.703516]  [<ffffffffa0011911>] ? usb_set_configuration+0x5a1/0x8d0 [usbcore]
[8974464.711901]  [<ffffffffa0011cb1>] ? driver_set_config_work+0x71/0xa0 [usbcore]
[8974464.720178]  [<ffffffff8105d974>] ? worker_run_work+0xa4/0xf0
[8974464.726804]  [<ffffffffa0011c40>] ? usb_set_configuration+0x8d0/0x8d0 [usbcore]
[8974464.735179]  [<ffffffff8104f10d>] ? process_one_work+0x14d/0x510
[8974464.742096]  [<ffffffff8104fefd>] ? worker_thread+0x11d/0x450
[8974464.748713]  [<ffffffff8104fde0>] ? manage_workers.isra.36+0x280/0x280
[8974464.756213]  [<ffffffff81055d64>] ? kthread+0xb4/0xc0
[8974464.762062]  [<ffffffff81055cb0>] ? kthread_worker_fn+0x160/0x160
[8974464.769076]  [<ffffffff81498ecd>] ? ret_from_fork+0x5d/0xb0
[8974464.775506]  [<ffffffff81055cb0>] ? kthread_worker_fn+0x160/0x160
[8974464.782511] ---[ end trace 87df794098df881e ]---
[8974466.461681] usb 3-4: new SuperSpeed USB device number 4 using etxhci_hcd-170202
[... 116 lines cut ...]
          TX packets:999 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:14548 (14.2 KiB)  TX bytes:113673 (111.0 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:25894523 errors:0 dropped:0 overruns:0 frame:0
          TX packets:25894523 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:7334692483 (6.8 GiB)  TX bytes:7334692483 (6.8 GiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

Any ideas ? 

Best Regards ! 

Claudio

mr-kek, 2024-11-08:
Essentially having the same problem.
2.5Gb USB, brand pluggable
DS 918+
Linux NAS 4.4.302+ #64570 SMP Thu Jul 20 00:07:29 CST 2023 x86_64 GNU/Linux synology_apollolake_918+
DSM 7.2-64570 Update 3

part of DMESG
[1646514.057977] usb 2-1: new SuperSpeed USB device number 110 using xhci_hcd
[1646514.077967] drivers/usb/core/hub.c (2955) Same device found. Change serial to 000000000
[1646514.218082] usb 2-1: reset SuperSpeed USB device number 110 using xhci_hcd
[1646514.331555] r8152 2-1:1.0 eth3: v2.17.1 (2023/06/13)
[1646514.337360] r8152 2-1:1.0 eth3: This product is covered by one or more of the following patents:
                 		US6,570,884, US6,115,776, and US6,327,625.

[1646514.355381] r8152 2-1:1.0 eth3: chip rev 12
[1646514.360416] r8152 2-1:1.0 eth3: rx_buf_sz will be overridden to 16384

lsusb
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F009CCE0A13F274)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         0bda:8156:3000 00  3.20 5000MBit/s 512mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 000000000)
  |__2-2         0bda:8156:3000 00  3.20 5000MBit/s 512mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 000000001)

bb-qq, 2024-11-13:
Did you check this page?
https://github.com/bb-qq/r8152/wiki/Troubleshooting#multiple-identical-devices-do-not-work

bb-qq, 2024-11-20:
Good. I close this issue.

closed 2024-11-20 by bb-qq