bb-qq/r8152 #452: 8156B adapter would be recognized as 1GbE adapter after BE-88U router restarted. re-enable the driver would fix it

1x2y3z4, 2024-10-27:
### Description of the problem

After all hardware and software installed, the 3rd network interface would display 2500Mb/s; 
But when BE-88U router reinited network interfaces or restarted by applying some network configurations,
the adapter would be in 1000Mb/s in DSM network interfaces UI. 
I would have to disable and enable the driver(2.18.1-1), it would then display as 2500Mb/s again. 
Is there any change to fix the root cause? or could I write a crontab job to disable and enable the driver when it is in 1000Mb/s mode?

### Description of your products

Linux 4.4.180+ #42962 SMP Mon May 29 14:38:23 CST 2023 x86_64 GNU/Linux synology_apollolake_918+
DS918+ (with DSM 7.1.1-42962 Update 6) ;

### Description of your environment
one of Asus BE-88U router 2.5GE port connected to "ULT-unite USB-A to 2.5GE" adapter with CAT-6 cable.
adapter plugged in the back USB port of DS 918+

### Output of ``dmesg`` command

[dmesg.log](https://github.com/user-attachments/files/17533287/dmesg.log)

### Output of ``lsusb`

````
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F0097B90B3A7D34)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-2         0bda:8156:3104 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 4013000001)
````

### Output of ``ifconfig -a`` command

````
docker0   Link encap:Ethernet  HWaddr 02:42:E7:B9:45:6E  
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          inet6 addr: fe80::42:e7ff:feb9:456e/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2521561 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2770069 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:3027443197 (2.8 GiB)  TX bytes:37519219086 (34.9 GiB)

dockera7c Link encap:Ethernet  HWaddr 4E:F7:3E:A0:11:A0  
          inet6 addr: fe80::4cf7:3eff:fea0:11a0/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2521561 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2769775 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:3062745051 (2.8 GiB)  TX bytes:37519176836 (34.9 GiB)

eth0      Link encap:Ethernet  HWaddr 00:11:32:B4:58:99  
          inet addr:192.168.50.112  Bcast:192.168.50.255  Mask:255.255.255.0
          inet6 addr: fe80::211:32ff:feb4:5899/64 Scope:Link
          inet6 addr: 2408:8207:2530:36c0:211:32ff:feb4:5899/64 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:170929 errors:0 dropped:0 overruns:0 frame:0
          TX packets:190719 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:121227456 (115.6 MiB)  TX bytes:144171415 (137.4 MiB)

eth1      Link encap:Ethernet  HWaddr 00:11:32:B4:58:9A  
          inet addr:169.254.172.2  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth2      Link encap:Ethernet  HWaddr 00:E0:4C:68:0B:C5  
          inet addr:192.168.50.111  Bcast:192.168.50.255  Mask:255.255.255.0
          inet6 addr: 2408:8207:2530:36c0:2e0:4cff:fe68:bc5/64 Scope:Global
          inet6 addr: fe80::2e0:4cff:fe68:bc5/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:9997302 errors:0 dropped:0 overruns:0 frame:0
          TX packets:10133328 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:14237104664 (13.2 GiB)  TX bytes:12286200619 (11.4 GiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:223204 errors:0 dropped:0 overruns:0 frame:0
          TX packets:223204 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:88468777 (84.3 MiB)  TX bytes:88468777 (84.3 MiB)

ovs-syste Link encap:Ethernet  HWaddr CA:AC:99:33:4B:F9  
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

bb-qq, 2024-11-20:
Can you disable auto-negotiation?
https://github.com/bb-qq/r8152/wiki/Troubleshooting#disable-auto-negotiation

This symptom may also occur when using poor-quality cables.

1x2y3z4, 2024-11-21:
I tried `sudo ethtool -s eth2 autoneg off advertise 0x802f` on kernel `4.4.180+`,  auto negotiation cannot be turned off:
```
Cannot set new settings: Invalid argument
  not setting autoneg
```

bb-qq, 2024-11-23:
Sorry, but I can't think of any other workarounds.

By the way, does this problem not occur when the adapter in question is connected to a PC?
To me it looks like a problem on the router side, so I would think the same symptoms would occur when connected to a PC.