bb-qq/r8152 #457: D-Link DUB-2315 only negotiates 100Mb/s

sardaukar, 2024-11-16:
### Description of the problem

I have two [DUB-2315](https://www.dlink.com/si/sl/products/dub-2315-usb-c-to-2-5g-ethernet-adapter)s connecting two Synologys, a DS-1621+ and a DS-923+. I've installed version 2.18.1-1 of the driver on both of them (V1000 and R1000 respectively) and after the SSH fix, the driver works fine and detects both devices. I have a CAT6 U/UTP 26AWG cable connecting them. However, setting the MTU isn't possible over 1500 on the UI and they only negotiate 100Mb/s. 

### Description of your products

`uname -a` on the DS1621+ running DSM 7.2.2-72806 Update 1:

```
Linux syno 4.4.302+ #72806 SMP Thu Sep 5 13:45:50 CST 2024 x86_64 GNU/Linux synology_v1000_1621+
```

and on the 923+ running DSM 7.2.2-72806 Update 1:

```
Linux tiny 4.4.302+ #72806 SMP Thu Sep 5 13:41:01 CST 2024 x86_64 GNU/Linux synology_r1000_923+
```

### Description of your environment

U/UTP 26AWG CAT6 cable connecting both dongles that are on the back USB port of the devices.

### Relevant output of ``dmesg`` command

On the DS1621+

````
[ 3216.997695] usb 2-2: reset SuperSpeed USB device number 2 using xhci_hcd
[ 3217.077964] r8152 2-2:1.0 eth4: v2.18.1 (2024/05/20)
[ 3217.082948] r8152 2-2:1.0 eth4: This product is covered by one or more of the following patents:
                        US6,570,884, US6,115,776, and US6,327,625.

[ 3217.098496] r8152 2-2:1.0 eth4: chip rev 14
[ 3217.102765] usbcore: registered new interface driver r8152
[ 3217.170876] IPv6: ADDRCONF(NETDEV_UP): eth4: link is not ready
[ 3218.347476] systemd[1]: Started Avahi daemon.
[ 3238.863906] IPv6: ADDRCONF(NETDEV_CHANGE): eth4: link becomes ready
[ 3238.870604] r8152 2-2:1.0 eth4: carrier on
````

On the 923+

````
[ 3573.970339] usbcore: deregistering interface driver r8152
[ 3625.619017] usb 2-3: reset SuperSpeed USB device number 4 using xhci_hcd
[ 3625.702122] r8152 2-3:1.0 eth2: v2.18.1 (2024/05/20)
[ 3625.707097] r8152 2-3:1.0 eth2: This product is covered by one or more of the following patents:
               		US6,570,884, US6,115,776, and US6,327,625.

[ 3625.722638] r8152 2-3:1.0 eth2: chip rev 14
[ 3625.726838] usbcore: registered new interface driver r8152
[ 3625.778303] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[ 3628.021992] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready
````

### Output of ``lsusb`` command

1621+

````
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:0b:00.3) hub
  |__1-1         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 650023C98881C618)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:0b:00.3) hub
  |__2-2         0bda:8156:3104 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 401000001)
|__usb3          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:0b:00.4) hub
  |__3-1         051d:0002:0106 00  1.10  1.5MBit/s 2mA 1IF  (APC Back-UPS ES 700G FW:871.O3 .I USB FW:O3  5B1339T04490  )
|__usb4          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:0b:00.4) hub
````

923+

````
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:08:00.3) hub
  |__1-1         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 6500426B912DF971)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:08:00.3) hub
  |__2-2         1058:25ed:1031 00  3.10 5000MBit/s 8mA 1IF  (Western Digital My Book 25ED 3551475A474B5946)
  |__2-3         0bda:8156:3104 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 401000001)

````

### Output of ``ifconfig -a`` command

For relevant interface on 923+

````
eth2      Link encap:Ethernet  HWaddr 00:E0:4C:44:59:EE  
          inet addr:10.10.10.2  Bcast:10.10.10.255  Mask:255.255.255.0
          inet6 addr: fe80::2e0:4cff:fe44:59ee/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:69 errors:0 dropped:0 overruns:0 frame:0
          TX packets:120 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:12050 (11.7 KiB)  TX bytes:19510 (19.0 KiB)
````

on 1621+

````
eth4      Link encap:Ethernet  HWaddr 00:E0:4C:44:5A:92  
          inet addr:10.10.10.1  Bcast:10.10.10.255  Mask:255.255.255.0
          inet6 addr: fe80::2e0:4cff:fe44:5a92/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:51 errors:0 dropped:0 overruns:0 frame:0
          TX packets:88 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:6614 (6.4 KiB)  TX bytes:13824 (13.5 KiB)
````

ethtool on 1621+

````
Settings for eth4:
        Supported ports: [ MII ]
        Supported link modes:   10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Full 
                                2500baseX/Full 
        Supported pause frame use: No
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  10baseT/Half 10baseT/Full 
                                             100baseT/Half 100baseT/Full 
        Link partner advertised pause frame use: No
        Link partner advertised auto-negotiation: Yes
        Link partner advertised FEC modes: Not reported
        Speed: 100Mb/s
        Duplex: Full
        Port: MII
        PHYAD: 32
        Transceiver: internal
        Auto-negotiation: on
        Supports Wake-on: pumbg
        Wake-on: g
        Current message level: 0x00007fff (32767)
                               drv probe link timer ifdown ifup rx_err tx_err tx_queued intr tx_done rx_status pktdata hw wol
        Link detected: yes
````

on 923+

````
Settings for eth2:
	Supported ports: [ MII ]
	Supported link modes:   10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Full 
	                        2500baseX/Full 
	Supported pause frame use: No
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full 
	                                     100baseT/Half 100baseT/Full 
	Link partner advertised pause frame use: No
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 100Mb/s
	Duplex: Full
	Port: MII
	PHYAD: 32
	Transceiver: internal
	Auto-negotiation: on
	Supports Wake-on: pumbg
	Wake-on: g
	Current message level: 0x00007fff (32767)
			       drv probe link timer ifdown ifup rx_err tx_err tx_queued intr tx_done rx_status pktdata hw wol
	Link detected: ye
````

I have tried `ethtool -s eth2(and 4) autoneg on advertise 0x802f` but it just disconnects them, and returns them to 100Mb/s. I've also tried with another CAT5E cable, with the same result.

sardaukar, 2024-11-16:
I also tried the pre-release version of the driver, with the same result

sardaukar, 2024-11-16:
After checking the adapters with a connection from my laptop to my PC, I see the same behavior so I guess they're faulty somehow, closing this ticket.

closed 2024-11-16 by sardaukar

thekondrashov, 2024-11-17:
try to replacing the cable with a shielded cat 6. should be bad cable.
I use a few of such D-Link adapters with the linux 6.11 kernel and the latest v2.19.2 driver from the Realtek.

```sh
$ uname -a
Linux 6.11.2-4-MANJARO #1 SMP PREEMPT_DYNAMIC Tue Oct  8 11:52:01 UTC 2024 x86_64 GNU/Linux

$ sudo dmesg --time-format iso
2024-11-17T21:08:08,660080+04:00 usb 2-4: new SuperSpeed USB device number 2 using xhci_hcd
2024-11-17T21:08:08,678329+04:00 usb 2-4: New USB device found, idVendor=0bda, idProduct=8156, bcdDevice=31.04
2024-11-17T21:08:08,678338+04:00 usb 2-4: New USB device strings: Mfr=1, Product=2, SerialNumber=6
2024-11-17T21:08:08,678342+04:00 usb 2-4: Product: USB 10/100/1G/2.5G LAN
2024-11-17T21:08:08,678344+04:00 usb 2-4: Manufacturer: Realtek
2024-11-17T21:08:08,678347+04:00 usb 2-4: SerialNumber: 401000001
2024-11-17T21:08:12,836331+04:00 usbcore: registered new interface driver cdc_ether
2024-11-17T21:08:12,842558+04:00 usbcore: registered new interface driver cdc_ncm
2024-11-17T21:08:12,848470+04:00 usbcore: registered new interface driver cdc_wdm
2024-11-17T21:08:12,853804+04:00 usbcore: registered new interface driver cdc_mbim
2024-11-17T21:08:12,886223+04:00 usbcore: registered new device driver r8152-cfgselector
2024-11-17T21:08:12,915537+04:00 cdc_ncm 2-4:2.0: MAC-Address: 00:e0:4c:xx:xx:xx
2024-11-17T21:08:12,915541+04:00 cdc_ncm 2-4:2.0: setting rx_max = 16384
2024-11-17T21:08:12,915602+04:00 cdc_ncm 2-4:2.0: setting tx_max = 16384
2024-11-17T21:08:12,915863+04:00 cdc_ncm 2-4:2.0 eth0: register 'cdc_ncm' at usb-0000:00:14.0-4, CDC NCM (NO ZLP), 00:e0:4c:xx:xx:xx
2024-11-17T21:08:12,916533+04:00 cdc_ncm 2-4:2.0 eth0: unregister 'cdc_ncm' usb-0000:00:14.0-4, CDC NCM (NO ZLP)
2024-11-17T21:08:13,090344+04:00 r8152-cfgselector 2-4: reset SuperSpeed USB device number 2 using xhci_hcd
2024-11-17T21:08:13,170532+04:00 r8152 2-4:1.0 eth0: v2.19.2 (2024/10/21)
2024-11-17T21:08:13,170535+04:00 r8152 2-4:1.0 eth0: This product is covered by one or more of the following patents:
                                                US6,570,884, US6,115,776, and US6,327,625.

2024-11-17T21:08:13,170557+04:00 usbcore: registered new interface driver r8152
2024-11-17T21:08:13,211324+04:00 r8152 2-4:1.0 enp0s20f0u4: renamed from eth0
2024-11-17T21:08:16,689397+04:00 r8152 2-4:1.0 enp0s20f0u4: carrier on
2024-11-17T21:08:20,913396+04:00 r8152 2-4:1.0 enp0s20f0u4: carrier off
2024-11-17T21:08:24,497222+04:00 r8152 2-4:1.0 enp0s20f0u4: carrier on

$ sudo lsusb
Bus 002 Device 002: ID 0bda:8156 Realtek Semiconductor Corp. USB 10/100/1G/2.5G LAN

$ ethtool -i enp0s20f0u4
driver: r8152
version: v2.19.2 (2024/10/21)
firmware-version: 
expansion-rom-version: 
bus-info: usb-0000:00:14.0-4
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no

$ ethtool enp0s20f0u4
Settings for enp0s20f0u4:
        Supported ports: [ MII ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
                                2500baseT/Full
        Supported pause frame use: No
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
                                2500baseT/Full
        Advertised pause frame use: No
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  100baseT/Half 100baseT/Full
                                             1000baseT/Full
                                             2500baseT/Full
        Link partner advertised pause frame use: No
        Link partner advertised auto-negotiation: Yes
        Link partner advertised FEC modes: Not reported
        Speed: 2500Mb/s
        Duplex: Full
        Auto-negotiation: on
        Port: MII
        PHYAD: 32
        Transceiver: internal
netlink error: Operation not permitted
        Current message level: 0x00007fff (32767)
                               drv probe link timer ifdown ifup rx_err tx_err tx_queued intr tx_done rx_status pktdata hw wol
        Link detected: yes
```

sardaukar, 2024-11-18:
One of the units I was using was only negotiating to 100Mb for some reason, the other was fine. I've replaced it and it just works now.