bb-qq/r8152 #458: WP-UT5 RTL8157

Sjekke, 2024-11-16:
I live in Belgium an dI'm for many years a fan of this solution of BBQQ. Now I want to test the WisdPi 5Gbps Base-T Ethernet Adapter ... but where can I buy it in he EU/Belgium? All Amazon in EU don't have them. 

Thx.
Serge

PS. You can always sent me one :-)

TheS1R, 2024-11-16:
Try wisdpi.com.

Sjekke, 2024-11-17:
The adapter + shipping + tax in Belgium of 21% , that adapter would cost me around 60$. No thanks ...

Cruistar, 2024-12-10:
WP-UT5 RTL8157: [Ali-express](https://vi.aliexpress.com/item/1005008134517506.html?spm=a2g0o.productlist.main.3.48ad2161o9n2Dw&algo_pvid=8c78c3d1-6b26-40e4-8931-78aa41658568&algo_exp_id=8c78c3d1-6b26-40e4-8931-78aa41658568-1&pdp_npi=4%40dis%21GBP%2137.49%2129.99%21%21%2146.41%2137.13%21%40211b81b117338267954098569e947c%2112000043933215427%21sea%21UK%213316145269%21X&curPageLogUid=IUO9hMpaOulF&utparam-url=scene%3Asearch%7Cquery_from%3A)
Same chip: https://vi.aliexpress.com/item/1005007869954253.html?src=google&gatewayAdapt=glo2vnm