bb-qq/r8152 #459: DS415+, adapter not showing up on rear USB3 ports, but showing up on front USB2 port

david-christ, 2024-11-19:
### Description of the problem

USB to 2.5GbE aadapter not showing up on rear USB3 ports, but showing up when connecting to front USB2 port.

I tried connecting a USB3 HDD to the rear port, and it seems to just work fine.

### Description of your products

> - DS415+
>     - `Linux xoanon 3.10.108 #42962 SMP Mon May 29 14:36:28 CST 2023 x86_64 GNU/Linux synology_avoton_415+`
>     - DSM version 7.1.1-42962 Update 6
> - Using [this adapter from AliExpress](https://www.aliexpress.com/item/1005007075357588.html), using an RTL8156B

### Description of your environment

> - connect directly to a PC or using a Ethernet switch
>     - PC: NIC model name or chipset name
>     - Ethernet switch: model name
> - cable category (Cat.5e or Cat.6 or Cat.7)

### Output of ``dmesg`` command

[dmesg.txt](https://github.com/user-attachments/files/17809308/dmesg.txt)

### Output of ``lsusb`` command

When just plugging the USB to ethernet adapter into the rear, no output is shown

```
$ lsusb
|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:16.0) hub
  |__1-1         8087:07db:0002 09  2.00  480MBit/s 0mA 1IF  ( ffffffd1ffffffb2ffffffdbffffffad) hub
    |__1-1.1     f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskSataion 6500496912B76D16)
|__usb2          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
|__usb3          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
```

The following shows the USB to ethernet adapter showing up on the front USB port, and a USB3 HDD on the rear

````
$ lsusb
|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:16.0) hub
  |__1-1         8087:07db:0002 09  2.00  480MBit/s 0mA 1IF  ( ffffffd1ffffffb2ffffffdbffffffad) hub
    |__1-1.1     f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskSataion 6500496912B76D16)
    |__1-1.3     0bda:8156:3104 00  2.10  480MBit/s 200mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 4013000001)
|__usb2          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
|__usb3          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__3-1         1058:25a3:1030 00  3.10 5000MBit/s 8mA 1IF  (Western Digital Elements 25A3 355147394E303446)
````

### Output of ``ifconfig -a`` command

````
$ ifconfig -a
eth0      Link encap:Ethernet  HWaddr 00:11:32:XX:XX:F1
          inet addr:192.168.1.2  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: 2401:d002:3a09:6200::1005/128 Scope:Global
          inet6 addr: fe80::211:32ff:fe38:1cf1/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:360125649 errors:0 dropped:0 overruns:0 frame:0
          TX packets:343473665 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:314621864765 (293.0 GiB)  TX bytes:496248381485 (462.1 GiB)

eth1      Link encap:Ethernet  HWaddr 00:11:32:XX:XX:F2
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:5387 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5387 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:823763 (804.4 KiB)  TX bytes:823763 (804.4 KiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tailscale Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          inet addr:100.100.100.2  P-t-P:100.100.100.2  Mask:255.255.255.255
          inet6 addr: fd7a:115c:a1e0::a301:dd0a/128 Scope:Global
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1280  Metric:1
          RX packets:117552 errors:0 dropped:0 overruns:0 frame:0
          TX packets:136068 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500
          RX bytes:6852708 (6.5 MiB)  TX bytes:321703950 (306.8 MiB)
````

bb-qq, 2024-11-23:
Whether a device is listed in `lsusb` or not is related to the USB host controller and host driver. USB class drivers like my driver cannot be involved.

I believe the problem is occurring at the hardware level, so your action in using a powered USB hub is correct.
If the problem persists despite this, I am sorry, but I cannot come up with a better solution.

david-christ, 2024-11-30:
It's very weird. The device works as it should on my Linux laptop, but not on the NAS. I bought one of the adapters that had been tested as working in the compatibility list, and it works. 🤷

bb-qq, 2024-12-31:
That really is weird. Maybe the connector is not well made and there is an electrical contact or some other problem.
I have no further advice that I can give you, so if you don't mind, I would like to close this issue.

david-christ, 2025-01-14:
Sure, go ahead. I bought an adaptor that had a report as "working" in the compatibility list for my NAS, and it works just fine.

closed 2025-01-14 by david-christ