bb-qq/r8152 #464: Should 2 usb 2.5G adapters work on one Diskstation?

alexz707, 2024-12-09:
HI! 
just tried some cheap ali express usb 2.5G Adapters on the diskstation.
with one plugged in I get a 3 lan adapter in DSM but when I plug in a second one it doesn't show up?
Any ideas what I can do? Stopping/starting the driver or restarting the diskstation doesn't help (DS620slim)

If I remove the first and plug in the second it shows up as soon as I restart the driver...

Cruistar, 2024-12-10:
Yes.
I just solved that out and I believe I used the same Ali express adapters as you mentioned.
After reading others comments elsewhere, I realised you have to change the maxlanport.
By default, my 920+ limits the amount of lan to 2. Fortunately you can change it.

There are two documents you need to change:
1. vi /etc.defaults/synoinfo.conf
2. vi /etc/synoinfo.conf
**Process:**

1. Log in as ssh, and then get root access, same as what you did when installing the driver.
2. Type in vi /etc.defaults/synoinfo.conf
3. Scroll down and find this line: maxlanport=「2」
4. press i in you keyboard to enter edit mode.
5. Change number 2 to 4 ( this will change the max lan number to 4)
6. Press Esc
7. Type :wq (Save and exit)
8. Repeat step 2->7  for vi /etc/synoinfo.conf
9. Restart your NAS

Hopefully after that you are able to see 4 lan in your NAS.
If so, you can enable SMB multi-channel for those two 2.5G ports and get 5G.
Fingers crossed.

knowhatamine, 2024-12-13:
Haha that is awesome. Pretty sure I got the same adapters as you guys. ;)

knowhatamine, 2024-12-14:
Nope, I've already had 8 adapters allowed. 

This however did the trick:

# bash /var/packages/r8152/scripts/install-udev-rules
Updating Hardware Database Index...
UDEV rules have been installed to /usr/lib/udev/rules.d
lrwxrwxrwx 1 root root 50 Dec 14 00:24 /usr/lib/udev/rules.d/51-usb-r8152-net.rules -> /var/packages/r8152/scripts/51-usb-r8152-net.rules

Can't say Link Aggregation is working very well though. I barely surpass 1 adapter's speed. Maybe 40% peak gain without using Synology's SMB implementation. I really regret buying this piece of shit. Synology is almost worse than Microshit who completely removed LACP drivers from Windows.

But hey, today my  X520-DA2 (2x10G SFP+) adapter arrived which I got for 35 bucks on AliExpress. 

Will report back if it's working or not.

All hail the great Ali!

bb-qq, 2024-12-31:
The problem seems to be resolved, so this Issue is closed. If you have any other questions, feel free to re-open.

closed 2024-12-31 by bb-qq