bb-qq/r8152 #465: Kernelmodul Call trace - Not working

falcon040, 2024-12-17:
DS418 : Latest Update 

DSM 7.2.2-72806 Update 2
Realtek RTD1296 SoC
USB 10/100/1G/2.5G LAN - Realtek Semiconductor Corp.

Network Resets directly and hangs starting a File transfer to PC. Mounting smb works. Filetransfer hangs.

2024-12-17T16:35:06+01:00 ds418 synofinderdb[11438]: synofinderdb.cpp:101 (main) synofinderdb tool desc: update synofinder.db

2024-12-17T16:36:30+01:00 ds418 kernel: [  220.151952] xhci-hcd xhci-hcd.2.auto: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 3 comp_code 1
2024-12-17T16:36:30+01:00 ds418 kernel: [  220.162736] xhci-hcd xhci-hcd.2.auto: Looking for event-dma 000000007e18e5f0 trb-start 000000007e18e4e0 trb-end 000000007e18e4e0 seg-start 000000007e18e000 seg-end 000000007e18eff0
2024-12-17T16:36:30+01:00 ds418 kernel: [  220.179199] xhci-hcd xhci-hcd.2.auto: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 3 comp_code 4
2024-12-17T16:36:30+01:00 ds418 kernel: [  220.189972] xhci-hcd xhci-hcd.2.auto: Looking for event-dma 000000007e18e680 trb-start 000000007e18e4e0 trb-end 000000007e18e4e0 seg-start 000000007e18e000 seg-end 000000007e18eff0
2024-12-17T16:36:46+01:00 ds418 kernel: [  235.987630] ------------[ cut here ]------------
2024-12-17T16:36:46+01:00 ds418 kernel: [  235.992356] WARNING: at net/sched/sch_generic.c:307
2024-12-17T16:36:46+01:00 ds418 kernel: [  235.997347] Modules linked in: fuse r8152(OE) vhost_scsi(O) vhost(O) tcm_loop(O) iscsi_target_mod(O) target_core_user(O) target_core_ep(O) target_core_multi_file(O) target_core_file(O) target_core_iblock(O) target_core_mod(O) syno_extent_pool(PO) rodsp_ep(O) udf isofs synoacl_vfs(O) btrfs ecryptfs zstd_decompress zstd_compress xxhash raid456 async_raid6_recov async_memcpy async_pq raid6_pq async_xor xor async_tx 8021q uas usb_storage syno_hddmon(P) ds418_synobios(O) hid_generic usbhid hid usblp uhci_hcd r8168(O) zlib_deflate sg dm_snapshot dm_bufio dm_mod crc_itu_t crc_ccitt psnap p8022 llc hfsplus md4 hmac sit tunnel4 ipv6 arc4 aes_neon_blk aes_ce_blk gf128mul ablk_helper cryptd ecb authenc des_generic ansi_cprng cts md5 cbc vxlan ip6_udp_tunnel udp_tunnel ip_tunnel loop sha256_generic synorbd(O)
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.070398]  synofsbd(O) dwc3_rtk xhci_pci xhci_plat_hcd xhci_hcd ehci_pci ehci_hcd dwc3 usbcore phy_rtk_usb3 phy_rtk_usb2 phy_rtk_rle0599 usb_common [last unloaded: ds418_synobios]
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.085780]
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.087310] CPU: 2 PID: 0 Comm: swapper/2 Tainted: P           OE   4.4.302+ #72806
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.095147] Hardware name: Realtek_RTD1296 (DT)
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.099784] task: ffffffc077503000 ti: ffffffc07750c000 task.ti: ffffffc07750c000
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.107450] PC is at dev_watchdog+0x234/0x24c
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.111910] LR is at dev_watchdog+0x234/0x24c
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.116369] pc : [<ffffffc0006eaab8>] lr : [<ffffffc0006eaab8>] pstate: 60000145
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.123939] sp : ffffffc07750fc50
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.127331] x29: ffffffc07750fc50 x28: 0000000000000100
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.132790] x27: ffffffc07750c000 x26: 0000000000000005
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.138249] x25: 00000000000000a0 x24: 00000000ffffffff
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.143708] x23: 0000000000000002 x22: ffffffc0009e4000
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.149168] x21: ffffffc00099e000 x20: ffffffc074ee5000
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.154628] x19: 0000000000000000 x18: 0000000000000006
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.160088] x17: ffffffbdc1d1f320 x16: ffffffc07531e500
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.165548] x15: 0000000000000006 x14: ffffffc0809eaa0f
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.171008] x13: ffffffc0009eaa17 x12: ffffffc0009ea4a8
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.176467] x11: 0000000000000049 x10: 000000000000037e
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.181927] x9 : 00000000000ef805 x8 : 756575712074696d
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.187386] x7 : ffffffc00099ea48 x6 : ffffffc000591820
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.192844] x5 : 0000000000000000 x4 : 0000000000000000
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.198304] x3 : 0000000000000000 x2 : ffffffc07df98230
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.203763] x1 : ffffffc07df96ae0 x0 : 0000000000000039
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.209223]
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.210748] Call trace:
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.213254] [<ffffffc0006eaab8>] dev_watchdog+0x234/0x24c
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.218784] [<ffffffc0002ec2d8>] call_timer_fn.constprop.0+0x28/0x80
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.225290] [<ffffffc0002ecd64>] run_timer_softirq+0x160/0x2bc
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.231263] [<ffffffc00029b5f0>] __do_softirq+0x100/0x270
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.236790] [<ffffffc00029b9b4>] irq_exit+0x84/0xc0
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.241786] [<ffffffc0002ddd24>] __handle_domain_irq+0x74/0xd0
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.247758] [<ffffffc000280d24>] gic_handle_irq+0x54/0xc0
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.253284] Exception stack(0xffffffc07750fdf0 to 0xffffffc07750ff20)
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.259877] fde0:                                   0000000000000000 0000008000000000
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.267892] fe00: ffffffc07750ff50 ffffffc000284d70 0000000000000145 ffffffc0009860e8
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.275908] fe20: 0000000000000000 0000000000000000 ffffffc00099f8f0 0000000000000000
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.283923] fe40: 000000007d613000 00000000000001c0 00000000ffffe6fe ffffffc07df99f38
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.291938] fe60: ffffffc0775038c0 ffffffc07750feb0 0000000000000860 ffffffbdc0000000
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.299954] fe80: 0000004000000000 ffffffc077401308 ffffffc077401328 0000000000000001
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.307970] fea0: ffffffc07531e500 ffffffbdc1d1f320 000000000000000a 0000000000000000
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.315986] fec0: 0000000000000000 ffffffc07750c000 ffffffc07750ff60 ffffffc000983ac0
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.324001] fee0: ffffffc0009860e8 ffffffc000786778 ffffffc00099ea48 ffffffc0009e44eb
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.332017] ff00: ffffffc00099ead8 ffffffc07750ff50 ffffffc000284d6c ffffffc07750ff50
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.340034] [<ffffffc000283f80>] el1_irq+0x80/0x100
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.345029] [<ffffffc0002d1630>] cpu_startup_entry+0x1f0/0x280
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.351001] [<ffffffc00028b630>] secondary_start_kernel+0x11c/0x140
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.357417] [<000000000028111c>] 0x28111c
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.361520] ---[ end trace 2fe79443b7d4fe45 ]---
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.366285] r8152 3-1:1.0 eth2: Tx timeout
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.372835] r8152 3-1:1.0 eth2: Tx status -2
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.377230] r8152 3-1:1.0 eth2: Tx status -2
2024-12-17T16:36:46+01:00 ds418 kernel: [  236.381620] r8152 3-1:1.0 eth2: Tx status -2
[... 18 lines cut ...]
2024-12-17T16:36:49+01:00 ds418 kernel: [  238.573129] x23: ffffffc04f3dbbec x22: ffffffc04f3dbbae
2024-12-17T16:36:49+01:00 ds418 kernel: [  238.578589] x21: 000000000000a000 x20: ffffffc04f3dbb4c
2024-12-17T16:36:49+01:00 ds418 kernel: [  238.584049] x19: 0000000000000000 x18: 0000000000000006
2024-12-17T16:36:49+01:00 ds418 kernel: [  238.589508] x17: 0000007f9ec27080 x16: ffffffc0006ad2a4
2024-12-17T16:36:49+01:00 ds418 kernel: [  238.594968] x15: 0000000000000006 x14: ffffffc0809eaa0f
2024-12-17T16:36:49+01:00 ds418 kernel: [  238.600428] x13: ffffffc0009eaa17 x12: ffffffc0009ea4a8
2024-12-17T16:36:49+01:00 ds418 kernel: [  238.605888] x11: 0000000000000047 x10: 00000000000003b8
[... cut ...]
2024-12-17T16:36:49+01:00 ds418 kernel: [  238.673748] [<ffffffc0006e1d98>] dev_ioctl+0x118/0x6a0
2024-12-17T16:36:49+01:00 ds418 kernel: [  238.679011] [<ffffffc0006ab494>] sock_ioctl+0x2e4/0x310
2024-12-17T16:36:49+01:00 ds418 kernel: [  238.684362] [<ffffffc00039a958>] do_vfs_ioctl+0xa48/0xd10
2024-12-17T16:36:49+01:00 ds418 kernel: [  238.689890] [<ffffffc00039acd4>] SyS_ioctl+0xb4/0xd0
2024-12-17T16:36:49+01:00 ds418 kernel: [  238.694974] [<ffffffc000284670>] el0_svc_naked+0x24/0x28
2024-12-17T16:36:49+01:00 ds418 kernel: [  238.700410] ---[ end trace 2fe79443b7d4fe46 ]---

falcon040, 2024-12-17:
USb Adapter:
Asus USB-C2500 and Ugreen 2.5G both tested and both hang.

bb-qq, 2024-12-31:
This problem appears to be caused by the rtd1296 platform. See #275 for more information. This Issue is closed to integrate information.

closed 2024-12-31 by bb-qq