bb-qq/r8152 #466: How to install diver on DS218j directly from command line

giorgosbotis, 2024-12-18:
I got a DS218j with a failed ethernet port (after a lightning strike burned the router). 

I have a working connection to the serial port of the NAS, it boots normally and usb ports work perfectly. I am trying to install divers for a TP Link UA300 usb ethernet adapter. 

The problem is that the NAS run on busybox 1.6 that does not have 'make' so I cannot install the drivers. Does anyone has any idea of a possible solution?

bb-qq, 2024-12-31:
`synopkg install` command should work. Alternatively, extract the driver package and install `r8152.ko` manually.

giorgosbotis, 2024-12-31:
Unfortunately `synopkg` is also not available. What steps should I follow to install the driver manually? Should I just copy the file to a specific folder?

bb-qq, 2025-01-02:
Just copy the file and load the module by the `insmod` command.
Then, if a network connection is obtained, continue with the normal installation procedure.

closed 2025-06-08 by bb-qq