bb-qq/r8152 #468: RTL8157 (WavLink 5G) not working on DS916+

gut2ssomdn, 2024-12-28:
### Description of the problem

> Never really worked on DS916+ (Braswell). It kind of installs and you briefly see eth2 and ovs_eth2 but the system becomes very slow / unstable, and after rebooting DSM has automatically stopped the package and it comes back to a more normally working situation. Dmesg logs seem to indicate an issue with the driver.

### Description of your products

> - DSM916+
>     - output of ``uname -a`` command: Linux [hostname] 3.10.108 #72806 SMP Thu Sep 5 13:39:20 CST 2024 x86_64 GNU/Linux synology_braswell_916+
>     - DSM version 7.2 (latest update)
> - the product name of the ethernet dongle: WL-NWU340G (WavLink 5G adaptor based on RTL8157)

### Description of your environment

> - connect using a Ethernet switch
>     - Ethernet switch: GS110EMX
> - cable category: Cat.6

### Output of ``dmesg`` command

````

[  927.837266] r8152: module verification failed: signature and/or required key missing - tainting kernel
[  927.849588] usbcore: registered new interface driver r8152
[  927.981423] usb 2-2: reset SuperSpeed USB device number 2 using xhci_hcd
[  927.999586] usb 2-2: Parent hub missing LPM exit latency info.  Power management will be impacted.
[  928.010402] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff880467fd9540
[  928.020554] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff880467fd9580
[  928.030737] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff880467fd95c0
[  928.045149] docker0: port 1(dockerb264e5e) entered disabled state
[  928.052752] device dockerb264e5e left promiscuous mode
[  928.058547] docker0: port 1(dockerb264e5e) entered disabled state
[  928.093751] r8152 2-2:1.0: ram code version is wrong 0000(0013
[  928.250531] r8152 2-2:1.0 eth2: v2.19.2 (2024/10/21)
[  928.256120] r8152 2-2:1.0 eth2: This product is covered by one or more of the following patents:
                        US6,570,884, US6,115,776, and US6,327,625.

[  928.273599] r8152 2-2:1.0 eth2: chip rev 19
[  929.098175] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[  929.248961] device ovs_eth2 entered promiscuous mode
[  929.284489] device eth2 entered promiscuous mode
[  929.312900] IPv6: ADDRCONF(NETDEV_UP): ovs_eth2: link is not ready
[  941.401438] r8152 2-2:1.0 eth2: Promiscuous mode enabled
[  941.412836] r8152 2-2:1.0 eth2: carrier on
[  941.427404] IPv6: ADDRCONF(NETDEV_CHANGE): ovs_eth2: link becomes ready
[  942.291340] device docker9b5d5a5 entered promiscuous mode
[  942.300500] IPv6: ADDRCONF(NETDEV_UP): docker9b5d5a5: link is not ready
[  942.307962] docker0: port 1(docker9b5d5a5) entered forwarding state
[  942.315026] docker0: port 1(docker9b5d5a5) entered forwarding state
[  942.401461] docker0: port 1(docker9b5d5a5) entered disabled state
[  972.103319] r8152 2-2:1.0: read type=0x0033, index=0xd634 fail -110
[  972.110446] ------------[ cut here ]------------
[  972.115638] WARNING: at //source/r8152/r8152.c:1400 wait_tgphy_cmd_ready+0x92/0xa0 [r8152]()
[  972.125133] Modules linked in: r8152(OF) pci_stub vhost_net kvm_intel kvm nfnetlink xfrm_user xfrm_algo fuse bridge stp aufs macvlan veth xt_addrtype ipt_MASQUERADE xt_REDIRECT xt_nat tun vhost_scsi(O) vhost(O) tcm_loop(O) iscsi_target_mod(O) target_core_ep(O) target_core_multi_file(O) target_core_file(O) target_core_iblock(O) target_core_mod(O) syno_extent_pool(PO) rodsp_ep(O) vfat fat udf isofs synoacl_vfs(O) btrfs zstd_decompress ecryptfs zstd_compress xxhash iptable_nat nf_nat_ipv4 nf_nat xt_conntrack xt_mac xt_geoip(O) compat_xtables(O) nf_conntrack_ipv6 ip6table_filter raid456 ip6_tables async_raid6_recov async_memcpy async_pq xt_recent xt_iprange xt_limit xt_state raid6_pq async_xor xt_tcpudp xt_multiport xor async_tx xt_LOG nf_conntrack_ipv4 nf_defrag_ipv4 iptable_filter ip_tables x_tables
[  972.204840]  8021q leds_lp3943 aesni_intel glue_helper lrw gf128mul ablk_helper syno_hddmon(P) braswell_synobios(O) hid_generic usbhid hid usblp openvswitch(O) gre nf_defrag_ipv6 nf_conntrack i915 drm_kms_helper cfbfillrect cfbcopyarea cfbimgblt output drm fb fbdev intel_agp intel_gtt agpgart video backlight button uhci_hcd ehci_pci ehci_hcd usb_storage i2c_algo_bit igb(O) e1000e(O) sg dm_snapshot crc_itu_t crc_ccitt psnap p8022 llc zlib_deflate libcrc32c hfsplus md4 hmac sit tunnel4 ipv6 flashcache_syno(O) flashcache(O) syno_flashcache_control(O) dm_mod crc32c_intel cryptd arc4 sha256_generic sha1_generic ecb aes_x86_64 authenc des_generic ansi_cprng cts md5 cbc cpufreq_powersave cpufreq_performance acpi_cpufreq mperf processor thermal_sys cpufreq_stats freq_table vxlan ip_tunnel loop etxhci_hcd
[  972.283282]  xhci_hcd usbcore usb_common [last unloaded: braswell_synobios]
[  972.289660] CPU: 1 PID: 11937 Comm: snmpd Tainted: PF          O 3.10.108 #72806
[  972.297947] Hardware name: Synology Inc. DS916+/Type2 - Board Product Name, BIOS M.215 3/2/2016
[  972.307711]  ffffffff81490ef2 ffffffff81481ab9 0000000000000000 ffff88040f426700
[  972.316144]  0000000000000000 ffff880449d67c5e 0000000000000001 ffffffffa0b96802
[  972.324482]  ffffff920f426000 ffffffffa0b988bb ffff88040f426700 000000000000a402
[  972.332809] Call Trace:
[  972.335617]  [<ffffffff81490ef2>] ? dump_stack+0xc/0x15
[  972.341500]  [<ffffffff81481ab9>] ? warn_slowpath_common+0x56/0x6c
[  972.348450]  [<ffffffffa0b96802>] ? wait_tgphy_cmd_ready+0x92/0xa0 [r8152]
[  972.356184]  [<ffffffffa0b988bb>] ? ocp_write_word+0x4b/0x90 [r8152]
[  972.363317]  [<ffffffffa0b98dd8>] ? rtl_tgphy_access+0x98/0xc0 [r8152]
[  972.370692]  [<ffffffffa0ba29cf>] ? rtl8152_get_settings+0x8f/0x340 [r8152]
[  972.378518]  [<ffffffff813e474f>] ? ethtool_get_settings+0x14f/0x170
[  972.385637]  [<ffffffff8114ffff>] ? copy_mnt_ns+0x10f/0x2a0
[  972.391889]  [<ffffffff811398bd>] ? path_init.constprop.0+0x41d/0x570
[... 85 lines cut ...]
[ 1138.904842] docker0: port 1(docker2f6109c) entered forwarding state
````

### Output of ``lsusb`` command

````

|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (xhci_hcd 0000:00:14.0) hub
  |__1-5         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 6500635D81E41679)
|__usb2          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (xhci_hcd 0000:00:14.0) hub
  |__2-2         0bda:8157:3000 00  3.20 5000MBit/s 544mA 1IF  (Realtek USB 10/100/1G/2.5G/5G LAN 0003803F5DFD66B5)
|__usb3          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__3-1         0463:ffff:0100 00  1.10   12MBit/s 20mA 1IF  (EATON Eaton 5SC G131M52041)
|__usb4          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__4-2         152d:0576:7102 00  3.10 5000MBit/s 8mA 1IF  (TerraMaster TDAS 20230705390F)
````

### Output of ``ifconfig -a`` command

````
Not showing anything interesting when the system is back to usable (in particular, no eth2 and no ovs_eth2)
````

I’ve tried with both 2.18 versions before with basically the same result. I’ve installed, cleaned uninstalled, reinstalled etc… I don’t know how many times with always the same result for all versions.
I’m able to perform tests if you clearly state them.

bb-qq, 2024-12-31:
It appears that the device has become unresponsive rather than a driver problem. Have you read [the troubleshooting page](https://github.com/bb-qq/r8152/wiki/Troubleshooting)?

gut2ssomdn, 2024-12-31:
Absolutely. And among the things I have tested:
- Confirmed the port on the switch works properly
- reversing the usb port: indeed, it only works when plugged in one direction (I’m using an C-> A adapter)
- I’m not using an L-shaped connector
- fully uninstalled and reinstalled multiple times (testing all versions from 2.18-1)
- applied the hot plugging command
- changed the cable

What else would it make sense for me to test ? Do you know of any report of it working on DS916+ ?

bb-qq, 2025-01-01:
There are several reports of RTL8156 on braswel platforms.

I am not sure, but since the RTL8156 and RTL8157 have different power consumption, running the device through a self-powered USB hub may help.

vishnunuk, 2025-02-08:
I am using ‎WLIN-NWU340G with RTL8157.
Getting around 3.3Gbps.
Link speed not showing. 

<br class="Apple-interchange-newline">
[image]
[image]
[image]

bb-qq, 2025-02-16:
It appears to be a low-layer problem, like a signal. 
What about using another TypeA-C conversion adapter? Also, does the adapter you are using now work reliably on your Windows PC?

vishnunuk, 2025-02-17:
@bb-qq I tried three different adapters and observed the same behavior. Maybe it's an issue with the old kernel or the USB 3.1 5Gbps port in the DS920+.

n3man1, 2025-04-21:
I am facing the same issue. Updated to latest driver which has added missing network speed. Was working fine and showing 5000mbps last night only to now show 1000mbps this morning (verified decrease in speed with speed tests).

Are we just boned with the Braswell architecture?